tsduck/tsduck #1: TBS DVB tuners not detected

rezna11, 2017-09-01:
TBS5925, TBS5922 nor DVBSky S960 are detected as input DVB devices. What DVB tuners are supported?

I have other tuners too, should I try those? Is there a list of compatible tuners?
```

C:\Users\Administrator>tsp -I dvb -d "TBS 5922 USB DVB-S2 Tuner"
* Error: dvb: DVB device "TBS 5922 USB DVB-S2 Tuner" not found

C:\Users\Administrator>tslsdvb -ve
tslsdvb: no DVB device found

=== Device category KSCATEGORY_BDA_NETWORK_PROVIDER

=== Device category KSCATEGORY_BDA_TRANSPORT_INFORMATION

=== Device category KSCATEGORY_CAPTURE

=== Device category KSCATEGORY_TVTUNER

=== Device category KSCATEGORY_BDA_RECEIVER_COMPONENT

device "TBS 5922 USB DVB-S2 Capture"
  Some supported interfaces:
    interface IAMDeviceRemoval
    interface IAMFilterMiscFlags
    interface IAMovieSetup
    interface IBaseFilter
    interface IKsControl
    interface IKsPropertySet
    interface IKsTopologyInfo
    interface IMediaFilter
    interface IMediaSeeking
  IKsPropertySet properties support:
  IKsControl properties support:
  IKsTopologyInfo:
    Categories: KSCATEGORY_BDA_RECEIVER_COMPONENT
    No node found

  - Pin "MPEG2 Transport", direction: input
    Some supported interfaces:
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMemInputPin
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

  - Pin "MPEG2 Transport", direction: output
    Some supported interfaces:
      interface IAMBufferNegotiation
      interface IAMStreamConfig
      interface IAMStreamControl
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMediaSeeking
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

device "TBS 5925 DVBS2 Capture"
  Some supported interfaces:
    interface IAMDeviceRemoval
    interface IAMFilterMiscFlags
    interface IAMovieSetup
    interface IBaseFilter
    interface IKsControl
    interface IKsPropertySet
    interface IKsTopologyInfo
    interface IMediaFilter
    interface IMediaSeeking
  IKsPropertySet properties support:
  IKsControl properties support:
  IKsTopologyInfo:
    Categories: KSCATEGORY_BDA_RECEIVER_COMPONENT
    No node found

  - Pin "MPEG2 Transport", direction: input
    Some supported interfaces:
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMemInputPin
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

  - Pin "MPEG2 Transport", direction: output
    Some supported interfaces:
      interface IAMBufferNegotiation
      interface IAMStreamConfig
      interface IAMStreamControl
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMediaSeeking
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

=== Device category KSCATEGORY_BDA_NETWORK_TUNER

device "TBS 5922 USB DVB-S2 Tuner"
  Some supported interfaces:
    interface IAMDeviceRemoval
    interface IAMFilterMiscFlags
    interface IAMovieSetup
    interface IBaseFilter
    interface IKsControl
    interface IKsPropertySet
    interface IKsTopologyInfo
    interface IMediaFilter
    interface IMediaSeeking
  IKsPropertySet properties support:
  IKsControl properties support:
  IKsTopologyInfo:
    Categories: KSCATEGORY_BDA_NETWORK_TUNER
    No node found

  - Pin "Input0", direction: input
    Some supported interfaces:
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMemInputPin
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

  - Pin "MPEG2 Transport", direction: output
    Some supported interfaces:
      interface IAMBufferNegotiation
      interface IAMStreamConfig
      interface IAMStreamControl
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMediaSeeking
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

device "TBS 5925 DVBS2 Tuner"
  Some supported interfaces:
    interface IAMDeviceRemoval
    interface IAMFilterMiscFlags
    interface IAMovieSetup
    interface IBaseFilter
    interface IKsControl
    interface IKsPropertySet
    interface IKsTopologyInfo
    interface IMediaFilter
    interface IMediaSeeking
  IKsPropertySet properties support:
  IKsControl properties support:
  IKsTopologyInfo:
    Categories: KSCATEGORY_BDA_NETWORK_TUNER
    No node found

  - Pin "Input0", direction: input
    Some supported interfaces:
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMemInputPin
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

  - Pin "MPEG2 Transport", direction: output
    Some supported interfaces:
      interface IAMBufferNegotiation
      interface IAMStreamConfig
      interface IAMStreamControl
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMediaSeeking
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:
```

lelegard, 2017-09-02:
Hello,

Thanks for providing `tslsdvb -e` output, this is exactly the right thing to do.

According to this output, this seems to be a Windows issue. Several DirectShow components are missing. The device categories `KSCATEGORY_BDA_NETWORK_PROVIDER`, `KSCATEGORY_BDA_TRANSPORT_INFORMATION` and `KSCATEGORY_CAPTURE` are empty. This should not be the case in a fully installed Windows system. For comparison, the end of this message contains the same output for a Windows 10 system without tuner. You can see that the network provider section contains support for ATSC, DVB-C, DVB-S and DVB-T.

In your case, the tuner is seen but Windows has no network provider filter to connect to it and TSDuck is unable to build a complete DirectShow graph.

Honestly, I do not know how you got this. As far as I know, a default Windows installation has always had a full featured DirectShow infrastructure. Did you specifically tuned the installation, or removed component, or unregistered DirectShow services?

I do not really know how to fix this. This is a Windows issue, really. I may propose to "reinstall" but it can more harmfull than helpfull on this operating system.

Which Windows version are you using? I know for sure that DVB tuners work on Windows 7 and 10. Before 7, DVB-S2 was not supported, only DVB-S.

As a final note, when you get it to work, if you get a corrupted reception with "dropped packets" messages using verbose mode, the lastest fixes in TSDuck may help. These fixes are committed in the code but no binary installer is provided yet. The fix tries to mitigate the defects of several tuner Windows drivers which tend to loose packet synchronization in the presence of bad reception. The stream of packets is then transmitted out of sync. TSDuck now tries to detect this and manually resynchronizes the stream. This has nothing to do with your problems but I wanted to let you know.

Let me know if you manage to fix your Windows installation.

```
=== Device category KSCATEGORY_BDA_NETWORK_PROVIDER

device "Microsoft ATSC Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
    interface IBDA_TIF_REGISTRATION
    interface IBaseFilter
    interface IFrequencyMap
    interface IMPEG2_TIF_CONTROL
    interface IMediaFilter
    interface IScanningTuner
    interface ITuner

  - Pin "Antenna Out", direction: output
    Some supported interfaces:
      interface IPin
      interface IQualityControl

device "Microsoft DVBC Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
    interface IBaseFilter
    interface IMediaFilter
    interface IScanningTuner
    interface ITuner

  - Pin "Antenna Out", direction: output
    Some supported interfaces:
      interface IPin
      interface IQualityControl

device "Microsoft DVBS Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
    interface IBaseFilter
    interface IMediaFilter
    interface IScanningTuner
    interface ITuner

  - Pin "Antenna Out", direction: output
    Some supported interfaces:
      interface IPin
      interface IQualityControl

device "Microsoft DVBT Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
[... 151 lines cut ...]
      interface IKsPropertySet
      interface IMediaSeeking
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

  - Pin "Video Camera Terminal", direction: input
    Some supported interfaces:
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMemInputPin
      interface IPin
      interface IQualityControl
      interface IStreamBuilder
    IKsPropertySet properties support:
    IKsControl properties support:

=== Device category KSCATEGORY_TVTUNER

=== Device category KSCATEGORY_BDA_RECEIVER_COMPONENT

=== Device category KSCATEGORY_BDA_NETWORK_TUNER
```

rezna11, 2017-09-02:
Thanks for checking. I can confirm this is Windows issue. On Win 10 it works, but does not on Win 2008R2. The other programs work probably because they use StreamReaderBDA interface, not directshow. I think it would be useful if you added StreamreaderBDA too as it's quite popular. But for now I will use Win 10 for further testing.

lelegard, 2017-09-02:
I am glad you made it work.

I am curious about StreamReaderBDA. I never heard of it and cannot find positive information using Google. Is this a Windows internal API like DirectShow or is this a third-party product?

Originally, ten years ago (!), I interfaced DVB tuners on Linux. The OS interface to DVB devices is LinuxTV, a very simple and reliable ioctl-based interface. A few years later, for consistency, I ported the application to Windows and looked for the Windows equivalent of LinuxTV. Hell ! I discovered that only DirectShow could interface with DVB devices. I found no other way to access BDA drivers. If you compare the amount of code to interface tuners in the source code of TSDuck between Linux and Windows, you will realise what a pain Windows is here. I even had to write my own DirectShow filter since their is no standard way to extract a stream out of DirectShow. Everything takes place _inside_ DirectShow. COM programming, a awful piece of sh.. And the experience demonstrated that the reception on Windows is much less reliable.

rezna11, 2017-09-02:
it's correct name is StreamReaderEx-BDA, developed by CrazyCat (the author of blindscan tool CrazyScan), more info here https://sourceforge.net/p/crazyscan/blog/

it's a pretty reliable way to work with DVB tuners in Windows, it's not universal API, not all tuners are supported but the best ones are

lelegard, 2017-09-02:
Thanks for the information. I will have a look at StreamReaderEx-BDA source code. If it uses a simpler way to use BDA drivers without DirectShow, I will consider using a similar interface. However, if it uses specificities of a few known drivers but cannot use all tuners in a generic way, then this would degrade the features of TSDuck which uses DirectShow to hide those specificities.

Win 2008 is a server product, unlike Win7 or Win10. So, maybe DirectShow is not installed by default. I have seen that there is a "Media Services" package for 2008 SR2. Maybe it is there. Have a look at:

https://support.microsoft.com/en-us/help/963697/how-to-install-windows-media-services-for-windows-server-2008-r2

Let me know if and how you managed to use DirectShow (and TSDuck) on 2008SR2. That would be a good information to add in the documentation.