tsduck/tsduck #2: Dektec DTA-107 low power in Win 2008 R2 SP1

rezna11, 2017-09-02:
I'm getting very low power (-47dBm) output from DTA-107 modulator. Had the same issue with Dektec's Streamxpress but after installing Dektec SDK, I'm getting full power in Streamxpress but still very weak from TSDuck. Haven't tested another OS, just W2008R2SP1.

rezna11, 2017-09-02:
The previous issue was fixed by installing Dektec SDK and rebooting PC again. But now I'm getting 100% power on the output and --level parameter has no effect

rezna11, 2017-09-02:
Looks like the --modulation nor --level parameters work for DTA-107... tsp uses the last values that were previously set by StreamXpress!

lelegard, 2017-09-02:
Looking at the source code, the two options are actually passed to DTAPI, the Dektec API that drives the device. However, I have not used any DTA-107 for years and I do not have any now. I will ask some colleagues if they have any to test.

rezna11, 2017-09-03:
After trying in Windows 10 and Windows 2016, it really seems to me that the tsp command line parameters are not set correctly, not only the level and modulation but perhaps other too. FEC and SR seem to work, at least in some cases.

lelegard, 2017-09-03:
Could you provide a sample command with all parameters? How do you determine that they are not set correctly? Could you post the values you measure if you have a spectrum analyser or equivalent?
Thanks

rezna11, 2017-09-03:
Sorry, I'm an idiot, I entered "-30" (habit from working with DtPlay) but it should have been "-300", no wonder the modulator was acting strange. During my tests I also didn't encounter the issue with modulation settings, I think it was all caused by wrong level I was entering.

lelegard, 2017-09-03:
No problem, we all make mistakes sometimes.
Closing this one.

closed 2017-09-03 by lelegard