tsduck/tsduck #3: Remote IP source?

rezna11, 2017-09-02:
So I have a second computer with DVB tuner in another city. Over there I'm running EBSPro which can stream TCP or UDP. How do I use this as IP source for tsp at my home? When I try` tsp -I ip REMOTE_PUBLIC_IP:7000`, tsp outputs an error complaining that this is not a multicast IP. However, I don't want to use multicast in this case, I just want to connect to remote IP and get the TS via TCP or UDP. I'm running Windows on both ends. Thanks for ideas.

lelegard, 2017-09-02:
Hello,

Thanks for using TSDuck. With all due respect, this cannot work that way because this is not the way IP networking is working.

Your source, let's name it SERVER, sends UDP/IP packets which must be received at your destination, let's name it HOME, which uses tsp. There are two ways to do this.

First option, use multicast. SERVER sends IP packets with a _destination_ multicast address. These packets can be received by multiple systems, including HOME. All these systems must explicitly listen to this multicast address. This is the address parameter of tsp. Note that this address is the _destination_ multicast address, not the source address of SERVER. There is no way in IP systems to listen to packets based on source address.

Output of `tsp -I ip --help`:
```
Usage: tsp -I ip [options] [address:]port

Parameter:
  The parameter [address:]port describes the destination of UDP packets.
  The 'port' part is mandatory and specifies the UDP port to listen on.
  The 'address' part is optional. It specifies an IP multicast address
  to listen on. It can be also a host name that translates to a multicast
  address.
```

However, for this to work, the network infrastructure must allow those multicast packets to fly from SERVER to HOME. This works inside a home or business internal network. But, in the general case, this does not work on the Internet. Otherwise, if anyone is allowed to send multicast packets all over the planet, the network would explode. Inside the managed network of an ISP, multicast is usually allowed and routed only for the traffic which is generated by the ISP. This is the way the ISP provides IP-TV for instance.

In your case, if SERVER and HOME are in different places and communicate through the Internet, there is probably no way to route multicast between the two.

Second option, use unicast. In that case, SERVER shall send the packets to the IP address of HOME, explicitly. At HOME, tsp does not need any address because the unicast packets are sent to it anyway. Only the port number is required and must be the same as the destination port as specified by SERVER.

If HOME has several network interfaces, you may optionally force which local interface to use. But this is strictly optional.

```
  -l address
  --local-address address
      Specify the IP address of the local interface on which to listen.
      It can be also a host name that translates to a local address.
      By default, listen on all local interfaces.
```

If HOME has a public IP address and is normally routed to the Internet, just specify its address as destination.

Now, if HOME is connected to the Internet through a NAT router, as it is the case with most ISP's, you must open the specified port on the router and create a redirection to HOME inside your home network. Then, SERVER must use the public external address of your NAT router.

Sorry for the long explanation if you already knew all of this.

rezna11, 2017-09-02:
Thanks for detailed answer. I found out about TSDuck yesterday and I loved it from the first moment ... at least the idea, because I still can't get it to work :) But I feel it could be great alternative to Dektec's overpriced Muxxpert, which I almost bought since I would like to use Dektec devices for a fun project, so money is always an issue.

Anyway, I think I didn't understand how tsp works with ip sources. I thought it could read TCP/UDP source just like it reads files. You know, when I open VLC and enter the IP and port of the SERVER (TCP), I can watch the channels - so VLC reads the stream from the SERVER. And I thought tsp could do this too, but apparently not.

So what I can do, I think, is to run tsp on the SERVER, filter out unwanted services and use ip output module to generate uni/multicast directed to HOME...? That would be good. My TBS6903 tuner on the server is recognized by tsp. Multicast could maybe work too through EoIP tunnel which I could set up (I have Mikrotiks on both ends).

lelegard, 2017-09-02:
`tsp -I ip` works exactly the same way as VLC. When you receive a multicast stream on VLC, you specify the multicast address, which is a _destination_ address. You cannot receive using the _source_ address of the server host. It is not a matter of application, tsp or vlc, this is simply not possible on IP networks.

I am not convinced that tsp can replace Dektec Muxxpert. Tsp is not a MUX. Muxing is very difficult, hence the price of real mixing software. That being said, Dektec products, both hardware and software, are significantly cheaper than other professional products.

TSDuck is not a production muxing or streaming tool. It is a flexible toolkit to analyse and/or transform MPEG streams in an infinite number of ways. A lab and debug tool. However, because of this flexibility, some of my former colleagues used if for things I really did not anticipate. So, you may found some original usages too. Have fun...

rezna11, 2017-09-02:
The difference probably is that the stream from EBSPro which I used is unicast and TCP, so I can receive it anywhere by simply connecting to the server IP and port. I admit I don't know much about UDP and multicast. 

But I'm wasn't sure if tsp really couldn't handle a simple TCP connection to read the source from. Basically something more elegant than the following solution:

`wget -O - http://my_public_up:7001 | tsp -I file -O dektec --device 0 --frequency .....`

Something where I could drop the wget part.

closed 2017-09-03 by lelegard