tsduck/tsduck #4: Variable source TS issue with speed/stuttering, can tsp add null packets?

rezna11, 2017-09-02:
I'm struggling with this and I'm not sure if I can do this at all, so I'd like to ask:

- I have live IP source with filtered TS (so it's variable bit rate) or I could even take the full transponder
- I want to output this to Dektec DTA-107, for now without further processing, but I want set different modulation parameters than used on the source (for example to convert DVB-S to DVB-S2)
- As an parameter for the output, I set symbol rate to 30000000 for example, so it's more than the original stream
- Then I tune into this mux on my PC with TBS tuner to analyze the TS - and in the TS, I can see tsp adds null packets, so it looks OK
- BUT when I try to play this on my TV, i can see stuttering and errors 

Any idea what's missing?

lelegard, 2017-09-02:
Tsp does not add packet. It can only transform or remove input packets. If you have a variable bitrate input and need to output at a fixed bitrate like DVB-S/S2, tsp is not really suitable for this. If you see null packets on output, this is probably the DTA-107 which is smart enough to add them when its input is starving.

But the output is certainly not smooth. The irregular input probably creates bursts in the output. This is the kind of behaviour a TV or STB does not like. This kind of device expects a smooth real-time stream. PC applications may be more tolerant.

rezna11, 2017-09-02:
Alright, so are you saying that tsp can be used to remove packets but it won't reduce the TS bitrate since the removed packets must be replaced with null packets in order to keep the constant bit rate of the TS? If that's the case, is another plugin needed for converting vbr mux to cbr? Something like tsvbr2cbr from Opencaster?

lelegard, 2017-09-02:
No, removed packets are removed, hence reducing the bitrate. 

But in your case you want to output a higher bitrate that the input. This would need to create new packets. By the way, I forgot about the option `--add-input-stuffing` with tsp. Look at the user's manual. This is a way to artificially increase the input bitrate.

However, you still have the issue of the variable input bitrate. The option `--add-input-stuffing` will increase the bitrate but it will remain variable. But we may use it to increase the bitrate so that the lowest bitrate values are at least as high as the required output bitrate. Then, we would need a plugin that clips at a given bitrate, dropping null packets only. That could work. Just another plugin to develop. Are you volunteer?

rezna11, 2017-09-02:
Right, I need to output higher bitrate because I don't know the max bitrate of the input but I know it's less than the original TS since it was filtered before reaching tsp. However, TSDuck is already working out for me just fine because I was able to convert source VBR to CBR using tsvbr2cbr tool with just a minor modification in the code of tscbr2vbr (read from stdin instead of looking for a file).

With this command in cygwin, I'm able to watch smooth video on a receiver:

`wget -O - http://my_remote_ip:7001 | tsp -I file -O file | ./tsvbr2cbr.exe - 24191000 2> /dev/null | tsp -I file -O dektec --device 0 --frequency 1055000000 -r 7/8 --modulation DVB-S-QPSK --symbol-rate 15000000`

Of course it would be nicer to have a plugin that will do this without having to use such workaround but this proves how good and flexible TSDuck already is! I'm sorry but I'm in no way volunteering for any development as I'm not a real developer. I'd love to but I don't have the skills and time.

closed 2017-09-03 by lelegard