tsduck/tsduck #5: regulate plugin no output

EQMOD, 2017-09-04:
HI i tried to use tsp but when using regulate plugin , there is no output. Input is from a file and output is VLC via UDP

I am using 12.04 Ubuntu on virtualbox.

lelegard, 2017-09-04:
Hi,

Reading a TS file and multicasting it over UDP is the typical use case for the regulate plugin. It was developed for that purpose. This works as far as I used it. See the user's manual section 5.2.4 for an example.

Could you be more specific? What do you mean by "no output"? Could you investigate further? Run Wireshark on the same system as tsp and check the IP and UDP headers in the outgoing IP packets. If you do not see any IP packet here, retry without output plugin and pipe the output to tsdump to see if the tsp command generates any TS packet (it is easy to remove all packets with an incorrect filter). 

If you want others to help you, please provide precise technical details, command with parameters, results of measurements, observations or investigations. "Does not work" is not enough to provide a diagnosis.

rezna11, 2017-09-04:
works fine for me... make sure the TS you try to play contains PAT (PID 0), otherwise VLC won't play

I tested this command:

`tsp -I file testfile.ts -P regulate -b 3590409 -O ip 192.168.20.11:1234
`
and in VLC opened URL `udp://@192.168.20.11:1234` and it played smoothly

EQMOD, 2017-09-05:
I did what is exactly written in the manual ,  tsp –I file   <myfile.ts>    -P pcrbitrate     -P regulate   -O ip <myIP:port>

If i remove the regulate plugin - i see some video frames and VLC gets choked. I also see output when it is in stdout output mode but once i put the regulate plugin - tsp program sends out at stdout a single packet per second.

lelegard, 2017-09-05:
The plugin `regulate` "regulates" the pass-through speed at a given bitrate. If you want to see some consistent video, this bitrate shall be the right one for that video. Otherwise, you will get poor result.

The first solution is to specify a bitrate yourself using option `-b` as in @rezna11 comment. But you must specify the right value for that specific file. So, using the same value as another example is unlikely the right one for your input file and you won't get the right result. If your video has a constant bitrate, use the command `tsanalyze` to get a complete report on the input file, including bitrates, per TS, per service and per PID.

The second solution is to have the bitrate transmitted from a previous plugin in the chain. Normally, the input plugin computes the input bitrate if possible. For demodulators, the bitrate is computed from modulation parameters. For files, the bitrate is computed based on PCR analysis. Then, each plugin passes the bitrate information to the next one. By default, the plugin `regulate` uses this information as target pass-through speed. If the TS has not been modified between the input and `regulate`, there is nothing to do since `regulate` receives the input bitrate. If you have removed PID's in between, like `zap` to a given service, the input bitrate information is no longer valid. The plugin `pcrbitrate` is here to recompute the actual bitrate, based on PCR's, at this point and propagate the new bitrate information downstream to the next plugin.

However, for this to work, the PCR's must be present and correct. If your input file has missing or incorrect PCR's, the bitrate information will be incorrect. Try option `tsp -v` (verbose mode). The plugins `pcrbitrate` and `regulate` will display the bitrate information. If these values seem incorrect, PCR's are probably incorrect. Try `regulate --dts` to use DTS instead of PCR as synchronization base. In all cases, do a `tsanalyze` on your input file to get an idea on bitrates.

Last possibility, this could be a system issue. The plugin `regulate` uses the real-time clock of the operating system to regulate the output speed. Since you are using a VirtualBox virtual machine, there is _maybe_ a risk that the real-time clock of the guest VM is incorrect. Try the same command on a physical system (either Linux or Windows). But, this seems the less probable cause.

lelegard, 2017-09-16:
Any news on this? Do you still have the problem?

Note: In my previous comment, I wrote `regulate --dts`. This is incorrect. It should have been `pcrbitrate --dts`. But reading the user's manual would have given you the answer already.

lelegard, 2017-09-20:
Closing this issue because of lack of additional information

EQMOD, 2017-09-28:
ok i tried to debug using Ubuntu 12.04. I inserted a debugging code to src/libtsduck/tsMonotonic.cpp 
 and got the following BIG NUMBER on _value thus causing a very looong wait time ;

**Inside Monotonic wait - nanonsecpersec: 1,000,000,000 _value:1,506,573,853,577,471,806 due.tv_sec: 1,506,573,853  due.tv._nsec:577,471,806**

Program was executed with -P regulate -b 30000000 using the following cout added code;

 // Compute due time
    ::timespec due;
    due.tv_sec = time_t(_value / NanoSecPerSec);
    due.tv_nsec = long(_value % NanoSecPerSec);

  std::cout <<  "Inside Monotonic wait - nanonsecpersec: " + Decimal(NanoSecPerSec)+ " _value:" + Decimal(_value) + " due.tv_sec: " + Decimal (due.tv_sec)+ "  due.tv._nsec:" + Decimal (due.tv_nsec) + "\n";

    // Loop on clock_nanosleep, ignoring signals
    int status;
    while ((status = ::clock_nanosleep(CLOCK_MONOTONIC, TIMER_ABSTIME, &due, NULL)) != 0) {

        if (status != EINTR) {
            // Actual error, not interrupted by a signal
            throw MonotonicError("clock_nanosleep error", errno);
        }
    }

lelegard, 2017-09-28:
Hi,

Thanks for taking the time to investigate. Note that the call to `clock_nanosleep` uses the flag `TIMER_ABSTIME` which means an absolute time value, not a duration value. So, this huge value you see is the number of seconds since 1979-01-01, not the duration you are waiting for.

Your value means:
```
$ TZ=UTC date --date=@1506573853 '+%Y-%m-%d %H:%M:%S'
2017-09-28 04:44:13
```

This seems quite consistent with the date of your report.

To explain the context, the plugin `regulate` needs a monotonic clock to avoid the drift which would be caused by the general overhead of the system and application. Using an absolute time is a requirement to get a monotonic clock. Any use of duration value would lead to time drift.

In you case, if you want to display the waiting time, you should display the difference between this value and `::clock_gettime(CLOCK_REALTIME, &res)` as done in `ts::Time::UnixRealTimeClockNanoSeconds()`.

EQMOD, 2017-09-28:
Yes i can confirm that the time values (current time and the passed clock_nanosleep time) are valid. It seems the clock_nanosleep() function blocks at  initial call and does not return to your wait() routine. Will investigate further ...

lelegard, 2017-09-28:
I just reproduced the issue on Ubuntu 16.04 running as a guest O/S in a VirtualBox VM on a Windows 7 host. So, in my list of potential causes, it seems that the VM issue was the right one. Using Google, I have found a couple of problems with `clock_nanosleep` on virtual machines. When you need real-time precision, VM's are not so good. You have to use a physical machine.

Anyway, since many people use VM's, I just pushed a fix. The default behaviour remains the use of `clock_nanosleep` but I added the option `--no-realtime-clock` to `tsp` to disable its usage and fallback to a more standard clock. This was the default behaviour on macOS since macOS does not support `clock_nanosleep`. Now, you can manually activate it on Linux.

This is a global `tsp` option, to be placed before the first plugin.

Could you please pull to current state of the repository and check if it works on your VM?

EQMOD, 2017-09-28:
i made the git clone/pull and compiled/installed the binaries and added the --no-realtime-clock option and it is now working :) Thanks Thanks :)

lelegard, 2017-09-28:
I am glad to read that. For your information, I just tried on two other VM's, Fedora 26 and Ubuntu 17, but using VMware on macOS Sierra. Same result. So, VMware does not seem better that VirtualBox.

closed 2017-09-29 by lelegard