tsduck/tsduck #6: descrambler - PID selection?

rezna11, 2017-09-04:
Alright so let's say I have a basic DVB scrambler which scrambles all PIDs, including PMT, Video, Audio. I'm doing the scrambling, so I know the control word - it's constant, not changing. A regular receiver can't decrypt PMTs, so I would like to use TSDuck to decrypt PMT but pass everything else encrypted to the output and let the receiver handle the rest (decryption of all A/V PIDs). This is just an example, I could also want to decrypt other PIDs. But right now I can't see this as possible, unless I'm not getting something.

lelegard, 2017-09-04:
Currently, the `descrambler` plugin is very basic (unlike the `scrambler` one). It descrambles all scrambled packets, typically for tests.

I keep the suggestion to specify a list of PID's to descramble, leaving alone the others. I will add it when I have time.

I keep this issue open as a reminder.

Thanks for the suggestion.

rezna11, 2017-09-04:
cool, thanks

lelegard, 2017-09-05:
Done
```
$ tsp -P descrambler --help
...
  -p value
  --pid value
      Descramble packets with this PID value. Several -p or --pid options may be
      specified. By default, all PID's with scrambled packets are descrambled.
```

closed 2017-09-05 by lelegard

rezna11, 2017-09-05:
works great, thanks!

commit b88eb28a (2017-09-05): Added option --pid to plugin descrambler

Fixes #6