tsduck/tsduck #7: Somewhat inconsistent output from tslsdvb and tsdektec -a

rezna11, 2017-09-04:
There might be a reason for this but...
If I do `tslsdvb `and I have only one DVB tuner installed, I get

`TBS 5925 DVBS2 Tuner (DVB-S)`

But if I install 2 tuners, I receive

```
0: TBS 5922 USB DVB-S2 Tuner (DVB-S)
1: TBS 5925 DVBS2 Tuner (DVB-S)
```

When doing tsdektec -a with just one adapter installed, I get 

`0: DTA-107 (DTA-107 QPSK Modulator/Upconverter)`

**First issue** is that I don't understand why the first command doesn't return "0: " in front of the device name. Almost seems like a bug to me. If that's on purpose, then why doesn't `tsdektec -a` behave the same way, why does it include 0: prefix?

**Second issue**, but not a bug for sure, is the fact that the output of this command can't be used as `--device-name` directly because the index prefix and the " (DVB-S)" part must first be stripped, which is not very intuitive.

lelegard, 2017-09-05:
Good point. I have no excuse, just explanations to offer :)

At the very beginning, the ancestor of TSDuck was developed on Linux only. Even after porting it to Windows, DVB tuners were supported on Linux only. On Linux, there is a consistent framework for DVB tuners. All tuner devices are named `/dev/dvb/adapter0`, `/dev/dvb/adapter1`, etc, regardless of the brand. So adapter names and numbers were simple. 

Then came Windows. It took me some time to tame DirectShow. On Windows, tuners do not have device names. The only distinctive name was the complex name of the vendor-provided DirectShow filter for the tuner, usually with mixed-case and embedded space. Not very handy. If there is only one tuner in the system, it is used by default and we do not really care about names. When there are several tuners, it is easier to use the adapter number with options `-a 1` or `--adapter 1`. This is why, on Windows, numbers are displayed only when there is more than one tuner.

That being said, I admit that this is not very consistent. I will modify output to something like this, even if there is only one tuner.
```
0: "TBS 5922 USB DVB-S2 Tuner" (DVB-S)
1: "TBS 5925 DVBS2 Tuner" (DVB-S)
```
Adding quotes seems better to clearly indicate where to copy/paste if you really want to use adapter names in TSDuck commands. 

Nonetheless, if you have two identical tuners on the same Windows system, it is probable that they will get the same name and using adapter numbers will remain the only solution to address them specifically. We do not have that kind of issue with Linux.

I keep this issue open as a reminder.

rezna11, 2017-09-05:
Thanks, having the index in front of each device no matter what's the number of items in the list would be nice and make a semi-automated tsp control easier (like a GUI to let the user select a device and build tsp command line with all parameters).

lelegard, 2017-09-05:
For automation, several commands have a `--normalized` option which creates a stable and easily parsed output format. This includes `tsdektec`. I can add one for `tslsdvb`. My use case was scripting but UI is another possible use case.

lelegard, 2017-09-05:
Done. On my Windows system with one USB tuner:
```
C:\> tslsdvb
0: "PCTV 291e/461e BDA 28179 TVTuner S" (DVB-S)
```

closed 2017-09-05 by lelegard

rezna11, 2017-09-06:
also works good with multiple tuners

[image]

commit cd80f5f2 (2017-09-05): Adjusted tslsdvb output on Windows

Fixes #7