tsduck/tsduck #8: Show dektec modulator tuning parameters?

rezna11, 2017-09-06:
Is it possible to print the exact tuning parameters of a modulator, most importantly the symbol rate, if it's not entered in the command line? In the example below, the symbol rate gets smartly calculated to 29102, which is great, I didn't have to specify it manually. But then the problem is that only blindscan (or a DVB bandwidth calculator) can actually tell me what's the symbol rate. Is there a way to show the resulting symbol rate (and perhaps other parameters) after running tsp?

This is my command:

`tsp -v -I dvb --adapter 2 --delivery-system DVB-S2 --modulation QPSK --lnb 9750,10600,11700 --frequency 11495000000 --polarity vertical --fec-inner 9/10 --symbol-rate 44100000 --pilots on --roll-off auto -O dektec --device 1 --modulation DVB-S2-8PSK --convolutional-rate 8/9 --lnb 5150 --satellite-frequency 4095000000 --level -390 --stuffing`

and output I receive

```
* dvb: using TBS 5922 USB DVB-S2 Tuner (DVB-S)
* dvb: tuned to transponder --frequency 11495000000 --symbol-rate 44100000 --fec-inner 9/10 --spectral-inversion auto --polarity vertical --delivery-system DVB-S2 --modulation QPSK --pilots on --roll-off auto --lnb 9750,10600,11700 --satellite-number 0
* dvb: input bitrate is 77,004,969 b/s
* dektec: using Dektec device 1, output channel 0 (DTA-107 port 1)
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes
* dektec: setting output carrier frequency to 1,055,000,000 Hz
* dektec: initial output bitrate: 77,004,969 b/s
* dektec: DTA-107 output FIFO load is 6,710,848 bytes, starting transmission
```

and the screenshot shows blindscan result - as you can see, the symbol rate of 29102 is nowhere to be seen in the tsp output shown above

[image]

lelegard, 2017-09-07:
The plugin displays the parameters which are passed to the Dektec API, including the default values which were not specified on the command line. 

If I am not wrong, you are asking for the actual parameters which are applied on the output signal. Only the modulator knows them and I have not seen any way to obtain those informations from the Dektec API. Sorry.

rezna11, 2017-09-07:
Yes, what I was asking about was the symbol rate of the output signal from Dektec. Since TSDuck knows the bitrate and all the tuning parameters (except symbol rate in my case), it could be able to calculate the symbol rate which will be used by the modulator.

lelegard, 2018-02-07:
Hi @rezna11 

This was an old one I finally fixed. In the output plugin `dektec`,  when the output bitrate is specified but not the symbol rate, the computed output symbol rate is displayed on start. This is a verbose level message (use `tsp -v`).

Fixed in commit 2c8133fb3e2de2267d740d60ad552a7dbf023949, version 3.8-526.

closed 2018-02-07 by lelegard