tsduck/tsduck #9: Plugin idea: T2-MI extractor

rezna11, 2017-09-06:
There's a simple [command line tool](https://github.com/newspaperman/t2-mi) by @newspaperman to extract the TS from a T2-MI data stream. I think this would be also a cool plugin for TSDuck. I'm not sure how to do it myself since the plugins I see and I understand are processing a single packet, but with t2-mi it's necessary to process and output multiple packets at a time, so it's not THAT easy.

I could provide various T2-MI streams to anyone interested.

lelegard, 2017-09-06:
Good idea. I never had to opportunity to work with T2-PI and did not read the related ETSI documents about it. I had a brief look at the code you mentioned. It does not seem too complicated. 

TSP plugins process packets one at a time since it is the simplest way to do. But most plugins need a global view. For instance, demuxing a table (NIT, PMT, whatever) needs to read many packets to get a complete table and need many other packets to insert the modified table, if required. This is done with minimal effort.

So, developing a T2-MI extractor should not be difficult for whoever understands the concept of T2-MI. A good exercise... Providing  a stream with embedded M2-TI and a few explanations about it and about the expected output is definitely necessary. If you have this and can take some time to write explanations, please do.

rezna11, 2017-09-07:
I uploaded couple of files:

[T2MI_CAPITAL.zip](https://github.com/tsduck/tsduck/files/1284901/T2MI_CAPITAL.zip)

Extract the zip to get a TS. It's a fresh recording from 40W (3667L) of the full transponder, including null packets. In the TS, you'll find PIDs 0, 33, 64 and 8191.

[image]

PID 33 = PMT
PID 64 = T2MI encapsulated data

Checking the content in 4T2 Content Analyzer will reveal the content of the data PID 64, and it will show the PLP (102).

[image]

[image]

With PLP and the PID, you can run newspaperman's t2-mi extractor with a command like

`cat T2MI_CAPITAL.ts | ./t2-mi.exe 64 102 > capital_extracted.ts`

Then you'll get capital_extracted.ts which will be the final and playable TS. In this case, there's one regional TV channel from Bogotá in HD. You should be able to run t2-mi extractor without PLP, just data PID and it finds the PLP by itself.

[image]

newspaperman, 2017-09-21:
Hi!

thanks for keeping me up to date with this new great tool.
I didn't find time to answer your last mails. Did you make any progress with the ISE?
I've heard AFN will switch completely to HD this year and all GIs will get new decoders, I guees this will be the end of the game.

kind regards

Newsy
--------------------------------------------

 

 
 
 
 
 
 
 You are
 receiving this because you were mentioned.
 it on GitHub, or mute
 the thread.

lelegard, 2017-09-22:
@newspaperman The best way to be kept up to date is to "watch" the repository, as usual with GitHub.

Concerning the M2-TI extractor, I have no time and no testing environment to do it. Maybe you could do it since you already wrote an extraction tool. I can assist you with the TSDuck integration. Let's discuss this offline it you are interested.

lelegard, 2017-10-27:
Hi @rezna11,

Considering the obvious lack of volunteering ;), I started to look at DVB T2-MI standard documents. As a result, I implemented a candidate version for a T2-MI extractor plugin. The plugin is simply named `t2mi`. By default, without parameter, it detects the first PID carrying T2-MI in the TS and the first PLP inside that stream. The output of the plugin is the extracted TS from T2-MI. It completely replaces the input TS.

From the sample stream which was uploaded, the extraction command is:
```
tsp -I file T2MI_CAPITAL.ts -P t2mi -O file CAPITAL.ts
```
The output stream is consistent and the video of the matches the posted image.

I did that "the TSDuck way", meaning the plugin itself is mainly a simple command wrapper. The logic is located inside a C++ class named T2MIDemux which can be used anywhere, including in external applications. The demux class is also used in `tsanalyzer` so that the TS analysis now includes T2-MI information such as PID's and PLP's.

**However**, (there is always a _however_ somewhere), the extraction code may not work on all cases of T2-MI encapsulation. While demuxing T2-MI packets is relatively easy (see ETSI TS 102 773), demuxing TS packets from them is a real mess (see ETSI EN 302 755). There are many modes and adaptation options as in most complex transmission systems. The `t2-mi` program from @newspaperman is rather obscure to read but it seems obvious that it does not handle all modes either (there are also a few questionable issues that I cannot challenge for a lack a reference streams). All I can say for now is that the `t2mi` plugin can extract at least the same type of T2-MI streams as the `t2-mi` command line program. However, I anticipate that some T2-MI streams which would be inserted with different options may fail.

Could you please test the `t2mi` plugin on as many streams as you can? Please remember that I have no T2-MI reference stream. I only tested the `T2MI_CAPITAL.ts` stream that you provided. And then, please provide a feedback on how it worked. If you have any issue with a T2-MI stream, please post a sample stream reproducing the issue.

Currently, I have not yet produced an official version. To test the `t2mi` plugin, you need to pull the git repository and build the installer yourself. If you do not have the right environment to do so, please indicate which platform you use so that I can build an installer.

rezna11, 2017-10-29:
Great, I'll try but I will first need to fix my C-band dish since most T2MI signals are in C band... and today is quite windy, so hopefully during the week.

rezna11, 2017-11-05:
So I finally got to testing the t2-mi extraction and the 3 providers I tested all worked. I tried various RTVC feeds at 40W, a russian mux at 40E and the RCN mux at 40W. I'm not aware of other T2-MI muxes I could try. There are more russian ones at 40E and 53E but it's the same provider so I suppose they will all work. RTVC muxes worked automatically without any extra parameters. For RTVC and the russians, I had to enter --pid and --extract parameters and then it worked too.

lelegard, 2017-11-06:
Hi,

Thank you for the effort and the feedback.

Could you upload a sample of a TS for which you had to specify --pid? I assume that there is no T2MI_descriptor in the PMT of the service, meaning that there is no public evidence of the presence of the T2-MI stream.

Normally, --extract should not be necessary, this is the default, once a PID is identified, either automatically through the presence of a T2MI_descriptor or manually through option --pid.

rezna11, 2017-11-06:
Here are those 2 streams:

https://mega.nz/#!5wpQgJ7I!n7krkMmYt6sqfhHFyUq7g_bXN1f-jwRpwoTZZsmIrd4
https://mega.nz/#!tkwESb7K!4f38IwTbXt5grjsikAfYXsYfi71gFtvp4e4LkGzxSOM

both work with --pid, you're right that --extract wasn't necessary

lelegard, 2017-11-06:
Thanks for the files.

Indeed, there is no T2MI_descriptor in the PMT's. So there is no way to determine that the services are carrying T2-MI encapsulation. I am not a specialist in T2-MI but it seems that the signalization is incorrect according to DVB. Anyway, I have seen worse...

That means that the receiving equipment must use hard-coded service ids or, much worse, hard-coded PID's.

Two strange things:

- The Columbian T2-MI service has ECM's, while being in the clear. Maybe the encapsulated stream is encrypted at some periods of the day. Although the usage, in that case, is to stop ECM broadcasting during clear periods.
- The decapsulated Russian stream contains several services. In one of them, there is an old French movie. In the audio track, we can hear the original dialog in French, *over-dubbed* with translated Russian dialogs after a small shift. I wonder how people can accept to watch that crap.

OK, I think we can close this issue for now.

rezna11, 2017-11-06:
That wouldn't be surprising if their equipment is configured to handle the streams without T2MI_descriptor. These muxes aren't supposed to be received on basic consumer-grade receivers. 
I don't find the other points strange at all. The ECMs are there because the T2MI PID was originally encrypted (I didn't mention it as I didn't think that was necessary). And the Russian dubbing was considered normal in the past. All TV shows and movies were dubbed like this and it's still the case in Poland, even on premium pay-tv channels. On Russian premium channels they do a proper voice acting now I think.

lelegard, 2017-11-06:
OK, thanks for the feedback.

On French TV, we still have a horrible dubbing. But at least it is simple dubbing, completely replacing the original track. With Digital TV, premium channels added the original audio track and subtitles tracks. So, you could choose between French dubbing and original audio (with or without subtitles). 20 years after launching DTV, we still have some TV channels with French dubbing *only*. This probably explains why French people are so terrible at speaking English while Scandinavian people are much better, probably because they do not have dubbing at all and listen to English audio tracks from the beginning.

I have no experience with Eastern Europe TV. So, it was new to me to hear a *mixture* of languages in one single audio track. 

I close this issue. If you ever encounter a T2-MI stream with different encapsulation options for which tsp fails to demux, please enter a new issue with a sample TS.

closed 2017-11-06 by lelegard

Jenny256765, 2018-01-09:
Hi Rezna11,

Thanks for the T2-MI streams. ^_^
I had donwloaded T2MI_CAPITAL.ts, 3635R.ts and 4009R.ts
I can extract T2MI_CAPITAL.ts and 3635R.ts to a TS file.
But I can't extract 4009R.ts to a TS file.

[image]

Try to search T2MI packet_header start, but can't find any useful Baseband frame type 0x00.
Do you have any suggestion ?

rezna11, 2018-01-10:
Sorry I don't have any suggestions, but please don't consider 4009R a representative sample. It might work with newspaperman's original tool but TSDuck is maybe more strict and won't process the stream if there are errors in it.

lelegard, 2018-01-10:
Hello @Jenny256765 ,

Which command did you try? Did you expect the `t2mi` plugin to find the T2-MI stream automatically?

The stream 4009R has an incorrect signalization in the sense that the T2-MI PID is not properly signaled in the PMT of its service. This means that there is no way to know that there is a T2-MI PID here. Normally, the PMT of a service carrying T2-MI shall have a T2MI_descriptor in the T2-MI component. This is the way the `t2mi` plugin automatically recognizes the stream to extract.

Now, doing a `tsanalyze` on the file, you can see that PID 0x1000 seems a good candidate for a "hidden" T2-MI stream. So, let's specify it:
~~~
tsp -I file 4009R.ts -P t2mi --pid 0x1000 -O file 4009R.t2mi.ts
~~~
I get a valid extracted stream with clear video and audio. It seems to be a Spanish-speaking TV show talking about some election (I don't understand Spanish).

Jenny256765, 2018-01-11:
Hello @lelegard and @rezna11,
Thanks for the reply.  ^_^

I used my own little tool t2mi2ts to extract 4009R.ts and the result is fail. 
So today I try to use https://github.com/newspaperman/t2-mi
cat 4009R.ts | t2-mi 4096 0 > 4009R_extracted.ts
And get a valid TS stream.

I have know what is wrong with my tool. I should skip the invalid packet type and look for next PUSI=1 to collect T2MI packet.

Now my tool can extract 4009R.ts to valid TS stream. 
Thank you!!

robiul0011, 2024-06-26:
https://www.lyngsat.com/muxes/Thaicom-6_SE-Asia-C_4052-H.html
[image]
my need to help how to open this channel

robiul0011, 2024-06-26:
https://www.mediafire.com/file/2xa6y13mze3r0x0/20240627+0040+-+4050H+SID+0x01+-+instant+record.rar/file

TS File