tsduck/tsduck #10: Support for ISDBT

EQMOD, 2017-10-10:
In one of the old DtPlay source files, there is a code there that supports ISDBT modulator initialization. This is done by scanning the incoming TS data for NIT information or by directly initializing the IsdbtPars structure (see below) using data from a config file or passed as a plugin parameter - Would it be possible to include such on the next releases of tsduck ? :)

void ts::InitIsdbtPars(DtIsdbtPars& IsdbtPars)
{
    DTAPI_RESULT dr;
    FILE *config_file_fp;
    // const int c_PckSize=204;
    // const int c_TempBufSize = 1*1024*1024;

    // char* pTempBuf = new char [c_TempBufSize];

    int pid;
    char layer;

    IsdbtPars.m_DoMux = true;
    IsdbtPars.m_FilledOut = true;
    IsdbtPars.m_BType = DTAPI_ISDBT_BTYPE_TV;
    IsdbtPars.m_Mode = 3;
    IsdbtPars.m_Guard = DTAPI_ISDBT_GUARD_1_16;
    IsdbtPars.m_PartialRx = 1;
    IsdbtPars.m_Emergency = 0;
    IsdbtPars.m_IipPid = 0x1FF0;
    IsdbtPars.m_LayerOther = DTAPI_ISDBT_LAYER_B;

    IsdbtPars.m_LayerPars[0].m_NumSegments = 1;
    IsdbtPars.m_LayerPars[0].m_Modulation = DTAPI_ISDBT_MOD_QAM64;
    IsdbtPars.m_LayerPars[0].m_CodeRate = DTAPI_ISDBT_RATE_2_3;
    IsdbtPars.m_LayerPars[0].m_TimeInterleave = 3;

    IsdbtPars.m_LayerPars[1].m_NumSegments = 12;
    IsdbtPars.m_LayerPars[1].m_Modulation = DTAPI_ISDBT_MOD_QAM64;
    IsdbtPars.m_LayerPars[1].m_CodeRate = DTAPI_ISDBT_RATE_3_4;
    IsdbtPars.m_LayerPars[1].m_TimeInterleave = 2;

    IsdbtPars.m_LayerPars[2].m_NumSegments = 0;

    /* For TSMuxeR... */
    /* PAT */
    IsdbtPars.m_Pid2Layer[0x0]=DTAPI_ISDBT_LAYER_B;

    /* NIT */
    IsdbtPars.m_Pid2Layer[0x1f]=DTAPI_ISDBT_LAYER_B;

    /* PMT */
    IsdbtPars.m_Pid2Layer[0x100]=DTAPI_ISDBT_LAYER_B;

    /* VIDEO */
    IsdbtPars.m_Pid2Layer[0x1011]=DTAPI_ISDBT_LAYER_B;

    /* AUDIO */
    IsdbtPars.m_Pid2Layer[0x1100]=DTAPI_ISDBT_LAYER_B;

    /* PCR */
    IsdbtPars.m_Pid2Layer[0x1001]=DTAPI_ISDBT_LAYER_B;
    int result;
    dr = IsdbtPars.CheckValidity(result);
    if ( dr != DTAPI_OK ){
        switch(result){
        case DTAPI_ISDBT_E_NSEGM:
        printf("DTAPI_ISDBT_E_NSEGM\n");
        break;

            case DTAPI_ISDBT_E_PARTIAL:
        printf("DTAPI_ISDBT_E_PARTIAL\n");
        break;
        default:
        printf("result = %d, dr = %d\n", result, dr);
        }

        fprintf(stderr,"Failed to initialise ISDB-T parameters\n");
    }

}

lelegard, 2017-10-10:
Hi,

Good idea. I assume you would like to drive ISDB-T Dektec modulators using the `dektec` output plugin, right?

However, this kind of feature shall be tested thoroughly. I have no access to ISDB-T stream, no ISDB-T Dektec modulator, no ISDB-T STB or TV set. This is precisely why the `dektec` plugin does not support ISDB-T yet.  I cannot blindly write code, insert it in TSDuck and wait for bug reports. So, I am not the right person to implement this. Since this is probably only a matter of adding a few options in the plugin and translate them to DTAPI calls, this should be accessible to a developer with an ISDB-T test environment.

Any volunteer?

EQMOD, 2017-10-11:
Im trying to compile it with the output plugin but it seems the compiler cannot locate "DtIsdbtPars" which is actually define in DTAPI.h

lelegard, 2017-10-11:
Yes, `DtIsdbtPars` is defined in DTAPI.h indeed.

From where did you use it? If you want to enhance the `dektec` plugin, you must enhance the class `DektecOutputPlugin` which is in files `tsDektecOutputPlugin.h` and `.cpp` but not in a plugin file.

As a general rule, all code referencing DTAPI shall be in the same DLL / shareable image `tsduck.dll` or `.so`. This is why the plugin `dektec` and the tool `tsdektec` are only small wrappers to classes in the DLL/so. The reason is that DTAPI is only provided as a static library on Windows and compiled without `dllexport` attribute. So, you can use DTAPI either from one single DLL or from an executable / other DLL's, but not both. Since `tsduck.dll` needs to reference DTAPI, it cannot be used from elsewhere. This is the usual Windows mess.

Could this be the source of your problem?

braghetto, 2018-07-20:
Hi guys!

I made a product using the Dectek demodulator DTA-2138B in ISDB mode. I have one with me for tests.

I intend to contribute to the project, let's see if I can implement it.

But my first try will be implementing the ISDB-Tb (brazilian version) SI to analyse TS files.
The core japanese ISDB specs are the same in brazilian version, just only some additions.

ABNT NBR 15603-2 Standart:
http://www.telemidia.puc-rio.br/~rafaeldiniz/public_files/normas/SBTVD/pt_BR/ABNTNBR15603-2_2007Vc_2008Ing.pdf
(english version)

I will try to get some free time for these tasks, I need to remember C++  :)

lelegard, 2018-08-08:
Hi @braghetto 

You are welcome to contribute. Here in Europe, we are DVB-oriented and we lack experience in ISDB. So, anyone based in an ISDB area is welcome.

Your TS files would be appreciated. There is a stream repository [here](https://tsduck.io/streams/) where I can make them public.

braghetto, 2018-08-10:
Ok here are the links for download the ts files:

TS File 1:
https://drive.google.com/file/d/1c51_lkgs1d90AYoJG0_uShk_VpywoYd3/view?usp=sharing

TS File 2:
https://drive.google.com/open?id=1fs1dJhNsx_ehJzRR_kUtFEgqfo_9NKhx

TS File 1 (after resync):
https://drive.google.com/open?id=1wlWi0yIihAycRAwIXIoMbQEOaZlXrnir

Ts File 2 (after resync):
https://drive.google.com/open?id=15J6flUCZvCxgwTFvgzG6BnxQPiSnDwxz

Each file has 925MB.

The original files, the firts two, are recorded using the Dektec demodulator DTA-2138B in ISDB mode here in Brazil.
The last two files are the same after tsduck resync to be able to analyse them. But the first ones are more interessting :)

These files were recorded in a poor reception condition, worse than indoors reception. In order to test the signal locking capacity of the demodulator and its effects on ts files.
All files have null/stuffing packets, and the first two still have the sync and corruption problems on the stream.

About stream contents:
The two files contain two programs each, one h264 1080i, and one low-resolution copy for mobile reception called OneSeg.

lelegard, 2018-08-10:
Thanks,

I have downloaded the four files.

> The last two files are the same after tsduck resync to be able to analyse them. But the first ones are more interessting :)

What do you mean by "more interesting" ? The size difference between the unsync'ed and sync'ed files is only 68 bytes. The unsync'ed files start with a 52-byte "header" containing only FF and are truncated in the middle of the last TS packet to get a round-figure size of 969984000 bytes. These are probably an artifact of the capture tool. Apart from that, unsync'ed and sync'ed files are strictly identical.

braghetto, 2018-08-10:
Hello @lelegard !
You are right!

Sorry, I quickly tested these files with tsduck. I did not even compare the files after the sync.
I initially considered that the sync tool would have made major changes to the files.

And yes, the artifacts at the beginning of the stream come from the demodulator buffer when started.

The tool used for recording are the C++ sample code from Dektec API. 
Actually a really simple code, it only sends the parameters of initialization and configuration to the demodulator. 
After that this code only do a loop reading the demodulador internal buffer and transfering these data to a file.
Exactly the same as the RF demodulator recorded the data in the buffer.

Although the signal lock status from the demodulator also considers the data stream flow beyond the physical layer flow to determine the lock condition.
But it does not make any kind of interpretation of this data.
It would be something of a kind. "I have OK status in demodulation! Do we have data flowing? Yes?! So the overall status is OK!"

P.S.: I've been studying tsduck code, documentation, and some commits. Where would be the most appropriate place to post questions about the overall system architecture?

I'm trying to make a decision on the best way to implement the ISDB service, but taking advantage of the DVB descriptors and tables. In fact the EN300 is almost all contained in the ISDB specs.
I think of starting contributing with some extra descriptors and tables that isdb has.

Regards!

lelegard, 2018-08-10:
> P.S.: I've been studying tsduck code, documentation, and some commits. Where would be the most appropriate place to post questions about the overall system architecture?

Please open an issue here. The GitHub "issue" system is not reserved to _problems_, it is a powerful tools for exchanging information and knowledge.

lelegard, 2018-08-11:
Hi @braghetto,

I have uploaded your ISDB-Tb files in the [public streams repository](https://tsduck.io/streams/). I kept only the resync'ed files, with the names of the original files. Thank you for your contribution. This will allow TSDuck users to study ISDB-Tb streams.

peter-stanton, 2022-04-22:
I'm also interested in getting this working (as I said in issue 979), but I don't really know the code. I'm going to start trying to understand how the other modulations work and maybe I could write an implementation at some point. I have hardware to test ISDB-T