tsduck/tsduck #11: Tuning Space is Invalid

ggnull35, 2017-10-29:
Hi,

I have TBS 5925 DVB-S2 Tuner Card (external USB). I try to tune, but I fail. Here is the log:

C:\Program Files (x86)\TSDuck\bin>tslsdvb.exe
tslsdvb: fail to set default tuning space GkDVB-C: The handle is invalid.
0: "TBS 5925 DVBS2 Tuner" (DVB-S)

C:\Program Files (x86)\TSDuck\bin>tsp -I dvb --tune "11096:h:0:30000"
* Error: dvb: fail to set default tuning space GkDVB-C: The handle is invalid.
* Error: dvb: no input DVB signal after 5000 milliseconds

BTW, both 64-bits and 32-bits version throw the same error.

Probably tuning space is set wrong (GkDVB-C) ?

ggnull35, 2017-10-29:
Verbose Log of Tuner:
C:\Program Files (x86)\TSDuck\bin>tslsdvb.exe -v
tslsdvb: fail to set default tuning space GkDVB-C: The handle is invalid.

0: "TBS 5925 DVBS2 Tuner" (DVB-S)

  Signal locked:    no
  Network provider: Microsoft DVBS Network Provider

  DirectShow graph:

    - Filter "NetworkProvider"
      class GUID: CLSID_Microsoft DVBS Network Provider
      Some supported interfaces:
        interface IAMovieSetup
        interface IBDA_NetworkProvider
        interface IBDA_TIF_REGISTRATION
        interface IBaseFilter
        interface IFrequencyMap
        interface IMPEG2_TIF_CONTROL
        interface IMediaFilter
        interface IScanningTuner
        interface ITuner

    - Output pin "Antenna Out", id "Antenna Out"
      Some supported interfaces:
        interface IPin
        interface IQualityControl

    - Media major type KSDATAFORMAT_TYPE_BDA_ANTENNA
      subtype MEDIASUBTYPE_None
      format FORMAT_None

    - Input pin "Input0", id "0"
      Some supported interfaces:
        interface IKsControl
        interface IKsPin
        interface IKsPropertySet
        interface IMemInputPin
        interface IPin
        interface IQualityControl
        interface IStreamBuilder
      IKsPropertySet properties support:
        SIGNAL_STRENGTH (SignalStats) : get
        SIGNAL_QUALITY (SignalStats) : get
        SIGNAL_PRESENT (SignalStats) : get
        SIGNAL_LOCKED (SignalStats) : get
        RF_TUNER_FREQUENCY (FrequencyFilter) : get set
        RF_TUNER_POLARITY (FrequencyFilter) : get set
        RF_TUNER_RANGE (FrequencyFilter) : get set
        RF_TUNER_FREQUENCY_MULTIPLIER (FrequencyFilter) : get set
        LNB_LOF_LOW_BAND (LNBInfo) : get set
        LNB_LOF_HIGH_BAND (LNBInfo) : get set
        LNB_SWITCH_FREQUENCY (LNBInfo) : get set
      IKsControl properties support:
        SIGNAL_STRENGTH (SignalStats) : get
        SIGNAL_QUALITY (SignalStats) : get
        SIGNAL_PRESENT (SignalStats) : get
        SIGNAL_LOCKED (SignalStats) : get
        RF_TUNER_FREQUENCY (FrequencyFilter) : get set
        RF_TUNER_POLARITY (FrequencyFilter) : get set
        RF_TUNER_RANGE (FrequencyFilter) : get set
        RF_TUNER_FREQUENCY_MULTIPLIER (FrequencyFilter) : get set
        LNB_LOF_LOW_BAND (LNBInfo) : get set
        LNB_LOF_HIGH_BAND (LNBInfo) : get set
        LNB_SWITCH_FREQUENCY (LNBInfo) : get set

    - Filter "Tuner"
      class GUID: CLSID_Generic WDM Filter Proxy
      Some supported interfaces:
[... 166 lines cut ...]
           interface IAMPushSource
           interface IMPEG2PIDMap
           interface IPin
           interface IQualityControl

       - Media major type KSDATAFORMAT_TYPE_MPEG2_SECTIONS
         subtype KSDATAFORMAT_SUBTYPE_DVB_SI
         format FORMAT_None

       - Input pin "IB Input", id "IB Input"
         Some supported interfaces:
           interface IMPEG2PIDMap
           interface IMemInputPin
           interface IPin
           interface IQualityControl

       - Filter "TIF"
         class GUID: CLSID_BDA MPEG2 Transport Information Filter
         Some supported interfaces:
           interface IAMovieSetup
           interface IBaseFilter
           interface IFrequencyMap
           interface IMediaFilter
           interface IMediaSeeking
           interface ITuneRequestInfo

ggnull35, 2017-10-29:
Also -e output is here:

C:\Program Files (x86)\TSDuck\bin>tslsdvb.exe -e >d:\out
tslsdvb: fail to set default tuning space GkDVB-C: The handle is invalid.
tslsdvb: IPropertyBag::Read: The system cannot find the file specified.

0: "TBS 5925 DVBS2 Tuner" (DVB-S)

=== Device category KSCATEGORY_BDA_NETWORK_PROVIDER

device "Microsoft ATSC Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
    interface IBDA_TIF_REGISTRATION
    interface IBaseFilter
    interface IFrequencyMap
    interface IMPEG2_TIF_CONTROL
    interface IMediaFilter
    interface IScanningTuner
    interface ITuner

  - Pin "Antenna Out", direction: output
    Some supported interfaces:
      interface IPin
      interface IQualityControl

device "Microsoft DVBC Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
    interface IBaseFilter
    interface IMediaFilter
    interface IScanningTuner
    interface ITuner

  - Pin "Antenna Out", direction: output
    Some supported interfaces:
      interface IPin
      interface IQualityControl

device "Microsoft DVBS Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
    interface IBaseFilter
    interface IMediaFilter
    interface IScanningTuner
    interface ITuner

  - Pin "Antenna Out", direction: output
    Some supported interfaces:
      interface IPin
      interface IQualityControl

device "Microsoft DVBT Network Provider"
  Some supported interfaces:
    interface IAMovieSetup
    interface IBDA_NetworkProvider
    interface IBDA_TIF_REGISTRATION
    interface IBaseFilter
    interface IFrequencyMap
    interface IMPEG2_TIF_CONTROL
    interface IMediaFilter
    interface IScanningTuner
    interface ITuner

  - Pin "Antenna Out", direction: output
    Some supported interfaces:
      interface IPin
[... 358 lines cut ...]

  - Pin "Input0", direction: input
    Some supported interfaces:
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMemInputPin
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

  - Pin "MPEG2 Transport", direction: output
    Some supported interfaces:
      interface IAMBufferNegotiation
      interface IAMStreamConfig
      interface IAMStreamControl
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMediaSeeking
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

lelegard, 2017-10-29:
Hi,

Thanks for providing the right information. What is your Windows version? Can you confirm that you tried with TSDuck 32-bit ?

The last time I had this very same problem, I used TSDuck 64-bit on Windows 7 64-bit. The tuner was a PCTV DVB-S2, USB. There was some issue with this device's driver or DirectShow filter when used from a 64-bit application. Using TSDuck 32-bit, it worked fine. Several other tuners worked well on Windows with TSDuck 64-bit. So, it is very dependent on the driver.

ggnull35, 2017-10-29:
Hi, my pleasure and thanks to all for this great project work. I use Windows 10 Insiders Edition. I installed both TSDuck 32bit and 64bit editions from setup. I tried both versions in their related bin dirs. I can try other Windows 10 installations on different PCs as well. I will try to test with different HW also. BTW, i can supply HW foe tour test also.

ggnull35, 2017-10-30:
Hi,

I fixed the issue. I guess the problem is related with installation. I removed 32 bits & 64 bits versions of TSDuck. Re-installed 64 bits version, rebooted and the error message is gone. Maybe installing both versions on same PC makes confusion ?

lelegard, 2017-10-30:
Could you also try `tslsdvb --debug` please?

lelegard, 2017-10-30:
Anyway, your tuner is actually successfully found since it is listed. The error message is probably a spurious one. The real problem is why does this error occur and how could it be removed. But otherwise, your tuner is probably functional. Did you try to receive a signal using the `dvb` plugin?

To give some background on DVB tuners on Windows, you may read the relevant section in the last chapter "hardware device support" of the TDuck user's guide. Compared to Linux which has a consistent and simple API for DVB drivers, Windows uses the complex DirectShow framework. There is no - or I have not found any - simple way to "open" a tuner and "read" TS packets. You must create a DirectShow graph with many DirectShow filters, some from Microsoft, some from the tuner vendor and one from TSDuck. And, to be even more complex, there is no unique and straightforward way to build a DirectShow graph to interface a tuner. TSDuck has to try several combinations until one works. In the process of finding the right combination of filters to interface a given tuner, many errors occur when filters refuse to connect one to another. But these errors are normally not reported because they are normal, precisely. The error you encounter is not a normal one and it is reported. However, since the tuner is finally detected, TSDuck has managed to build a DirectShow graph to use it. This is why it is probably functional and you should try it.

But this error puzzles me and I would like to understand where it comes from. Especially because "GkDVB-C" is a weird (and non-standard) name for a tuning space and it should not be associated to a DVB-S tuner.

lelegard, 2017-10-30:
Sorry, I missed your answer about fixing the issue.

Normally, having both 32 and 64 bits versions installed on the same machine should not interfere. Except from adding the TSDuck directory in `%Path%`, the installation does nothing at system level. All TSDuck commands locate their files and plugins from their own executable directory. If you specify the complete path of the command exe, it works. I do that regularly: Visual Studio generates 32-bit and 64-bit binaries in separate directories and I test them from where they are, either 32 or 64 bits.

From the description of your problems, it is possible that the DirectShow filters repository was somehow messed up. Did you just install / upgrade the tuner's drivers, without rebooting?

Let me know if I can close the issue.

ggnull35, 2017-10-30:
Hi again,

Actually I spoke early. I just tried tslsdvb.exe and saw no error:

C:\TSDuck\bin>tslsdvb.exe --debug
debug level set to 1
found tuner filter "TBS 5922 USB DVB-S2 Tuner"
trying provider filter "Microsoft ATSC Network Provider"
found tuning space "Local ATSC Digital Antenna" (ATSC)
found tuning space "Local ATSC Digital Cable" (ATSCCable)
failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
trying provider filter "Microsoft DVBC Network Provider"
found tuning space "Local DVB-T Digital Antenna" (DVB-T)
found tuning space "GkDVB-C" (GkDVB-C)
found tuning space "Local ISDB-T Digital Antenna" (ISDB-T)
failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
trying provider filter "Microsoft DVBS Network Provider"
found tuning space "DD2 DVB-S" (DD2 DVB-S)
found tuning space "Default Digital DVB-S Tuning Space" (DVB-S)
found tuning space "GkDVB-S" (GkDVB-S)
found tuning space "Default Digital ISDB-S Tuning Space" (ISDB-S)
using provider filter "Microsoft DVBS Network Provider", tuning space "DD2 DVB-S" ("DD2 DVB-S")
trying direct connection from tuner (no receiver)
failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
trying receiver filter "TBS 5922 USB DVB-S2 Capture"
trying TIF "BDA MPEG2 Transport Information Filter"
using TIF "BDA MPEG2 Transport Information Filter"
using receiver filter "TBS 5922 USB DVB-S2 Capture"
IBDA_DigitalDemodulator in tuner: 0
IBDA_DigitalDemodulator2 in tuner: 0
IBDA_SignalStatistics in tuner: 0
IKsPropertySet in tuner: 2
0: "TBS 5922 USB DVB-S2 Tuner" (DVB-S)

This is 64bit version. Than I tried tsp with DVB input and IP output:

C:\TSDuck\bin>tsp --debug -I dvb --tune 11096:h:0:30000 -O ip 233.33.33.1:5001
debug level set to 1
* Debug: debug level set to 1
trying to load C:\TSDuck\bin\tsplugin_dvb.dll
symbol tspNewOutput not found in C:\TSDuck\bin\tsplugin_dvb.dll
symbol tspNewProcessor not found in C:\TSDuck\bin\tsplugin_dvb.dll
trying to load C:\TSDuck\bin\tsplugin_ip.dll
symbol tspNewProcessor not found in C:\TSDuck\bin\tsplugin_ip.dll
* Debug: dvb: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: dvb: found tuner filter "TBS 5922 USB DVB-S2 Tuner"
* Debug: dvb: trying provider filter "Microsoft ATSC Network Provider"
* Debug: dvb: found tuning space "Local ATSC Digital Antenna" (ATSC)
* Debug: dvb: found tuning space "Local ATSC Digital Cable" (ATSCCable)
* Debug: dvb: failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
* Debug: dvb: trying provider filter "Microsoft DVBC Network Provider"
* Debug: dvb: found tuning space "Local DVB-T Digital Antenna" (DVB-T)
* Debug: dvb: found tuning space "GkDVB-C" (GkDVB-C)
* Debug: dvb: found tuning space "Local ISDB-T Digital Antenna" (ISDB-T)
* Debug: dvb: failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
* Debug: dvb: failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
* Debug: dvb: trying provider filter "Microsoft DVBS Network Provider"
* Debug: dvb: found tuning space "DD2 DVB-S" (DD2 DVB-S)
* Debug: dvb: found tuning space "Default Digital DVB-S Tuning Space" (DVB-S)
* Debug: dvb: found tuning space "GkDVB-S" (GkDVB-S)
* Debug: dvb: found tuning space "Default Digital ISDB-S Tuning Space" (ISDB-S)
* Debug: dvb: failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
* Debug: dvb: IDVBSLocator = 2107911733216, IDVBSLocator2 = 2107911733216
* Debug: dvb: setting IDVBSLocator2.DiseqLNBSource to 1
* Error: dvb: ITuneRequest::put_Locator: Type mismatch.
* Debug: dvb: failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
* Debug: dvb: trying provider filter "Microsoft DVBT Network Provider"
* Debug: dvb: found tuning space "Local DVB-T Digital Antenna" (DVB-T)
* Debug: dvb: found tuning space "GkDVB-C" (GkDVB-C)
* Debug: dvb: found tuning space "Local ISDB-T Digital Antenna" (ISDB-T)
* Debug: dvb: failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
* Error: dvb: ITuneRequest::put_Locator: Type mismatch.
* Debug: dvb: failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
* Debug: dvb: trying provider filter "Microsoft Network Provider"
* Debug: dvb: Cannot map Microsoft Network Provider (class CLSID_Microsoft Network Provider) to known tuner type
* Error: dvb: No DVB tuner device
* Debug: Report logging thread terminated

It says tuner not found. I read through the debug info and you are right it seems my PC is messed up. On other Windows machine, same HW, same driver same TSDuck 64bit version works.

Any idea how to reset that PC settings ?

ggnull35, 2017-10-30:
Here is a GraphEdit Test:
[image]

lelegard, 2017-10-30:
I notice the following weird display:
```
trying provider filter "Microsoft DVBS Network Provider"
found tuning space "DD2 DVB-S" (DD2 DVB-S)
found tuning space "Default Digital DVB-S Tuning Space" (DVB-S)
found tuning space "GkDVB-S" (GkDVB-S)
found tuning space "Default Digital ISDB-S Tuning Space" (ISDB-S)
using provider filter "Microsoft DVBS Network Provider", tuning space "DD2 DVB-S" ("DD2 DVB-S")
```

TSDuck explores the various predefined network provider filters and, for each of them, ask it which tuning space it supports. Here, your "Microsoft DVBS Network Provider" supports 4 tuning spaces. On my PC, I can see only two, DVB-S and ISDB-S. They are predefined by Windows. Here, you have two more: "DD2 DVB-S" and "GkDVB-S". I do not know where they come from.

Moreover, "DD2 DVB-S" is selected because it is compatible and comes first. On my PC, the Microsoft-defined "Default Digital DVB-S Tuning Space" is used. This can make a difference.

You wrote that you have two PC's, one on which TSDuck works and one on which it does not. For each of them, could you indicate:
- Windows version (7/8/10, 32/64 bits)
- Actual tuners which are plugged, only one, same one?
- Same version of the tuner vendor software or different
- Presence of "DD2 DVB-S" and "GkDVB-S" in the debug output

ggnull35, 2017-11-01:
Both PCs use Windows 10 64-bits. (One is using Insider 170919-1503, other is normal one.)
The Insider version PC is a development PC, which has VS17 SDKs and several others.
I use exactly same HW, TBS 5922.
I use exactly same driver: 2.0.3.2

Working PC tslsdvb output log:

debug level set to 1
found tuner filter "TBS 5922 USB DVB-S2 Tuner"
trying provider filter "Microsoft ATSC Network Provider"
found tuning space "Local ATSC Digital Antenna" (ATSC)
found tuning space "Local ATSC Digital Cable" (ATSCCable)
failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
trying provider filter "Microsoft DVBC Network Provider"
no tuning space found
failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
trying provider filter "Microsoft DVBS Network Provider"
found tuning space "Default Digital DVB-S Tuning Space" (DVB-S)
found tuning space "Default Digital ISDB-S Tuning Space" (ISDB-S)
using provider filter "Microsoft DVBS Network Provider", tuning space "DVB-S" ("Default Digital DVB-S Tuning Space")
trying direct connection from tuner (no receiver)
failed to connect pins, status = 0x80040217, No combination of intermediate filters could be found to make the connection.
trying receiver filter "TBS 5922 USB DVB-S2 Capture"
trying TIF "BDA MPEG2 Transport Information Filter"
using TIF "BDA MPEG2 Transport Information Filter"
using receiver filter "TBS 5922 USB DVB-S2 Capture"
IBDA_DigitalDemodulator in tuner: 0
IBDA_DigitalDemodulator2 in tuner: 0
IBDA_SignalStatistics in tuner: 0
IKsPropertySet in tuner: 2
0: "TBS 5922 USB DVB-S2 Tuner" (DVB-S)

No "DD2 DVB-S" and "GkDVB-S" filters exist.

ggnull35, 2017-11-01:
It looks like those filters belongs to Microsoft:

[image]

ggnull35, 2017-11-01:
I deleted registery entries DD2 DVB-S, GkDVB-S, GkDVB-C and problem is fixed totally.
TSP is working now.

closed 2017-11-01 by ggnull35