tsduck/tsduck #13: Static Build

ggnull35, 2017-11-01:
Hi,

Is it possible to static build tools under Windows ? Such as tsp.

Kind Regards,

lelegard, 2017-11-01:
Hi,

Technically, it is possible to statically build all TSDuck tools. All of them will work correctly, _except_ `tsp`.

TSDuck contains a lot of code and many executables and plugins. But the architecture of TSDuck is such that the vast majority of the code is made of reusable C++ classes. All these classes are gathered in a big DLL `tsduck.dll`. All tools and plugins contain not much code and are very small. They reference `tsduck.dll` for most of the features. 

On Windows (and Unix), we build both the DLL `tsduck.dll` (`tsduck.so`) and the corresponding static library `tsducklib.lib` (`libtsduck.a`). The unitary tests are built and run on the two versions to make sure the code behaves correctly when used from the shared or static library.

So, in principle, all TSDuck tools can be built statically. However, this would be very inefficient. The total amount of binary would be much too large because the common C++ classes would be duplicated.

For `tsp`, this is different. `tsp` loads plugins which are DLL's. All these DLL's also uses `tsduck.dll`. So, in the same process, you get one executable `tsp.exe` and many DLL's `tsplugin_*.dll` and all of them reference the same unique `tsduck.dll`. If you statically build `tsp.exe` and all `tsplugin_*.dll`, you have two problems:

1. You have a space problem. Inside the virtual space of the process, the common code will be duplicated 10 times, for instance, if you have 9 plugins.

2. You have a functional problem because some common classes have static shared data. When statically built, `tsp` and the plugins will access different - and gradually divergent - copies of the same data. So, at best, it does not work as expected and, at worst, it crashes.

So, no, you should not build `tsp`statically.

ggnull35, 2017-11-01:
Thank you for detailed explanation. I understood well.

closed 2017-11-01 by ggnull35