tsduck/tsduck #14: Looking for testers : DVB tuners on Windows

lelegard, 2017-11-10:
Hi TSDuck users,

TSDuck [version 3.4-20171110](https://github.com/tsduck/tsduck/releases/tag/v3.4-20171110) is out.

Among other features and improvements, the support for DVB tuners on Windows has been partly rewritten. The previous version was written 10 years ago on XP or Vista (I cannot even remember). Since then, there have been some significant improvements in DirectShow. With version 3.4, TSDuck uses DirectShow filters which did not exist 10 years ago. This should improve the support for DVB tuners on Windows. As a start, it solved the 32-bit vs. 64-bit issue which was reported with some drivers.

However, messing with DirectShow can be tricky since different tuners come with different BDA drivers and DirectShow filters. The experience had shown (but it was a long time ago) that some proprietary drivers may require very specific forms of DirectShow graphs.

So, I would appreciate if all TSDuck users with DVB tuners on Windows could try this new version and report their feedbacks in this thread. If it works, please indicate the tuner type and characteristics so that I could include this in the documentation. If if does not work, please add the output of `tslsdvb --debug`.

Thanks in advance for your cooperation.

PS: @rezna11, @ggnull35, your feedback would be greatly appreciated

rezna11, 2017-11-11:
I briefly tried couple of tuners, all worked. I tuned in DVB-S2 or DVB-T frequency and streamed to VLC to check. Tried with the 64-bit version of TSDuck.

TT-budget S2-4100 DVBS/S2 Tuner (Windows 10, Ku-band)
TeVii DVB-S/S2 Receiver (Windows 10, Ku-band)
TBS 6284 DVBT/T2 Tuner (Windows 10, DVB-T)
TBS 6903 DVB-S/S2 (Windows 10, C-band)
TBS 5925 USB (Windows Server 2016, Ku-band)
TBS 5922 USB (Windows Server 2016, Ku-band)
DVBSky S960 USB (Windows Server 2016, Ku-band)

lelegard, 2017-11-11:
Thanks Jan,
This is good news. It is quite fortunate that you have so many different tuners !
Thanks again for your help.

lelegard, 2017-11-13:
@rezna11  I am including your test results in the TSDuck documentation. Could you please specify if the TeVii DVB-S/S2 is a S660 or S662 ?

rezna11, 2017-11-13:
It's actually TeVii S482, I forgot to mention that.

By the way I also tried "IT9135 BDA Filter" (DVB-T) and it tuned and streamed, but after some time the picture started pixelating. I used -P regulate -P pcrbitrate, but that didn't solve it.

alexander-nf, 2018-01-10:
I have successfully tested by `tsp -I dvb -a 0 --uhf-channel 30 -P skip 5000 -P until -s 120 -P analyze -o 30ch.log -O drop` two DVB-T2(/C) tuners under Windows 7 64-bit.

1. USB PCTV nanoStick T2 (290e):
```
> tslsdvb
0: "PCTV 290e/460e BDA 2875 TVTuner" (DVB-T)
1: "PCTV 80e/290e/460e BDA 2875 TVTuner C" (DVB-C)
```

2. PCIe Blackgold BGT3620 (the value for `-a` was changed from `0` to `3`):
```
> tslsdvb
0: "7231 BDA DVBC Tuner" (DVB-C)
1: "7231 BDA DVBC Tuner" (DVB-C)
2: "7231 BDA DVBT Tuner" (DVB-T)
3: "7231 BDA DVBT Tuner" (DVB-T)
4: "7231 BDA DVBC Tuner (MCE)" (DVB-T)
5: "7231 BDA DVBC Tuner (MCE)" (DVB-T)
```

lelegard, 2018-01-10:
Hi @alexander-nf 

Thanks a lot for the feedback. I will add these two devices in the documentation.

alexander-nf, 2018-01-11:
GOTVIEW MasterHD 3 USB tuner was successfully tested by `tsp -I dvb -a 0 --uhf-channel 30 -P skip 5000 -P until -s 120 -P analyze -o 30ch.log -O drop`, DVB-T2, Windows 7 64-bit:

```
> tslsdvb
0: "GOTVIEW MasterHD3 DVB-T/T2/C Tuner" (DVB-T)
```

lelegard, 2018-01-11:
Thank you all for your feedback. I originally opened this one to check that a substantial change in the Windows implementation did not break anything. But I realise that your feedback is also important to build a list of supported tuners.

Consequently, I close this specific issue and opens the #24 to collect more general feedbacks.

Please do not add comment to this one, use #24.

closed 2018-01-11 by lelegard