tsduck/tsduck #15: Custom EPG table injection

Nethe, 2017-11-16:
Hi guys. 

I am just new into the tsduck and trying to play with it. My current issue is that i am trying to inject custom EPG tables into the transport stream, but it is not working for me. 

The procedure:
1. get example binary file from current live stream or transport stream file (*.ts), everything is okay:
`tsp -I file TS.ts  -P tables --binary-output test.bin -O drop`
2. Decompile the binary file to XML, everything is okay, the file looks good:
`tstabcomp -d test.bin -o test.xml`
3. Made some changes and compile the XML file back to the binary, seems to be okay:
`tstabcomp -c test.xml -o test.bin`
4. Now, the main step, trying to inject this EPG tables back to the TS (or live stream):
**TS file:**
`tsp -I file TS.ts -P inject test32.bin -O ip 239.0.0.155:1234`
**Live streaming:**
`323  tsp -I ip 239.100.0.25:1234 -r -P inject test2.bin --replace --pid 18 -O ip -l 239.0.0.155:1234`

The issue is that custom EPG tables are not injected, so what I am doing wrong?

rezna11, 2017-11-16:
Try to run tsp with -v, maybe that will tell you more as to why it's not injecting ... or maybe it is actually injecting but the EIT data is not correct and your player is not able to display the EPG - did you check whether your new EIT tables are actually in the output or not by trying output to file and then look for the new tables?

lelegard, 2017-11-16:
Hi @Nethe,

Welcome to TSDuck.

I see a few potential issues with your commands. Probably not related to your problem however.

`tsp -I file TS.ts -P tables --binary-output test.bin -O drop` : you should add `--pid 18` to plugin `tables` to make sure you collect EIT's only.

`tsp -I file TS.ts -P inject test32.bin -O ip 239.0.0.155:1234` : this command should generate an error since the option `--pid` is mandatory in the plugin `inject` (otherwise we don't known where to inject the tables).

`tsp -I ip 239.100.0.25:1234 -r -P inject test2.bin --replace --pid 18 -O ip -l 239.0.0.155:1234` : this one should generate an error as well. The option `-l` specify the outbound local interface (in case you have many on your system) and its value is missing. More precisely `239.0.0.155:1234` is used as value and there are now two problems: The `-l` value in incorrect (`address:port` instead of `address`) and the mandatory destination `address:port` parameter has been swallowed by the option `-l`.

Generally speaking, try to replace the output by `-O file out.ts` and analyze the tables in PID 18 of the file `out.ts`. If the modified EIT's are there (they should be), then they are sent and the problem is probably in the player.

Nethe, 2017-11-20:
Hi @lelegard and @rezna11 , thanks for help. Will post feedback as soon as I test it.

lelegard, 2018-01-11:
Hi @Nethe 

Any news on this? Does it work? May we close the issue?

Nethe, 2018-01-11:
Hi, you can close it. 
Thanks.

closed 2018-01-11 by lelegard