tsduck/tsduck #16: Any plans for a Wiki or forum?

rezna11, 2017-11-16:
There are so many ways how to use TSDuck, wouldn't it be beneficial to have a Wiki page or similar resource where users could share examples of what they're doing with TSDuck? I don't mean to copy the manual but to share real-life use cases. It could be also used to ask questions because I'm not sure the Issues page here is the right place to just ask about how to achieve something, seems it's more to report bugs, right?

lelegard, 2017-11-16:
This is a good idea. In fact, I did not anticipate much feedback on such a very specific technical tool. It was made public about a year ago only, after more than 10 years of more or less active development (depending on periods). Making it public on GitHub was essentially a way to keep a safe backup of the repository when I started version 3.0. I am glad some people find it useful.

I just created a [Wiki](https://github.com/tsduck/tsduck/wiki) on GitHub. I am not completely sure that everyone is allowed to post articles here. The best way to know is that you write a useful article in it :)

Normally, a Wiki is a community reference knowledge base. For questions and discussions, a forum would be more appropriate. But I don't see any way to create a discussion forum, attached to GitHub project. And I do not feel like creating a forum in some external site.

rezna11, 2017-11-16:
Sure, no need for a forum yet, this is good enough to start with, but there's no option to add a page.

lelegard, 2017-11-16:
I changed the permissions of the Wiki. It was not cleared whether "_Restrict editing to users in teams with push access only_" applied to the Wiki only or if it was more general.

rezna11, 2017-11-17:
ok so I tried adding one new page - it works, but I'm not sure if it's possible to create some kind of a structure - now the page appears in the pages list on the right side, so all the pages would probably just add to the list, which might not look pretty

lelegard, 2017-11-17:
Yes, this is the limitation of the wiki over a discussion forum. Too bad that GitHub does not allow a discussion forum per project in addition to the wiki. SourceForge had it.

closed 2017-11-20 by lelegard