tsduck/tsduck #17: Raspberry Pi compiling issues: please check solutions

alexander-nf, 2018-01-10:
Hello Thierry!

First of all thank you for the project very much. I work in DVB field and I found TSDuck useful alongside professional proprietary software tools. Even more: your toolkit is very flexible.
Also I'd like to add I'm not a software developer, just code a bit for my own tasks, but it is obvious and clear fact: your project is very solid and made extremely professionally. (_It's not a gross flattery, but an admiring envy :-)_. ) The whole project from the architecture point of view to each small comment in documentation is deeply structured, harmonized and readable. Unit tests and final packaging for a distribution are perfect. The job is well done!

I have a few questions after using TSDuck, RTFM (I hope thoroughly) and looking across the code. Please read below (and I'll post a couple of issues/requests later):

I've got (_and solved?_) Rasberry Pi compilation problems, latest TSDuck release 3.6-439.

Hardware: **Rasberry Pi 3 Model B** V1.2
System software: **Rasbian 8**
Input device: USB DVB-T2 tuner **PCTV nanoStick T2** (290e)

```
$ uname -a
Linux dvb-pi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
```

1. Compilation error, **tsIPAddress.cpp**:
```
[CXX] tsIPAddress.cpp
tsIPAddress.cpp: In constructor 'ts::IPAddress::IPAddress(const sockaddr&)':
tsIPAddress.cpp:56:77: error: cast from 'const sockaddr*' to 'const sockaddr_in*' increases required alignment of target type [-Werror=cast-align]
         const ::sockaddr_in* sp = reinterpret_cast<const ::sockaddr_in*> (&s);
                                                                             ^
tsIPAddress.cpp: In member function 'void ts::IPAddress::copy(sockaddr&, uint16_t) const':
tsIPAddress.cpp:78:62: error: cast from 'sockaddr*' to 'sockaddr_in*' increases required alignment of target type [-Werror=cast-align]
     ::sockaddr_in* sp = reinterpret_cast< ::sockaddr_in*> (&s);
                                                              ^
tsIPAddress.cpp: In member function 'bool ts::IPAddress::resolve(const ts::UString&, ts::Report&)':
tsIPAddress.cpp:146:86: error: cast from 'sockaddr*' to 'const sockaddr_in*' increases required alignment of target type [-Werror=cast-align]
         const ::sockaddr_in* sp = reinterpret_cast<const ::sockaddr_in*> (ai->ai_addr);
                                                                                      ^
cc1plus: all warnings being treated as errors
../../Makefile.common:802: recipe for target 'release-arm/tsIPAddress.o' failed
make[2]: *** [release-arm/tsIPAddress.o] Error 1
```

SOLUTION: comment/delete `-Wcast-align` in Makefile.common.

1.1. Is it wrong or critical modification?

2. Compilation/linking error, **tsduck.so**:
```
[CC] release-arm/tsduck.so
/usr/bin/ld: ../../dektec/LinuxSDK/DTAPI/Lib/GCC4.4/DTAPI.o: Relocations in generic ELF (EM: 3)
...
... # (a lot of 'DTAPI.o: Relocations in generic ELF (EM: 3)' messages )
...
/usr/bin/ld: ../../dektec/LinuxSDK/DTAPI/Lib/GCC4.4/DTAPI.o: Relocations in generic ELF (EM: 3)
../../dektec/LinuxSDK/DTAPI/Lib/GCC4.4/DTAPI.o: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
Makefile:111: recipe for target 'release-arm/tsduck.so' failed
make[2]: *** [release-arm/tsduck.so] Error 1
```

SOLUTION: exclude Dektec support.
a)	Delete/comment `dtapi:` section in Makefile.
b)	Delete `./dektec` folder.

2.1. Is it possible to fix the point 2 properly and to use Dektec USB devices with Raspberry Pi?

lelegard, 2018-01-10:
Hi @alexander-nf 

First, thanks for the laudatory comments. I can just say that I know no other way to write code.

Thanks also for the detailed report and analysis.

I understand the two points which are both related to the ARM target. I used to have one ARM device a few years ago and did some work on it (hence a few ARM-related definitions in the common makefile which I use unchanged for virtually all projects). But I no longer have any. So, I never compiled TSDuck on ARM.

First issue: I permanently removed `-Wcast-align` on ARM targets. Browsing the Web, I found many similar issues and not only about `sockaddr` / `sockaddr_in`. The general consensus seems to avoid `-Wcast-align` on ARM. So let it be...

Second issue: The Dektec API provides binaries for Linux Intel only (i686 and x86_64). So, attempting to download it and use it on any other architecture is a non-sense. This is a bug of the makefile. I already excluded Dektec on MacOS but blindly used it on Linux, regardless of the architecture. I fixed this.

I pushed a proposed fix for the two problems (commit 96825ad0fc975c455bcf1de3357bd7a8bb2ef9d1).

Could you please pull the repository again, try it on your Raspberry and give us your feedback? I am waiting for your feedback before closing this issue.

Thanks

alexander-nf, 2018-01-11:
Hi, Thierry!

Thank you for the comments and the answers, Dektec issue is clear for me now.

I've just tried to compile TSDuck on Raspberry Pi from scratch and I've got:

```
  [CXX] tsGitHubRelease.cpp
In file included from tsGitHubRelease.cpp:48:0:
tsWebRequest.h: In member function 'const uint16_t& ts::WebRequest::proxyPort() const':
tsWebRequest.h:262:90: error: returning reference to temporary [-Werror=return-local-addr]
         const uint16_t& proxyPort() const { return _proxyPort == 0 ? _defaultProxyPort : _proxyPort; }
                                                                                          ^
cc1plus: all warnings being treated as errors
../../Makefile.common:806: recipe for target 'release-arm/tsGitHubRelease.o' failed
make[2]: *** [release-arm/tsGitHubRelease.o] Error 1
```

lelegard, 2018-01-11:
Thanks for the report.

The compiler seems definitely wrong and inconsistent here.
- The returned reference is not to temporary data but to a class member field. So the compiler is wrong.
- The same construct is used for the three adjacent members (proxyHost, proxyUser and proxyPassword) but the compiler does not complain. So the compiler is inconsistent. The compiler complains for an elementary type (uint16_t) but not for a class instance (UString).
- The code compiles correctly with GCC on Linux x86_64 and MS Visual Studio 2017.

Anyway, you may replace `const uint16_t& proxyPort()` with `uint16_t proxyPort()` in line 262 of tsWebRequest.h. This is functionally equivalent and should please the compiler. I will do the same in the base code for next releases. Using a const reference instead of a value makes sense for UString for performance reasons but does not make any difference with the integer. So, let's make the change permanent.

The affected code is new and was pushed the night before you pulled the new version. That can explain why you did not see it the first time. From my previous experience with ARM a few years ago, I recall a few compilation issues which were specific to GCC on ARM. If you encounter further compilation issues with TSDuck on ARM, do not hesitate to report them. But open a new issue for each of them so that they can be addressed separately.

What is your GCC version on ARM by the way? Which Linux distro do you use? Linaro?

alexander-nf, 2018-01-11:
I think it's possible to close the issue, I have created the new one for the new problem: #23.

closed 2018-01-11 by lelegard