tsduck/tsduck #18: How to select PLP while tuning/scanning RF-signal?

alexander-nf, 2018-01-10:
I realize that:
- Windows doesn’t have unified mechanism for the PLP selection, each DVB-tuner driver has its own implementation of such a mechanism.
- OS X/macOS – doesn’t work with DVB-devices _quite_ at all.
- Only Linux has TV framework/sybsystem.

I use USB DVB-T2 tuner PCTV nanoStick T2 (290e) supported by Linux kernels and it has multiple PLP functionality. For example VLC for Linux is able to select given PLP.

Is it possible lock frequency and specified PLP by Linux version of TSDuck?

lelegard, 2018-01-10:
Hi @alexander-nf 

Good point. I will do it. Stay tuned.

lelegard, 2018-01-11:
Hi,

It was not too difficult to add. This is now done in commit ebc898da6cbd3a62b181e25a0e5685c68a26b48b. The option `--plp value` has been added to the plugin `dvb` and to `tscan` (this is common code).

At expected, this option is effective only on Linux. You should get a warning on Windows.

However, I have no DVB-T2 reception available and I cannot test it. So, you have to be the tester. Could you please pull the repository again, rebuild, test and report feedback here?

Thanks.

alexander-nf, 2018-01-11:
I've checked two UHF-channels by `tsp -I dvb` with `--plp 1` and `--plp 2` (the second and the third PLPs) using modifications of

`tsp -I dvb --uhf-channel 30 --plp 1 -P skip 5000 -P until -s 60 -P analyze -o 30ch_PLP1.log -O drop`

and I've got correct reports about each PLP.

Thanks a lot, Thierry!

lelegard, 2018-01-11:
You're welcome.
Closing this one.

closed 2018-01-11 by lelegard