tsduck/tsduck #20: Detection of SCTE 35 cue messages in TS

alexander-nf, 2018-01-10:
Is it possible to add detection of SCTE 35 cue messages (signalling for ad insertion, TS marking and segmentation)?

Is it enough for the detection just to add the following descriptors?

- [x] .\src\libtsduck\tsduck.dvb.names, `[StreamType]`:
`0x86 = SCTE 35`
                                                      
[SCTE 35](http://www.scte.org/SCTEDocs/Standards/ANSI_SCTE%2035%202017.pdf), 9.5.1. 'Constraints on splice_info_section()'.

- [x] .\src\libtsduck\tsduck.dvb.names, `[TableId]`:
`0xFC = SCTE 35 splice info`

[SCTE 35](http://www.scte.org/SCTEDocs/Standards/ANSI_SCTE%2035%202017.pdf), 9.2. 'Splice info section', Table 5.

- [ ] Also I haven't added PMT registration descriptor:
> The registration descriptor shall be carried in the `program_info` loop of the PMT for each program that complies with this standard.

[SCTE 35](http://www.scte.org/SCTEDocs/Standards/ANSI_SCTE%2035%202017.pdf), 8.1. 'Registration descriptor' and Table 1 'registration_descriptor()'.

Should SCTE 35 registration descriptor be added to .\src\libtsduck\tsduck.xml or somewhere else?

lelegard, 2018-01-10:
Hi @alexander-nf 

I am not familiar with this. I will look into it. Is it a US-only standard? In which country is it used?
Could you provide a sample captured stream containing this type of information?

alexander-nf, 2018-01-11:
SCTE 35 is not only US standard _de facto_. About 7-8 years ago it has become to be used in Europe, and not only by cable operators. And the process has been intensified after mass digitalization of broadcasting. DTMF cue tones in analogue audio channels of TV content were replaced by SCTE 35 data signaling in DPI (digital program insertion) PIDs of digital broadcasting services.

And I should mark that ITU has recommendation [J.181: Digital program insertion cueing message for cable television systems](https://www.itu.int/rec/T-REC-J.181-201401-I/en) based on SCTE 35.

Actually I am from Russia, broadcasters and operators in Russia, Kazakhstan use SCTE 104/35 for DPI. But it's not a local initiative to use SCTE standards, it was caused by European broadcasters and operators, in other words by non-US experience.
Also a lot of video encoders have SCTE 104-to-35 functionality (SCTE 104 is out of scope TSDuck and describes in particular DPI signaling encapsulated within SDI; many broadcasters have injectors or/and automation systems with SCTE 104/35 support).

A couple of examples of software projects with SCTE 35 detection and parsers:
- Dektec partners from DVBControl have SCTE 35 viewer option in their [DVBAnalyzer](https://www.dvbcontrol.com/dvbanalyzer/).
- Your colleague, Java-programmer Eric Berendsen implemented SCTE 35 support in his open-source offline [DVB Inspector](http://www.digitalekabeltelevisie.nl/dvb_inspector/).

It would be great if you implement correct/proper detection of SCTE 35 DPI PIDs.

Optional parsing of SCTE 35 packets/tables is more complex task and if you be interested to solve it, I'd be happy to test the result.

In any case I will capture or find TS samples with SCTE 35 signaling during this week. By the way, there is a XML schema: [SCTE35.xsd](http://www.scte.org/schemas/35/SCTE35.xsd).

alexander-nf, 2018-01-13:
Hi, Thierry!

Here the TS record is: [20180112_1129_UTC_Russian_mux1_T2MI_3PLPs_incl_SCTE35_in_PLP0.ts](https://yadi.sk/d/Hp_GSz883RRtie).

Description:
- size is about 5 GBytes;
- T2-MI stream has 3 PLPs;
- only PLP 0 has DPI PIDs (1015, 1035, 1045, 1055) with SCTE 35 messages;
- probably it is enough to download 20-25% / 1 GB+ of the record, then you'll get SCTE 35 data from 3 of 4 captured DPI PIDs;
- SCTE 35 packets appear just seconds before splicing in/out-points (1-3 packet(s) with an equal content for the single point), most of the time you'll get _correct_ PID error for DPI PIDs, there is no SCTE 35 data in the stream except the moments;
- DPI PIDs / SCTE 35 sections in particular carry `pts_time` values which are equal to PTS value of the next I-IDR frame from video PID of the common TS service;
- a structure of the data in all SCTE 35 packets is quite identical, only the one SCTE 35 command (`splice_command_type = splice_insert` / `0x05`) is used in `splice_info_section()`;
- only the one type of splice descriptors (`avail_descriptor`) follows after the command body.

I'll try to get samples of the other commands and descriptors (for example, for a stream marking/segmentation), but I'm not sure if I can do this.

lelegard, 2018-01-15:
Hello @alexander-nf ,

Thank you for the TS. It was impossible to write proper code without testing real stuff.

I have added most of the SCTE signalization for now: 
- Various definitions you already prepared in your original report.
- Descriptors for the PMT: registration and cue_identifier. They can be also manipulated in XML form.
- Splice Info Table (the one in the info stream 0x86); Can be partially displayed. The details of the various subcommands are simply hex-dumped.

I have not implemented the following and I won't implement it on the short term because it will take more time than available now.
- Full support of Splice Info Table, including XML format.
- Reporting embedded triggers in the audio/video PID's.

One detail : The SCTE signalization remains fundamentally incompatible with DVB. The cue_identifier_descriptor, for instance, has tag 0x8A. In DVB terms, this is a private descriptor and requires a preceding private data specifier (PDS) descriptor. In a DVB environment, having this descriptor in the PMT in the middle of DVB descriptors makes the descriptor list invalid. In the TS you provided, the cue identifier descriptor is simply not used, probably a good workaround to avoid descriptor clashes. Otherwise, you need specific non-standard filtering in the local retransmitters, where the ads are inserted, and then PMT cleanup before retransmission to retail STB's or TV sets.

Anyway, there are many more or less clean ways to cope with this incompatibility. But we are moving toward non-standard and unorganized signalization, a mess that DVB successfully avoided in the last 20 years.

Fixed in commit 1a49f8dc723a1ac3a5b44b46c73146d2f2fccbb8

lelegard, 2018-01-27:
Hi @alexander-nf 

I went back to SCTE 35 for a while (I was just too curious). I added a few more information but I also added a new plugin named `rmsplice` which uses SCTE 35 splice information to locate and remove ads insertion.

I have tested it on the stream you provided and, indeed, it cuts programs at places which look like ads (I don't understand Russian but ads exhibit the same stupid images everywhere and it is not too difficult to detect them visually).

I noted a small glitch at the video transition even though the video stream was spliced at exactly the same PTS value as specified in the SCTE 35 splice commands. The splice commands are apparently aligned on video while audio and subtitles have slightly different PTS values. Use `tsp -v` to get verbose messages on splice commands and splicing points. I do not know if this is due to a non-splice-resistant video encoding or if my method is too harsh. Did you have the opportunity to watch real splice points with real local ads insertions on your TV? Any video glitch?

This plugin is implemented in 3.7-500, commit a64a853ec99025dc030b494c3143ac0d713591b3. Would you have the opportunity to test it on some of your streams and give us your feedback?

alexander-nf, 2018-01-31:
Oh, I think splicing is a whole new world with a lot of nuances, technical aspects and specific cases. IMHO, of course...
However, I have some information.

1. Yes, there are ad spots between start/stop SCTE 35 messages, more exactly - between in/out splicing points in the TS sample. And (almost) everywhere there is preceding set of ad spots before each in-point. Obviously, these are nation-/network-wide targeted sets of ads.

2. I have an assumption that something like this was not taken into account:
> Splicing between AVC streams with differing GOP structures and/or buffer delays may place further requirements on the selection of splice points.

I mean, splicing result depends on a structure of both: a source TS and inserted TS. Also, probably you have to insert/redistribute PES-packets to the source TS-packets chain instead of replacing TS-packets to splice visually seamlessly.

I took the information and the ideas from [SCTE 67](http://www.scte.org/SCTEDocs/Standards/SCTE%2067%202017.pdf), 'Recommended Practice for Digital Program Insertion for Cable'.

Please see also these SCTE 67 paragraphs:
- 14.2.1. 'Splicer Technology'
- 14.2.3.1. 'Seamlessness'
- 14.2.3.4. 'MPEG-2 Compliance of Output Stream from Splicer'

(BTW, I've found in SCTE 67, 5.1. 'Example SCTE 35 Decoder' a reference to Java code of SCTE 35 decoder. The code is available on SCTE portal: [scte35-master.zip](http://www.scte.org/documents/standards/scte35-master.zip).)

Moreover, there are [SCTE 172](http://www.scte.org/SCTEDocs/Standards/ANSI_SCTE%20172%202017.pdf), 'Constraints On AVC and HEVC Structured Video Coding for Digital Program Insertion'.

To your question: no, there are no corrupted decoded frames after a local splicing, I saw it personally and significantly more than once. Unfortunately, there is no aired signal with the spliced TS in my location. But I'll ask distant friends/colleagues if there is an opportunity to capture the stream. If I get the TS, I'll share it for you.

3. It would be easier to automate recording process without a need to monitor TS services with the following approach. My thought is about an application to make records of a TS between in/out splicing points. For example, it could be TSDuck function which records a fragment of the TS to a disk:
- from and including the TS packet with SCTE 35 message with the `pts_time` for splicing start
- to the moment of a splicing end (`pts_time` from the message for splicing stop) + some _delta_time_ (seconds/minutes)
- with an option to set a guard interval (hours for example by default) to stop recording without stop SCTE 35 message.

And I think it is reasonable to stop recording after the last splicing point in a series. What I mean:
```
Service 1: ------CUEI---[out+++++++CUEI++in]-----------------------------
...
Service M: ----------------------CUEI---[out++++++++++CUEI++in]----------
Service N: -------------------------CUEI---[out++++CUEI++in]-------------

Recording: ------(Begin=======================================|====End)--
```

4. I'll test `rmsplice` next week or a little later and will give my feedback. I am away from my "lab" and facilities at the moment. But I hope you've got a pinch of useful information from this message.

lelegard, 2018-01-31:
Hi,

Thanks for the feedback and advice. I noted the following:
- Get a SpliceInsert command, type=out, pts=A.
- Get the same command one second or so later (to cope with a previous packet miss I assume).
- On the video PID, get a TS packet with the start of a new PES packet (PUSI=1) and pts=A in the PES header (exactly A).
- The plugin `rmsplice` drops this TS packet and all subsequent packets on the video PID.
- Later get the same scenario with type=in and exact pts seen at the start of a video PES packet.
- The plugin `rmsplice` restarts passing packets on the video PID, starting with the packet containing the exact pts of the splice-in event.
- Optionally, the plugin `rmsplice` can adjust all subsequent PCR, PTS and DTS values so that the clock appears continuous, without a big hole during the removed sequence.

I assume (but I may be wrong here), that the PES packet defines a GOP and starts with an intra-frame. So, I thought that removing entire GOP's (between two splice points) would preserve the video transition. But I am not a video encoding expert. So I may have missed something.

I also noted that the exact PTS value of the splice event is seen only on the video PID. On the audio and subtitles PID's, the exact PTS value is not found. But there is always a PTS value which slightly higher. The plugin `rmsplice` cuts the other PID's of the service at these slightly higher values. But the glitch is seen on video, so this is unrelated.

Note that when I mention a video glitch, I see it on VLC, viewing the recorded stream. Maybe a TV set or STB would absorb it in a cleaner way.

One thing astonishes me. A video encoder generates frames and packets on the fly. And PTS values are extremely precise (1 / 90 000 second). How can we predict in advance the value of a PTS? I mean that the SpliceInsert command contains a precise PTS value which will magically appear a few seconds later in the video stream. Of course, you may simply a convert a frame rate into a number of PTS units and predict all subsequent PTS values. But it seems a bit risky.

Concerning the recording strategy, I do not like much the idea of starting and stopping on cue events. As a TV viewer, we want to record an event (a film, a TV show). But cue events define some (or all) ads sequences. Many TV channels insert one, two or more ads sequences in the middle of a movie. So, generally speaking, there is no predictable relationship between a film or TV show and a splice in/out sequence.

There is already a plugin named `time` in TSDuck which schedules sequences to pass and drop based on absolute local time. I think that the best (or at least safest) way to proceed is to have a pipe of plugins:
- `time` to cut the whole stream around the event to record, with a safety margin based on the typical delay of the target channel.
- `zap` to extract the target channel and drop all others.
- `rmsplice` to cut all ads insertions in the middle of the program.

I wonder how TV operators use cue events, for which purpose? To insert ads (business-friendly, viewer-hostile) or to signal interesting TV contents, films and show (viewer-friendly, business-hostile)?

alexander-nf, 2018-02-05:
1. I'm sorry, I've certainly misunderstood the concept of `rmsplice`, it's because I haven't tested the new build yet. I assumed in my previous message that you decided to splice/replace stream packets between cue messages.

I've check the guide, version 3.8 for additional information. My opinion about the algorithm of `rmsplice` you described is: I am pretty sure it is not a 'seamless' solution in general just to cut out packets in a live stream or to replace packets with TS stuffing.

I think so even if there is a theoretical permission for a decoder to keep/freeze the last decoded frame on a screen/in a decompressed stream after the decoder picture buffer gets empty or loses synchronisation.

The multitude of hardware decoders (STBs/TV sets) and actual implementations have showed me, my colleagues and partners it is definitely better to feed decoders by a TS with uninterrupted sequence of video PESes with continuous timestamping. In common words, it's better to provide usual/typical conditions for the decoder input. In case of complete removal of video packets, PCR PID and synchronisation is lost.

I have no convincing arguments but I can't fully agree that the safe/seamless removal depends on the original video only:
> If the original video encoding is carefully performed to resist to identified splice points, the transition should be smooth.

It has to be true only when you play and decode rmspliced TS record after ads complete removal. But also, if timebase / synchronisation of the stream was corrupted / shifted, it would be abnormal conditions for the decoder in a such rmspliced point.

Ads complete extraction may be useful for stream-to-file, file-to-file operations without further broadcasting or etc. In other words, it's great for rough stream backup/selective logging/program capture.

Anyway, I'll try `rmsplice` and check for the `--adjust-time` and `--stuffing` options.

2. I agree with you and I think an encoder calculations are about frame rate.

And by the way, there is an indicator `auto_return_flag` in SCTE 35 `break_duration()` field. It says that the start cueing message has a duration time of an upcoming splice segment. There is no need for the encoder to generate stop cueing message in this case. The actual stream has:
- the single SCTE 35 message with `pts_time` and `break_duration`,
- out-point marked by PTS = `pts_time`,
- in-point marked by PTS = `pts_time` + `break_duration`.

```
Service 1: ------CUEI---[out+++++++++++++in]-----------------------------
```

A distance between the cueing message and the in-point is even greater than seconds - a couple of minutes, but the difference between in/out-point PTS values and the time interval between I/IDR-frames in the stream is exactly equal to `break_duration`.

It was informative message in addition to your words about the encoder behaviour.

But there will be an issue probably: will `rmsplice` finish packets removal after `break_duration` without stop cueing message?

3. Local and European practice I met is to mark:
- ad segments,
- to mark content that has to be blocked in some areas due to copyrights or etc,
- to mark content for regional/local/sub-local content (non-ads) replacement.
These cue messages carried `splice_insert` and `splice_time` commands with additional descriptors.

More complex use of SCTE 35 was found by me only in googled documents such as '[Сontent Contribtion Specification](https://wikileaks.org/sony/docs/07/Documents/@Deals/Comcast/Comcast-SP-CCS-I01-140206-ContentProvider.pdf)'. (I don't know if it is correct to post the document, but it is available on the 3rd page of Google request '[scte 35 content segmentation](https://www.google.ru/search?q=scte+35+content+segmentation&start=20)'.)
There is a chapter 9 'In-Band Metadata'. There are sorts of ads: network, provider, product placement, etc. It is also noted that SCTE 35 segmentation descriptors of program starts/ends, chapters/segments starts/ends, breakaways/runovers have to be used in-band as a part of linear channel programming.

And this is the context why I thought it is potentially useful to `rmsplice` content to a disk instead of /dev/null. I don't insist on rationality of my idea, it's quite theoretical task, of course.

The great news for me was: SCTE 35 sections detection and logging were added to TSDuck.
Thanks a lot!

lelegard, 2018-02-05:
Hi,

Thanks for the feedback.

To be precise, the plugin `rmsplice` was designed with disk recording in mind. That means that, with option `--adjust-time`, without `--stuffing`, you get a time-consistent SPTS file with all ads removed (at least those which are signaled using SCTE 35 SpliceInsert commands). This is the typical use case of a TV viewer who likes films and hates ads (me for instance:).

Using it on a live stream, you get a black or frozen screen during the removed sequences. So, this is not the most appealing use case.

And yes, the `auto_return_flag` is used. When set, the plugin `rmsplice` automatically schedules the splice-in command for the end of the ads sequence.

On the stream you provided (and which is BTW the only one I have with SCTE 35 signalling), I noticed the following:
- All splice-out commands have a `break_duration` part.
- The  `auto_return_flag` is not set in the `break_duration`. So, it is informative only.
- A splice-in command is actually sent at the end of the sequence, with a PTS exactly matching the announced `break_duration`.

So, in that context, `rmsplice` does not use the `break_duration` (`auto_return_flag` is not set) and relies on the next splice-in command.

alexander-nf, 2018-02-09:
Hi, Thierry!

I have two ideas about the glitch you've seen.

1. Sometimes I see a short splash/flicker while playing rmslpliced TS with software players (VLC, [MPC-HC](https://mpc-hc.org/)). It's a broadcaster issue: SCTE 35 marking inaccuracy. My version: you have probably seen the first decoded frame of the first ad spot in a sequence.

Service 1050 SPTS timeline before `rmsplice`:
```
SCTE 35 PTSes: --------------------|out-----------in|----------------------
                                   |                |
                                   V                V
Decoded video: -video-sequence-1--|----ad-spots----|video-sequence-2------
```

Service 1050 SPTS decoded video after `rmsplice`:
```
Decoded video: -video-sequence-1--|-|ideo-sequence-2-----
```

Yes, I've cut the letter "v" from "ideo-sequence-2", because the first frame of the sequence was rmspliced correctly too.

Before this I:
- cut from the start of the TS file that I sent you about a month ago 1 GB of data,
- processed new TS file using
```
tsp ^
   -I file S:\20180112_1129_UTC_Russian_mux1_T2MI_3PLPs_incl_SCTE35_in_PLP0_1GB.ts ^
   -P t2mi --pid 0x1000 --plp 0 ^
   -P zap 1050 ^
   -P rmsplice ^
   -O file rmspliced_1050.ts
```
- opened the new SPTS in MPC-HC and stepped over the first splice point using `Ctrl + Right Arrow`,
- took screenshots of the decoded frames around the splice point
[image]

2. In addition to the splash I feel very short freezing in the splice point while playing/decoding `rmspliced_1050.ts` in VLC. I've tried to fix it by eliminating PCR/PTS/DTS values shift:
```
tsp ^
   -I file S:\20180112_1129_UTC_Russian_mux1_T2MI_3PLPs_incl_SCTE35_in_PLP0_1GB.ts ^
   -P t2mi --pid 0x1000 --plp 0 ^
   -P zap 1050 ^
   -P rmsplice --adjust-time ^
   -O file rmspliced_1050_time_adjustment.ts
```
But nothing has been made. Time stamps in `rmspliced_1050_time_adjustment.ts` haven't been recalculated:
[image]
(PCR/PTS/DTS analysis by [DVB Inspector](http://www.digitalekabeltelevisie.nl/dvb_inspector/))

Did I miss something?

I use TSDuck from TSDuck-Win64-3.7-512.exe installer.

lelegard, 2018-02-10:
Hi Alexander

Thank you for the detailed analysis. This is very helpful.

I made my tests on service 1010 instead of 1050. I did the same tests again today and noticed a strange thing. On this service, the transition scene is as follow : two squirrels dancing on the roof of a train / removed sequence / a star on top of a tower or church.

In the `tsp` command with `rmsplice`, when I save the output to a file and play the file using VLC, the transition is ugly. But, if I run the same command with `-O play` instead of `-O file`, VLC displays a smooth transition!

In the first case, VLC reads a disk file with the ability to seek back and forth. In the second case, VLC is launched by `tsp` through the plugin `play` and it reads the TS from a non-seekable pipe. So, I wonder if VLC, detecting something strange in the video, tries to seek back and forth, looking for a recovery or whatever in the first case. In the second case, VLC cannot do anything and the result is visually just better.

I am going to investigate the time stamp discontinuity.

lelegard, 2018-02-10:
Hi Alexander,

I am a fool and an idiot !!!!!

Using a '+' sign instead of a '-' sign doubled the time gap instead of canceling it !!!!

This is the same old stupid bug. It is so easy to get trapped. Anyway, now using `--adjust-time` in `rmsplice`, the PCR, PTS and DTS no longer have discontinuities and, guess what, the video transition is smooth, whether you stream the output to `-O play` or replay a saved file. So, removing complete PES packets at exact scheduled splice PTS seems valid, provided that you adjust all time stamps.

Thank you very much for your deep analysis and the time you spent on a bug I created. Sorry for that.

Version 3.8 is almost ready and I plan to release it during the week-end, including this fix.

alexander-nf, 2018-02-12:
He-h!.. Self-criticism and rational perfectionism are good traits, but it seems to me you're exceedingly cruel to yourself! :) Especially in the context of great quality and complexity of the project you have done.

I've downloaded, installed and tested TSDuck-Win64-3.8-534.exe.

New rmspliced SPTSes with `--adjust-time` are played by VLC and MPC-HC smoothly, without freezing and visually seamless. And also now the inaccuracy of SCTE marking and the first frame of the first ad spot in 1050 are visually detectable during every playback of the stream.

Nice job again!
And you are welcome; the time I spent is a small part of the appreciation for the time you spent developing the shared project. Thanks, Thierry!

lelegard, 2018-02-12:
Hi Alexander,

About the inaccuracy of SCTE marking, I wonder how could this be fixed at the broadcaster. I even wonder if that SCTE 35 thing is really viable for live broadcast.

Let's look at the scenario:
- My system use 25 fps (or 30 if you like)
- So one frame has a PTS duration of 1/25 * 90 000
- I need to send SliceInsert commands N seconds in advance (say 5 seconds).
- So, to compute the PTS in the SliceInsert command, I need to predict extremely precisely the number of frames from the last one I generated (and for which I know the PTS) and the first frame of the ad. How can this be possible in practice? Without introducing a time shift and buffering images during the N seconds I mean. Which would be unacceptable for live events like sports.
- Additionally, if the encoder is not able to artificially restart a GOP / PES packet exactly at the first frame of the ad, the splice-out must be a the GOP-boundary preceding the first frame of the ad (and then losing the last frames of the previous program).

Really, the whole thing looks very difficult to implement correctly.

alexander-nf, 2018-02-19:
There are different ways to signal SCTE messages. It is possible to insert SCTE 35 commands into a stream directly on a coder using GPI, RS-232, IP links. But I've seen only over-SDI-signaling implementations in the uni-directional communication chains like:
> [image]

(the scheme is from P.S., please see it at the end of my message).

At the same time [SCTE 67](http://www.scte.org/SCTEDocs/Standards/SCTE%2067%202017.pdf) recommends this method also:
> While it is possible to use IP or serial streams to signal an encoder via SCTE 104, this RP recommends using the VANC approach. With this method the programmer has a fully described stream that can be sent to multiple consumers who all have the same frame accurate information.

Workflow:
- Broadcaster's automation system processes current playlist and produces TV program signal (uncompressed bit-stream over SDI). There are events in the playlist connected to ads/segments.
- The events are set a few seconds before splice/segmentation point.
- The automation system generates SCTE 104 command using the events and injects a message with the command into SDI bit-stream. Only one SCTE 104 message in one frame is allowed (according to SMPTE ST 2010).
- SCTE 104 command carries _relative_ splice/segmentation time pointer - `pre_roll_time` measured in milliseconds. A value of `pre_roll_time` divided by 40 ms (in case of 25 fps) produces the quantity of frames to splice after the frame with the command.
- Compression system (encoder) calculates PTS value for the pointed frame, puts the value to SCTE 35 `pts_time` and encodes the frame as I/IDR.

About sports, yes, it looks like inapplicable usage, but broadcasters use a few seconds graphics for transitions. So it is possible to send SCTE 104 message in the beginning of the graphics video with `pre_roll_time` equal or less than the video duration (and not less than 4 seconds due SCTE 104/35 restrictions).

P.S.
While checking some details I've found fresh, one year old overview, it's compact but quite informative: [SCTE-104/35 and Beyond: A Look at Ad Insertion in an OTT World](http://www.tvtechnology.com/expertise/0003/scte10435-and-beyond-a-look-at-ad-insertion-in-an-ott-world/280536).

There is pretty the same description in the first part of the article. And there is the last section named 'Frame accuracy' that says:
> with the right equipment, frame accuracy can be maintained end-to-end.

It is not a detailed description how accuracy is achieved. :) But I can confirm the mechanism of an accurate SCTE 35 marking works and is very stable. I took part in measurements a few years ago and we got absolutely stable result on hundreds/thousands measures in a row using the scheme I referred to in the beginning of the message.

lelegard, 2018-02-19:
Thank you for pointing to this article. It is very interesting since it presents the relationships between all elements and puts them all together. Although its main point is OTT, the first part applies to broadcast as well and is quite clear.

So, if I understand the workflow correctly, SCTE 104 commands are fully synchronous in the baseband signal. Each command is associated with the frame it is in, not a later frame as in SCTE 35 commands. The automation system, when it switches back and forth between the main content and ads, knows exactly which frame starts a new scene in a new content. Then, the encoder must implement __a certain amount of time shift__ to insert SCTE 35 commands in the output stream, a few seconds before the frame to splice.

Am I right?

I noted the following point:
>  The splicer will also typically filter out the markers (so as not to facilitate the work of “commercial killers”). 

So, they want to defeat plugin `rmsplice` :-) I can understand that. But if the distribution feed (your capture of the T2-MI stream I assume) is accessible through satellite and in the clear, it is easy to work on it.

Thanks for all these information.

alexander-nf, 2018-02-20:
Hi Thierry,

Possibly I’m a bit lost in translation or did not describe the workflow thoroughly. I suppose you’re almost right except SCTE 104 full synchronization with the signal (_= video sequence?_). Yes, the commands are synchronized with the signal, but each command is injected in advance, `pre_roll_time / 40 ms` frames before the first spliced frame.

I’ve tried to explain it on the diagram:

```
                    1                   2        3                   4
Automation events: -|-------------------|--------|-------------------|---

Video sequence: ------------------------|===ad=spots===/===segment===|---
                                        ^                            ^
                                        |                            |
                    |<--pre_roll_time-->|        |<--pre_roll_time-->|
SCTE 104 triggers: -|out-----------------------in|-----------------------
                    |                            |
                    V (VANC injection)           V (VANC injection)
SDI bitstream: -----|-------------------|========|===================|---
```
Automations system playlist events:
1. Triggering SDI injector of the SCTE 104 message which carries `pre_roll_time` of the out-point.
2. Changing video source of playout system to the national commercials video storage or to a new source of video content.
3. The same as “1”, in-point triggering.
4. The same as “2”, changing video source.

lelegard, 2018-02-20:
My fault, I misunderstood the explanations. I understood that the pre_roll_time was an _information_ which was sent by the automation system inside an SCTE 104 command in the frame to splice in/out and later used by the encoder to send an SCTE 35 in advance. Now I understand that the SCTE 104 command is sent with the same advance, making the time shift useless in the encoder.

So, my initial concern remains about the prediction of the splice time. Take the example of a live TV show, followed by ads. When the automation system decides to send a splice out trigger with a pre_roll_time which is configured to the equivalent of, say, 50 frames (2 s), that means that the automation system knows exactly when there are exactly 50 remaining frames in the current TV show. Or alternatively, does the automation system _decide to cut off the TV show in 50 frames from now_, whatever it happens?

I probably focus on small details I am not concerned about. But, as usual, the devil is in the details...

closed 2018-03-10 by lelegard

delgemoon, 2018-03-30:
Hi @lelegard 
I cannot download 20180112_1129_UTC_Russian_mux1_T2MI_3PLPs_incl_SCTE35_in_PLP0.ts . Can you help me on this.
Thanks,
Tam

lelegard, 2018-03-30:
Hi,

This was a huge multi-gigabytes file. That kind of file cannot stay forever online. They are typically exchanged using « file transfer » services, not « file hosting ».

You may kindly request the original provider, @alexander-nf, to repost it. But this can be a lot of work for him. Uploading it without a good fiber link is a nightmare.

This file was essential in the process of developing SCTE 35 support. So, this was a good incentive for the original upload effort. 

Maybe someone with a dish pointing to the same russian satellite could provide a new - possibly smaller - capture.

delgemoon, 2018-03-30:
Hi Thierry,
Thanks for your reply. I see
Tam

lelegard, 2018-05-10:
Hi @delgemoon 

You will find the file [here](https://tsduck.io/streams/streams.php?name=russia-t2mi-scte35)

delgemoon, 2018-05-13:
Hi Thierry
Thank you so much :)
Tam

BartmanEH, 2021-01-19:
this is a wonderful thread on SCTE 35 analysis using TS Duck. I've dabbled with it a bit and can indeed extract crude info from a DVB TS. I have a few suggestions:
- "Duration PTS:" ideally there would be an option to show this value divided by the primary clock, usually 90 kHz, which would give a duration value in seconds. For now I'm doing the math manually.
- I can find no indication of preroll timing. There should be two PTS values: a Cmd time and a Splice time with the difference between the two being Arm time (AKA Preroll). Could this be shown somehow?

lelegard, 2021-01-20:
> - "Duration PTS:" ideally there would be an option to show this value divided by the primary clock
> - Splice event id: a hex value. please add a decimal equivalent

Could be easily done, will do...

> I can find no indication of preroll timing. There should be two PTS values: a Cmd time and a Splice time with the difference between the two being Arm time (AKA Preroll). Could this be shown somehow?

AFAIK, this information does not exist in the splice command which is extracted from the stream. This is an operational command which is sent to the inserter equipment so that the splice command can be inserted on time. In the TSDuck plugin `spliceinject`, this is implemented using options `--inject-interval` and `--inject-count`.  What you call the arm time is the former times the latter.

lelegard, 2021-01-20:
> As it is, I need to run -analyze to figure out what the SCTE35 PIDs are and rerun tsp with the "-P tables --pid (value)". Maybe there's a better way currently but some more automation for this would be great.

You can skip the first phase. You can simply use `-P --tid 0xFC`. The plugin examines all PID's and selects only the splice information tables (table id 0xFC). If you add option `--packet-index`, you can the TS packet number of each table. Given the bitrate, you can derive the duration between the tables.

TD-335, 2021-02-11:
Hi, i am looking to analyze a HLS TS segments to see if it's carrying SCTE35 data.  The script i have setup with TSDuck is as follows 
'tsp -I hls http://livestream.com/index.m3u8 \
    -P t2mi --pid 0x1000 --plp 0 \
    -P tables --tid 0x0FC \
              --text plp0.splice-info-hlstables.txt \
              --xml plp0.splice-info-hlstables.xml \
    -O drop
'
I am not getting any data on the output but I am not entirely sure if i am using tsp properly.  Any advice would be appreciated.

lelegard, 2021-02-11:
Hi @TD-335,

This is an old and closed thread. You would get more chance to get an answer by opening a new one.

Why are you using the `t2mi` plugin? This reminds me of some Russian stream in the TSDuck repository. Maybe you copied the command from there. In the Russian case, the input TS carried a T2-MI stream. The SCTE-35 tables were in the encapsulated stream inside the T2-MI tunnel. The purpose of the `t2mi` plugin is to extract the encapsulated T2-MI TS and replace the input TS with the decapsulated one.

If your stream is not a T2-MI stream, and I suspect it is not, simply remove the `t2mi` plugin. 

If your address is really http://livestream.com/index.m3u8, then it is another problem since it returns an HTML page saying "not found".