tsduck/tsduck #22: ip input has problem when specifying multicast address

tim-impulsescreen, 2018-01-11:
My setup has 5 x UDP MPEGTS streams going out from a single machine, each using a different multicast address:
- stream1 => 239.255.1.1 port 59000
- stream2 => 239.255.1.2 port 59000
- stream3 => 239.255.1.3 port 59000
- stream4 => 239.255.1.4 port 59000
- stream5 => 239.255.1.5 port 59000 

On another machine, when I try to analyze a single stream I seem to get information from the other streams. Such as...
`% tsp -I ip 239.255.1.5:59000 -P analyze -O drop`
 will return service and table info for stream1 and stream2.

If I try to zap a single service in stream1 (one that gets returned in the above analysis) I will get the following:
`% tsp -I ip 239.255.1.5:59000 -P zap 1376 -O drop`
`* Error: zap: service id 0x0560 not found in PAT`

If I stop sending the other streams (2, 3, 4 & 5) both the analyze and the zap plugins work as expected. Analysis contains only stream1 information and zap is able to find the given service.

If I move each stream to its own unique port: 
- stream1 => 239.255.1.1 port 59001
- stream2 => 239.255.1.2 port 59002
- stream3 => 239.255.1.3 port 59003
- stream4 => 239.255.1.4 port 59004
- stream5 => 239.255.1.5 port 59005 
I get expected results from both plugins.
`% tsp -I ip 239.255.1.1:59001 -P analyze -O drop`
will return only stream1 information and
`% tsp -I ip 239.255.1.1:59001 -P zap 1376 -O drop`
is able to find the service 1376.

The **ip** input is listening to more than 1 multicast destination - more than just the one I explicitly configure.

lelegard, 2018-01-11:
Hi Tim,

Thanks for the report. It seems to be a bug. I will fix it.

That being said, using the same UDP port for traffic on different multicast addresses is looking for trouble on the receiving side. There are many problem reports on that type of configuration. The Wikipedia article named [Promiscuous Traffic](https://en.wikipedia.org/wiki/Promiscuous_traffic) describes some of them.

Using the same destination port, a receiving application which binds to that port may accidentally receive any UDP packet with that destination port reaching the receiving machine, including unicast packets for that machine on the same port by the way. Being adjacent, your different multicast IP addresses map to distinct multicast MAC addresses (multicast MAC address collisions appear only in the upper 9 bits of the IP address). So, if there is only one receiving application on the system, listening to one single multicast address, it should work.

But the Web is full of problem reports which depend on the Linux kernel version or Windows version. What kind of system are you using on the receiving side?

As you noted, using distinct ports eliminates the problem. Beyond that specific issue with TSDuck, this is a recommended configuration anyway.

Now, there is a bug with TSDuck. When the `ip` plugins joins the multicast group and binds to the specified port, it assumes that all UDP packets come from the right traffic. A safe "belt and braces" check would be to verify the destination address of each UDP packet and ignore those with a destination which is different from the expected multicast address. I will make that fix in the future.

However, with multiple traffic on the same port, the fix will work correctly only it there is only one receiving application on the same system. If two instances of `tsp` are running on the same system, listening to distinct streams on the same port, it may work or fail. Each system and version of system has its own specific set of socket options to address this situation. I will try to do my best to provide the best support for each system when possible. But this configuration remains a source of potential trouble.

tim-impulsescreen, 2018-01-11:
Thierry,
Thanks for the fast response!

I will look into the cross-talking issue - it might make sense for me to move the senders each to a unique port number. Previously, I have used VLC, FFMPEG and FFPROBE to do some of the analysis and capture tasks and they seem to be fine with the multicast traffic on different destination addresses but the same port.

I am using a Linux host running Ubuntu 16.04 LTS. I have also been using Docker (version 17.04.0-ce) containers with TSDuck installed in some situations - but also based on Ubuntu 16.04 LTS.

```
$ uname -a
Linux ISMPER02 3.13.0-88-generic #135-Ubuntu SMP Wed Jun 8 21:10:42 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
$ docker --version
Docker version 17.04.0-ce, build 4845c56
$ tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.4-20171110
```

Let me know if I can be of any assistance.

lelegard, 2018-01-11:
Hi,

I assume that vlc and ffmpeg implement their own filtering or use specific system options. This is what I will implement in tsduck also.

I will let you know when I implement this for you to test it.

lelegard, 2018-01-16:
Hi @tim-impulsescreen 

To reproduce this issue on Linux, I need to have _two_ simultaneous applications listening on two distinct multicast addresses, same port. In that case, the two applications receive all packets from both streams at the same time.

If only one application is listening on the machine, it receives only its traffic. When I start a second receiver on a second multicast address, the first application starts to receive the second traffic as well. And the second application also receives both traffics. And when I stop the second application, the first application goes back to its previous state, receiving only its own traffic.

This situation is absolutely logical, regarding the way multicast MAC addresses are configured during join & drop group membership. This is the typical "promiscuous traffic" situation I mentioned in my first comment.

I will add a filtering on the destination multicast address in the ip plugin. Thus, even though the plugin potentially receives traffic from distinct streams, it will pass only the required traffic to the next plugin. This will solve the issue. I assume that vlc and ffmpeg implement a similar filtering mechanism.

Could you confirm that, when you listen to address A and receive streams from addresses A and B, you have another application on the same system listening to address B ? Maybe not TSDuck, any application, vlc, ffmpeg or whatever.

In all cases, I confirm what I wrote in my first comment. This configuration (distinct multicast addresses, same port) is really looking for trouble. After implementing the explicit filtering in the application, the feature will work ok. But each application receives and processes twice as many UDP packets as necessary (or three times for three addresses and simultaneous applications). The waste of resource (kernel buffers, CPU time, etc) is exponential with the number of applications. None of this would occur with distinct ports.

I will update this thread when a fix is available for you to test. Very soon anyway. Will you have the possibility to pull the repository and recompile TSDuck or do you need a binary package?

lelegard, 2018-01-16:
Also note that the problem does not exist on Windows (at least 7 and later). Even when two applications are listening simultaneously on two distinct multicast addresses, same port, each application receives exactly its own traffic, nothing else. I assume that Windows routes the packets to the applications sockets based on the multicast group they explicitly joined and nothing more.

I cannot decide which behavior is supposed to be correct and which one is buggy. Let's say that, here, Linux is not application-friendly because you ask for something and you get additional garbage in the middle of what you requested.

lelegard, 2018-01-17:
Hi @tim-impulsescreen 

The fix has been implemented in version 3.6-476 (commit 7ea0349dd618525a5549116c917db30cb1dd135f). The `ip` plugin now correctly filters UDP packets based on their destination address and only keeps packets with the expected multicast address.

Could you test it in your configuration? Do you need a binary package or can you pull the repo and rebuild TSDuck?

tim-impulsescreen, 2018-01-19:
Hi Thierry,

I don't have a CPP development environment so a binary dist would be ideal
for me. Are there nightly builds available anywhere?

Regards,
Tim.

[image: Impulse Screen Media] <http://www.impulsescreen.com/>
Tim Duncan
Software Architect
m:+61 412 222 800 <+61+412+222+800>
a:Suite 301, 123 Walker Street
  North Sydney NSW 2060
w:www.impulsescreen.com
e:tim.duncan@impulsescreen.com

lelegard, 2018-01-19:
Hi Tim,

I do not have the infrastructure to publish nightly builds. I have attached [tsduck_3.6-485_ubuntu.zip](https://github.com/tsduck/tsduck/files/1645867/tsduck_3.6-485_ubuntu.zip) which contains an Ubuntu package. It was built on Ubuntu 17.10. If the previous builds worked on your Ubuntu 16.04, this one should work too.

Let us know if this works.

In case of problem, TSDuck is easy to rebuild on Linux. See [details here](https://tsduck.github.io/doxy/html/building.html).

In short: On Ubuntu, install the prerequisites:
~~~
apt-get update
apt-get install git g++ curl pcscd libpcsclite-dev libcppunit-dev libcurl3 libcurl3-dev
~~~
Then, to build TSDuck:
~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
make deb
~~~
You can collect the `.deb` packages in the `installers` subdirectory.

lelegard, 2018-01-20:
Hi @tim-impulsescreen 
I just published the version 3.6-491. See the release section.

lelegard, 2018-01-29:
Hi @tim-impulsescreen 

Can you confirm that the problem is fixed in 3.6?

lelegard, 2018-02-07:
Closing as the fix was implemented

closed 2018-02-07 by lelegard