tsduck/tsduck #23: Raspberry PI / ARM compiling error

alexander-nf, 2018-01-11:
Advise 

> replace const `uint16_t& proxyPort()` with `uint16_t proxyPort()` in line 262 of tsWebRequest.h. 

 how to solve new [error](https://github.com/tsduck/tsduck/issues/17#issuecomment-356853749) from #17 doesn't work on my system. I've got new error message:

```
  [CXX] tsGitHubRelease.cpp
In file included from tsGitHubRelease.cpp:48:0:
tsWebRequest.h:262:36: error: type qualifiers ignored on function return type [-Werror=ignored-qualifiers]
         const uint16_t proxyPort() const { return _proxyPort == 0 ? _defaultProxyPort : _proxyPort; }
                                    ^
cc1plus: all warnings being treated as errors
../../Makefile.common:806: recipe for target 'release-arm/tsGitHubRelease.o' failed
make[2]: *** [release-arm/tsGitHubRelease.o] Error 1
```

GCC version:
```
$ gcc --version
gcc (Raspbian 4.9.2-10) 4.9.2
```
The distro:
```
$ cat /etc/debian_version
8.0

$ cat /etc/issue.net
Raspbian GNU/Linux 8

$ cat /etc/rpi-issue
Raspberry Pi reference 2017-07-05
```

This is an original Rasbian based on Debian 8 (Jessie).

lelegard, 2018-01-11:
Ah, ah ...  ;)

Look at the differences:

> replace `const uint16_t& proxyPort()` with `uint16_t proxyPort()` in line 262 of tsWebRequest.h. 

And:
> ```
> tsWebRequest.h:262:36: error: type qualifiers ignored on function return type [-Werror=ignored-qualifiers]
>          const uint16_t proxyPort() const { return _proxyPort == 0 ? _defaultProxyPort : _proxyPort; }
> ```

The _"type qualifiers ignored on function return type"_ is due to the extra `const` that should have been removed at the beginning of the line. The returned type was `const uint16_t&` and should be changed to `uint16_t`, not `const uint16_t`.

alexander-nf, 2018-01-11:
I see!.. :+1:
What a shame. :) I'm sorry!

I've just compiled, installed and ran TSDuck on Raspberry. No errors. There was no issue.

closed 2018-01-11 by lelegard

lelegard, 2018-03-08:
Hi @alexander-nf 

I just want to let you know that I have purchased a Raspberry Pi 3 (nice cheap little beast...) I will now anticipate ARM issues. Starting with the next version of TSDuck, I will release binaries for Raspbian / ARM.