tsduck/tsduck #24: Looking for feedbacks on DVB tuners - Linux or Windows

lelegard, 2018-01-11:
Hello TSDuck users,

You may have noticed that the TSDuck user's guide contains a chapter on hardware device support. Specifically, there is a table of "Tested DVB receiver devices".

The number of tuners available on the market is growing every year. It is virtually impossible to test them all. And no one owns them all anyway ! But choosing a tuner is a delicate matter. The level of support depends on the availability of drivers.
- Most tuners come with a Windows driver. But DirectShow, the framework which manipulates tuners on Windows, is complicated and there is no unique standard way to use a tuner from an application.
- On Linux, the situation is reversed. LinuxTV, the equivalent framework, is simple and robust with a unique way to handle tuners. But very few tuners come with a Linux driver. The [LinuxTV community](https://www.linuxtv.org/wiki/) is doing its best to develop them (usually with no information at all on the hardware!) but, like any voluntary effort, this work is incomplete. Some tuners have no driver or the provided driver does not work very well.

So it seems useful to build a list of tested devices which proved to work correctly with TSDuck. This must be a collaborative effort. This issue will remain open to allow anyone to provide feedback.

Please enter a comment in this thread if you have a tuner which works with TSDuck and which is not listed in the table from the TSDuck user's guide or if your operating system is marked as "not tested" for this device.

Please provide all required information for the table:
- Brand
- Model
- Modulation support (DVB-T, DVB-S, etc.)
- Bus (USB, PCIe, etc.)
- Tested systems (Linux, Windows)
- Output of `tslsdvb`
- Any additional remark you may find useful

I will update the TSDuck user's guide at regular intervals with the provided information.

Thank you all in advance for your cooperation.
-Thierry Lelegard, author of TSDuck

PS: I am also interested in any DVB-T tuner which works on MacOS with a fully documented API. I could not find any for now but I do not give up hope :)

alexander-nf, 2018-01-25:
I've got a few additional notes about devices in the guide (from Table 5: Tested DVB receiver devices):

- **GoTView MasterHD3** / **TeVii H640** / **MaxMedia HU372** with high probability are the same devices made by the one OEM manufacturer. Some users on the internet report that the devices have the same hardware, identical PCBs inside the USB sticks.

- **GoTView MasterHD3** has good support in Linux kernel releases since 4.2 up to 4.7, two frontends are available: `frontend0` is DVB-T, `frontend1` is DVB-T2/DVB-C.
The support in kernels after version 4.7 is partial, something was broken. The problem was met on ARMv7 and x86_64 platforms. The device starts but only with the one (DVB-T) frontend. The second frontend (Si2168 demodulator for DVB-T2 and DVB-C) doesn't start due to i2c error. 
Therefore TSDuck works only with one available frontend. See the other and older information about the device that could help to fix the problem in LinuxTV wiki.

- **Pinnacle PCTV nanoStick T2 290e** has kernel support for about seven years and the device has been successfully tested on Linux kernels 3.0+, no patches needed. TSDuck works with nanoStick fine (and also locks chosen PLP of multiple PLP DVB-T2 signal).

lelegard, 2018-01-25:
Thanks for the feedback. I will update the documentation accordingly.

joolzg, 2018-11-13:
TurboSight TBS 6704 ATSC frontend under Linux

encoder@joemex3:~$ tsp -v -I dvb --adapter 0 --frequency 479000000  -P until -s 10 -P analyze -O drop
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* dvb: using /dev/dvb/adapter0 (ATSC)
* dvb: tuned to transponder --frequency 479000000 --modulation 8-VSB --spectral-inversion auto
* dvb: input bitrate is 19,392,732 b/s

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ...... 30011 (0x753B)  |  Services: .............. 2  |
|  Bytes: ........................ 32,569,308  |  PID's: Total: ......... 13  |
|  TS packets: ...................... 173,241  |         Clear: ......... 13  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............. 769  |         With PCR's: ..... 2  |
|     Suspect and ignored: ................ 1  |         Unreferenced: ... 5  |

joolzg, 2018-11-13:
TurboSight TBS 6904 DVBS/S2 frontend under linux
===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ......... 15 (0x000F)  |  Services: ............. 30  |
|  Bytes: ........................ 48,594,428  |  PID's: Total: ......... 91  |
|  TS packets: ...................... 258,481  |         Clear: ......... 49  |
|     With invalid sync: .................. 0  |         Scrambled: ..... 42  |
|     With transport error: ............... 1  |         With PCR's: .... 19  |
|     Suspect and ignored: ................ 0  |         Unreferenced: .. 10  |

Works good

oviano, 2019-02-02:
Sorry I don't know why the formatting is like that, I just copied and pasted plain text and I can't seem to turn off that bold and sizing. When I edit, it doesn't appear like that.

lelegard, 2019-02-02:
> Sorry I don't know why the formatting is like that

I just fixed it. GitGHub markdown is [documented here](https://help.github.com/articles/basic-writing-and-formatting-syntax/).

> I think the bottom line is that this device, Hauppauge WinTV-soloHD DVBT Tuner (DVB-T), is working ok with TSDuck.

Thanks a lot. Is this on Windows or Linux?

oviano, 2019-02-02:
Thank you - yes that is on Windows 10 Pro 64-bit Release 1809.

lelegard, 2019-02-02:
Could you please post the output of `tslsdvb`? I am curious if DVB-T/T2 and DVB-C are seen as two tuners or one.

oviano, 2019-02-02:
It's seen as one:

>tslsdvb
0: "Hauppauge WinTV-soloHD DVBT Tuner" (DVB-T)

>

oviano, 2019-02-02:
The other feedback I have is that tsscan took nearly 2 hours to complete, whereas a channel scan with the supplied WinTV software took far less time (maybe 10 minutes).

I don't know if that is significant or expected, but I thought I should mention it.

lelegard, 2019-02-02:
It depends on which type of scan you select. On DVB-T, the default UHF-band scan tries all channels from 21 to 69 and then scans all offsets around each channel. If you do not care about offsets, use option `--no-offset` to speed up the process. The option `--verbose` displays which channels and offsets are scanned.

oviano, 2019-02-02:
Ok, maybe that's the reason then. I'm going too read up on what the offsets mean. I only started trying to learn how these capture cards work a couple of days ago!

oviano, 2019-02-02:
Actually there was one other thing relating to the tsscan - it's very hard to get it to respond to control+c to interrupt it. I've only succeeded once or twice. It seems to lock up the command prompt - once I had to reboot as I couldn't even force quit the command prompt.

skunge2000, 2019-02-03:
Here's the output for the Raspberry Pi Foundation TV uHAT.

This a GBP £20 DVB-T/T2 tuner in a new uHat format about the same footprint as a Raspberry Pi Zero. 
It connects to the Pi via the 40-pin GPIO connector, and uses SPI to send the transport stream from the demodulator to the Pi.  I2C is used - as with all HATs - to signal the presence of the uHAT to the host computer.

The Linux drivers appear to have had a lot of input from Sony (Sony manufacture Raspberry Pis in the UK) according to the Linux TV mailing lists.

`
/dev/dvb/adapter0 (Sony CXD2880, DVB-T, DVB-T2)

tslsdvb: ioctl FE_READ_BER on /dev/dvb/adapter0/frontend0: Resource temporarily unavailable
tslsdvb: ioctl FE_READ_SNR on /dev/dvb/adapter0/frontend0: Resource temporarily unavailable
tslsdvb: ioctl FE_READ_SIGNAL_STRENGTH on /dev/dvb/adapter0/frontend0: Resource temporarily unavailable
tslsdvb: ioctl FE_READ_UNCORRECTED_BLOCKS on /dev/dvb/adapter0/frontend0: Resource temporarily unavailable
  Status: 

  Bit error rate ................................. 0 (0%)
  Signal/noise ratio ............................. 0 (0%)
  Signal strength ................................ 0 (0%)
  Uncorrected blocks ............................. 0 
  Frequencies:
    Current ...................................... 0 Hz
    Min ................................ 174,000,000 Hz
    Max ................................ 862,000,000 Hz
    Step ..................................... 1,000 Hz
    Tolerance .................................... 0 Hz
  Spectral inversion ........................... off 
  Bandwidth ................................... auto 
  FEC(high priority) .......................... none 
  FEC(low priority) ........................... none 
  Constellation ............................... QPSK 
  Transmission mode ............................. 2K 
  Guard interval .............................. 1/32 
  Hierarchy ................................... none 

  Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 4/5, FEC 5/6,
    FEC 7/8, FEC auto, QPSK, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM,
    QAM auto, transmission mode auto, guard interval auto, 2nd generation,
    recover, mute TS`

Is the verbose output of tslsdvb when the tuner is idle.

samuraibrian, 2019-03-06:
'Generic' device, allegedly "DVB-T/T2/C+FM+DAB".
Working ok with tsp in Ubuntu 18.04 on UK DVB-T muxes, but unable to lock to DVB-T2 muxes:

tslsdvb: /dev/dvb/adapter0 (Realtek RTL2832 (DVB-T), DVB-T)
Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8, FEC auto, QPSK, 16-QAM, 64-QAM, QAM auto, transmission mode auto, guard interval auto, hierarchy auto, recover, mute TS

- which would explain why it can't see the (256-QAM) UK DVB-T2 muxes!

oviano, 2019-03-14:
PCTV 292e/461e BDA 28179 TVTuner S

This one seems to work on Windows.

[tslsdvb.txt](https://github.com/tsduck/tsduck/files/2967592/tslsdvb.txt)

[astra_10847000000_vertical.txt](https://github.com/tsduck/tsduck/files/2967594/astra_10847000000_vertical.txt)

lelegard, 2019-03-14:
> PCTV 292e/461e BDA 28179 TVTuner S ("WinTV Nova-S2 - Model 01677" on Amazon UK)

Are you sure? WinTV Nova are Hauppauge devices while PCTV (formerly Pinnacle) is another brand.

I used to have one PCTV 461e stick but I dropped it. It worked on Windows only (unreliable behaviour on Linux) and it dropped all null packets. As a consequence, the captured stream is missing the stuffing PID, computed bitrates are incorrect and it is not possible to cleanly reprocess and reinject the stream in a modulator. In your capture, I can see that the stuffing PID is missing as well. I also have a recent DVB-T2 tuner from PCTV with the same issue (no stuffing PID). So, I assume that this is an engineering choice from PCTV. It is true that we do not need the stuffing PID to watch TV but we, the DTV engineers, need it.

oviano, 2019-03-14:
Yes I’m sure, it came in a Hauppauge box and instructions to install WinTV.

That is the name that shows up in tslsdvb.

Perhaps Hauppauge acquires them or something.

oviano, 2019-03-14:
“Hauppauge Computer Works is a division of Hauppauge Digital Inc. In December of 2008, Hauppauge Digital Inc. acquired the Pinnacle PCTV product line from Avid, and now runs PCTV tuner products as a separate division.”

http://www.hauppauge.com/pages/about/about.html

lelegard, 2019-03-14:
You are probably right about the acquisition. PCTV products are now described on the Hauppauge site : http://www.hauppauge.co.uk/pctv/en/prods.html

But, on this page, I see this:
[image]

So, PCTV 292e and 461e look like different products. Which one is yours? Mine is the bottom one, 461e. In all cases, yours and mine seem unable to get the stuffing PID.

oviano, 2019-03-14:
Mine is this one

http://www.hauppauge.co.uk/site/products/data_novas2.html

lelegard, 2020-04-05:
First tested ISDB-T tuners: A user from Japan reported that the following two tuners work on Linux and Windows: MyGica S270 and PLEX PX-S1UD. They are internally identical.

piotrserafin, 2020-04-12:
Brand: **TBS**
Model:  **TBS5580**
Modulation support (DVB-T, DVB-S, etc.): 
**DVB-T/T2/C/C2/ISDB-T, DVB-T, DVB-T2, ISDB-T, DVB-C,DVB-S/S2/S2X, DVB-S, DVB-S2, DSS**
Bus (USB, PCIe, etc.): **USB**
Tested systems (Linux, Windows): **Linux, Windows, Raspberry PI 3+ (Raspabian)**

Output of tslsdvb:

_For Frontend 0:_
```
tslsdvb -d /dev/dvb/adapter0:0 -v
/dev/dvb/adapter0 (TBS 5580 CI USB2.0 DVB-T/T2/C/C2/ISDB-T, DVB-T, DVB-T2, ISDB-T, DVB-C, 2)

tslsdvb: error reading status on /dev/dvb/adapter0/frontend0: Resource temporarily unavailable
  Delivery system ............................ DVB-T
  Frequencies:
    Current ............................ 498,000,000 Hz
    UHF channel ................................. 24
    Min ................................ 174,000,000 Hz
    Max ................................ 862,000,000 Hz
    Step ................................... 250,000 Hz
    Tolerance .................................... 0 Hz
  Spectral inversion .......................... auto
  Modulation ................................ 64-QAM
  Bandwidth .................................. 8-MHz
  FEC(high priority) .......................... auto
  FEC(low priority) ........................... auto
  Transmission mode ............................. 8K
  Guard interval .............................. 1/32
  Hierarchy ................................... none

  Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8,
    FEC auto, QPSK, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM, QAM auto,
    transmission mode auto, guard interval auto, hierarchy auto, multistream,
    2nd generation, mute TS
```

_For Frontend 1:_
```
tslsdvb -d /dev/dvb/adapter0:1 -v
/dev/dvb/adapter0:1 (TBS 5580 CI USB2.0 DVB-S/S2/S2X, DVB-S, DVB-S2, DSS)

tslsdvb: error reading status on /dev/dvb/adapter0/frontend1: Resource temporarily unavailable
  Delivery system ............................ DVB-S
  Frequencies:
    Min ................................ 850,000,000 Hz
    Max .............................. 2,300,000,000 Hz
    Step ......................................... 0 Hz
    Tolerance .................................... 0 Hz
  Symbol rates:
    Min .................................. 1,000,000 sym/s
    Max ................................. 45,000,000 sym/s
    Tolerance .................................... 0 sym/s
  Spectral inversion ........................... off
  FEC(inner) .................................. auto
  Modulation .................................. QPSK

  Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8,
    FEC auto, QPSK, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM, QAM auto,
    transmission mode auto, guard interval auto, hierarchy auto, multistream,
    2nd generation, mute TS
```

lelegard, 2020-04-13:
> Model: TBS5580
Modulation support (DVB-T, DVB-S, etc.):
DVB-T/T2/C/C2/ISDB-T, DVB-T, DVB-T2, ISDB-T, DVB-C,DVB-S/S2/S2X, DVB-S, DVB-S2, DSS

Wow, that is quite a beast ! 

I couldn't find it yet on Amazon but, for $120 on TBS site, it looks quite a good investment. Is it a new device? Are you satisfied with it?

piotr-serafin-epam, 2020-04-13:
Depends on the use case, currently I'm using it with Raspberry PI 3B+ (with headless Raspbian Lite installed) as a streaming/dev server. I'm playing a lot with TS manipulation. 

For Windows 10 it is working fine but not checked with high number of DVB Players/Software, working definitely with TSDuck.

For Linux, checked on Debian based systems (mainly Ubuntu and it's flavors) and on Arch, there was a lack of driver support for newer kernels, and still installation requires some manual work (compiling, moving FW binaries) but it works fine. Checked with TSDuck as well:)

Raspberry PI was a pain, there was no driver support for ARM targets but at the beginning of 2020 they shared on TBS forum new drivers archive with support for ARM. It is working but it seems that driver has an issue with signal strength for DVB-T FE. On My TV sets it show >50%  on dvb-fe-tool, and tsscan it shows <10% do there are some issues with this driver. 

From mechanical perspective, the device is very soli, enclosed in metal casing. 

It can get quite hot when used but entire case is big radiator.

EricBerendsen, 2020-07-18:
Got some tuners that work with TSDuck;

All tested using Windows 10, all using USB.
 
### Brand: Technotrend
### Model: TT-connect CT-3650 CI

Modulation: DVB-C, DVB-T 

Works with DVB-C, not tested with DVB-T (not available any more). Does not work with DVB-T2 (not to be expected for a device from 2008).  tslsdvb incorrectly says DVB-T2 

tslsdvb output:
0: "USB 2.0 BDA DVB-T Tuner" (DVB-T, DVB-T2)
1: "USB 2.0 BDA DVB-C Tuner" (ATSC, DVB-C, DVB-C/C)

### Brand: Hauppage
### Model: WinTV-HVR-935

Modulation: DVB-C, DVB-T2

tslsdvb output:
0: "Hauppauge Cx23100 BDA DVBC Tuner" (ATSC, DVB-C, DVB-C/C)
1: "Hauppauge Cx23100 BDA DVBT Tuner" (DVB-T, DVB-T2)

### Brand: Hauppauge
### Model: WinTV-dualHD

Modulation: DVB-C, DVB-T2

Warning: This tuner discards all null packets (PID 0x1FFF), for both DVB-C and DVB-T2. So not to be used for bitrate measurements etc.

tslsdvb output:
0: "Hauppauge WinTV-dualHD DVBC Tuner" (ATSC, DVB-C, DVB-C/C)
1: "Hauppauge WinTV-dualHD DVBT Tuner 2" (DVB-T, DVB-T2)
2: "Hauppauge WinTV-dualHD DVBT Tuner" (DVB-T, DVB-T2)
3: "Hauppauge WinTV-dualHD DVBC Tuner 2" (ATSC, DVB-C, DVB-C/C)

lelegard, 2020-08-11:
@EricBerendsen, thanks for your feedback.

>  Does not work with DVB-T2 (not to be expected for a device from 2008). tslsdvb incorrectly says DVB-T2

This is unfortunately a limitation of Windows. DirectShow is, as far as I know, unable to report specific DVB-T2 support in addition to DVB-T. So, if DirectShow reports DVB-T support, TSDuck [assumes DVB-T2 as well](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/windows/tsDirectShowNetworkType.cpp#L93). Of course, tuning to DVB-T2 on a tuner which does not support it will fail. But it is better than TSDuck not supporting DVB-T2 on a tuner which supports it. The same problem exists with DVB-S vs. S2.

There is no such problem on Linux where the driver reports the exact capabilities of the tuner.

> Warning: This tuner discards all null packets (PID 0x1FFF), for both DVB-C and DVB-T2. So not to be used for bitrate measurements etc.

This is a problem I also got with some PCTV tuners. This is really a pain when transforming a TS on the fly before re-injecting it into a modulator.

nicst35, 2020-12-22:
Hi

After patching the Siano driver to make it report the modulation parameters for ISDB-T broadcasts, I noticed that `tsscan` fails to report layer parameters.

After scanning a few frequencies with `dvbv5-scan` using the following command:

    $ dvbv5-scan -C CL -G <some channel list in dvbv5 format>

I get something like this:

    [Canal 13 HD]
        SERVICE_ID = 42016
        NETWORK_ID = 1313
        TRANSPORT_ID = 1313
        VIDEO_PID = 811
        AUDIO_PID = 812
        FREQUENCY = 527142857
        BANDWIDTH_HZ = 6000000
        INVERSION = AUTO
        GUARD_INTERVAL = 1/8
        TRANSMISSION_MODE = 8K
        ISDBT_LAYER_ENABLED = 3
        ISDBT_PARTIAL_RECEPTION = 1
        ISDBT_SOUND_BROADCASTING = 0
        ISDBT_SB_SUBCHANNEL_ID = 0
        ISDBT_SB_SEGMENT_IDX = 0
        ISDBT_SB_SEGMENT_COUNT = 0
        ISDBT_LAYERA_FEC = 2/3
        ISDBT_LAYERA_MODULATION = QPSK
        ISDBT_LAYERA_SEGMENT_COUNT = 1
        ISDBT_LAYERA_TIME_INTERLEAVING = 4
        ISDBT_LAYERB_FEC = 3/4
        ISDBT_LAYERB_MODULATION = QAM/64
        ISDBT_LAYERB_SEGMENT_COUNT = 12
        ISDBT_LAYERB_TIME_INTERLEAVING = 2
        ISDBT_LAYERC_FEC = AUTO
        ISDBT_LAYERC_MODULATION = QAM/AUTO
        ISDBT_LAYERC_SEGMENT_COUNT = 0
        ISDBT_LAYERC_TIME_INTERLEAVING = 0
        COUNTRY = CL
        DELIVERY_SYSTEM = ISDBT

Guard intervals, transmission mode, per layer FEC and time interleaving are reported correctly. However, when I do a scan with `tsscan` using the following command line:

    $ tsscan --brazil -l -u --show-modulation --first-channel 23 --last-channel 23

I get this:

    * UHF channel 23 (527.142 MHz)
      Transport stream id: 1313, 0x0521
      Carrier frequency: 527,142,857 Hz
      Guard interval: 1/32
      Bandwidth: 6-MHz
      Transmission mode: 8K
      Layers: ABC
      Partial reception: on
      Layers A segment count: 255
      Layers A time interleaving: 255
      Layers B segment count: 255
      Layers B time interleaving: 255
      Layers C segment count: 255
      Layers C time interleaving: 255

      Name           Provider ServId TSId   ONetId Type PMTPID
      -------------- -------- ------ ------ ------ ---- ------
      Canal 13 HD    Harmonic 0xA420 0x0521 0x0521 0x01 0x032A
      13 .2          Harmonic 0xA421 0x0521 0x0521 0x01 0x0334
      Canal 13 Movil Harmonic 0xA438 0x0521 0x0521 0xC0 0x1FC8

Although guard intervals and transmission mode are reported and match with the dvbv5-scan output but no layer information is shown. Driver bug or tsduck bug?

lelegard, 2020-12-22:
>  Driver bug or tsduck bug?

Most probably a TSDuck bug since it works with `dvbv5-scan`. That's the problem with modulations for which I have no network to test. I cannot make sure by myself that it works. I will have a look but I will have to ask you to test afterwards.

I hope that your patch of the Siano driver will be merged in the kernel.

lelegard, 2020-12-22:
Hi @nicst35,

Only looking at the code, there is no obvious problem. So, this is the kind of problem that should be debugger under real conditions, either using a debugger or traces. As already mentioned, I cannot do anything on this because of the lack of actual ISDB-T network. Do you have the environment to do this?

The function that reads tuning information from the driver is `ts::Tuner::Guts::getCurrentTuning()`, starting [line 623 of file `src/libtsduck/dtv/linux/tsTunerGuts.cpp`](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/linux/tsTunerGuts.cpp#L623)

Reading ISDB-T parameters starts at [line 807](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/linux/tsTunerGuts.cpp#L807).

If you can debug this, please submit a [pull request](https://tsduck.io/doxy/contributing.html).

Sorry for not being able to actually work on this. Life would be easier if the whole world used DVB 😄

jof-cw, 2021-02-25:
Hello this is Jesus.

I'm using a Raspberry Pi TV HAT on a Raspberry 3B+ I tested on DVB-T and T2 multiplex and work fine. I'm using Raspbian Lite with TVHeadend.

0: /dev/dvb/adapter0 ("Sony CXD2880", DVB-T, DVB-T2)
Device: /dev/dvb/adapter0/frontend0

Status:

Delivery system ............................ DVB-T
Frequencies:
Current ............................ 546,000,000 Hz
Min ................................ 174,000,000 Hz
Max ................................ 862,000,000 Hz
Step ..................................... 1,000 Hz
Tolerance .................................... 0 Hz
Spectral inversion ........................... off
Modulation .................................. QPSK
Bandwidth .............................. 8,000,000 Hz
FEC(high priority) .......................... none
FEC(low priority) ........................... none
Transmission mode ............................. 2K
Guard interval .............................. 1/32
Hierarchy ................................... none

Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 4/5, FEC 5/6,
FEC 7/8, FEC auto, QPSK, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM,
QAM auto, transmission mode auto, guard interval auto, 2nd generation,
recover, mute TS

Best regards.

lelegard, 2021-02-26:
Hi @jof-cw,
Thanks for the feedback. This looks like a very cost-effective solution for DVB processing!

rphanikrishna, 2021-07-27:
Brand: HanfTek (https://www.amazon.in/gp/product/B071WFJKNY/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)
Model: ID 15f4:0131 HanfTek 
Modulation support (DVB-T, DVB-S, etc.): DVB-T, DVB-T2,DVB-C
Bus (USB, PCIe, etc.): USB
Tested systems (Linux, Windows): Raspberry PI 3+ (Raspberry pi os 5.10.17-v8+)

Output of tslsdvb:
**For Frontend 0:**

tslsdvb -d /dev/dvb/adapter0:0 -v
/dev/dvb/adapter0 ("Realtek RTL2832 (DVB-T)", DVB-T)
Device: /dev/dvb/adapter0/frontend0

  Status: 

  Delivery system ............................ DVB-T 
  Frequencies:
    Min ................................ 174,000,000 Hz
    Max ................................ 862,000,000 Hz
    Step ................................... 166,667 Hz
    Tolerance .................................... 0 Hz
  Spectral inversion ........................... off 
  Modulation ................................... QAM 
  Bandwidth ...................................... 0 Hz
  FEC(high priority) .......................... auto 
  FEC(low priority) ........................... auto 
  Transmission mode ........................... auto 
  Guard interval .............................. auto 
  Hierarchy ................................... auto 

  Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8,
    FEC auto, QPSK, 16-QAM, 64-QAM, QAM auto, transmission mode auto,
    guard interval auto, hierarchy auto, recover, mute TS

**For Frontend 1:**

tslsdvb -d /dev/dvb/adapter0:1 -v
/dev/dvb/adapter0:1:1:1 ("Panasonic MN88473", DVB-T, DVB-T2, DVB-C)
Device: /dev/dvb/adapter0/frontend1

tslsdvb: error reading status on /dev/dvb/adapter0/frontend1: Resource temporarily unavailable
  Delivery system ............................ DVB-T 
  Frequencies:
    Min ................................. 42,000,000 Hz
    Max .............................. 1,002,000,000 Hz
    Step ......................................... 0 Hz
    Tolerance .................................... 0 Hz
  Spectral inversion ........................... off 
  Modulation ................................... QAM 
  Bandwidth ...................................... 0 Hz
  FEC(high priority) .......................... auto 
  FEC(low priority) ........................... auto 
  Transmission mode ........................... auto 
  Guard interval .............................. auto 
  Hierarchy ................................... auto 

  Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8,
    FEC auto, QPSK, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM, QAM auto,
    transmission mode auto, guard interval auto, hierarchy auto, multistream,
    2nd generation, mute TS

**Will try recording and analysing few Streams from Indian DVB-C 64-QAM and 256-QAM soon.**

lelegard, 2021-08-01:
Hi @rphanikrishna 

Thanks for the feedback. It seems that the actual brand is Artek but is also referenced as HanfTek or Astrometa on some Linux sites.

The firmware `dvb-demod-mn88473-01.fw` is already part of the OpenELEC package which can be installed using the command `tsconfig --install-dvb-firmware` in the latest version of TSDuck (previously, there was in a dedicated script in the `build` subdirectory).

EnoSat, 2021-10-30:
Brand: **TBS**
Model: **TBS5520SE**
Modulation support: DVB-S/S2/S2X/T/T2/C/C2/ISDB-T
Bus: **USB**
Tested systems: **Windows11**

Output of tslsdvb:
set as **DVB-T/T2**
e:\TV\TSDuckBeta\bin>tslsdvb --debug=1 -v
tslsdvb: Debug: debug level set to 1
tslsdvb: Debug: ====> tslsdvb --debug=1 -v
tslsdvb: Debug: found tuner filter "TBS 5520 BDA Tuner"
tslsdvb: Debug: tuner device path: @device:pnp:\\?\usb#vid_734c&pid_5521#5&44cb01&0&4#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{b45b50ff-2d09-4bf2-a87c-ee4a7ef00857}
tslsdvb: Debug: DirectShowGraph init, graph build is valid: true, media control is valid: true
tslsdvb: Debug: adding filter "NetworkProvider", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Tuner", graph valid: true, filter not null: true
tslsdvb: Debug: initializing tuning space "TSDuck DVB-T Tuning Space"
tslsdvb: error setting UniqueName on TSDuck DVB-T Tuning Space: The specified object is too large.
tslsdvb: error setting FriendlyName on TSDuck DVB-T Tuning Space: The specified object is too large.
tslsdvb: Debug: user-specified receiver filter name: ""
tslsdvb: Debug: trying direct connection from tuner (no receiver)
tslsdvb: Debug: adding filter "Tee", graph valid: true, filter not null: true
tslsdvb: failed to connect pins: Žiadna kombinácia filtrov pre prepojenie sa nedala nájsť.
tslsdvb: Debug: failed to connect filters
tslsdvb: Debug: trying receiver filter "TBS 5520 BDA Capture"
tslsdvb: Debug: adding filter "Receiver", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Tee", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Sink/Capture", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Demux", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "TIF", graph valid: true, filter not null: true
tslsdvb: Debug: using receiver filter "TBS 5520 BDA Capture"
tslsdvb: Debug: IBDA_DigitalDemodulator in tuner: 7
tslsdvb: Debug: IBDA_DigitalDemodulator2 in tuner: 7
tslsdvb: Debug: IBDA_SignalStatistics in tuner: 7
tslsdvb: Debug: IKsPropertySet in tuner: 2
tslsdvb: Debug: found tuner device "TBS 5520 BDA Tuner"
0: "TBS 5520 BDA Tuner" (DVB-T, DVB-T2)
   Device: @device:pnp:\\?\usb#vid_734c&pid_5521#5&44cb01&0&4#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{b45b50ff-2d09-4bf2-a87c-ee4a7ef00857}
     Signal locked:    yes
     Signal strength:  0 dB

set as **DVB-S/S2**
e:\TV\TSDuckBeta\bin>tslsdvb` --debug=1 -v
tslsdvb: Debug: debug level set to 1
tslsdvb: Debug: ====> tslsdvb --debug=1 -v
tslsdvb: Debug: found tuner filter "TBS 5520 BDA Tuner"
tslsdvb: Debug: tuner device path: @device:pnp:\\?\usb#vid_734c&pid_5521#5&44cb01&0&4#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{b45b50ff-2d09-4bf2-a87c-ee4a7ef00857}
tslsdvb: Debug: DirectShowGraph init, graph build is valid: true, media control is valid: true
tslsdvb: Debug: adding filter "NetworkProvider", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Tuner", graph valid: true, filter not null: true
tslsdvb: Debug: initializing tuning space "TSDuck DVB-S Tuning Space"
tslsdvb: Debug: user-specified receiver filter name: ""
tslsdvb: Debug: trying direct connection from tuner (no receiver)
tslsdvb: Debug: adding filter "Tee", graph valid: true, filter not null: true
tslsdvb: failed to connect pins: Žiadna kombinácia filtrov pre prepojenie sa nedala nájsť.
tslsdvb: Debug: failed to connect filters
tslsdvb: Debug: trying receiver filter "TBS 5520 BDA Capture"
tslsdvb: Debug: adding filter "Receiver", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Tee", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Sink/Capture", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Demux", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "TIF", graph valid: true, filter not null: true
tslsdvb: Debug: using receiver filter "TBS 5520 BDA Capture"
tslsdvb: Debug: IBDA_DigitalDemodulator in tuner: 2
tslsdvb: Debug: IBDA_DigitalDemodulator2 in tuner: 2
tslsdvb: Debug: IBDA_SignalStatistics in tuner: 2
tslsdvb: Debug: IKsPropertySet in tuner: 2
tslsdvb: Debug: found tuner device "TBS 5520 BDA Tuner"
0: "TBS 5520 BDA Tuner" (DVB-S, DVB-S2)
   Device: @device:pnp:\\?\usb#vid_734c&pid_5521#5&44cb01&0&4#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{b45b50ff-2d09-4bf2-a87c-ee4a7ef00857}
[... 46 lines cut ...]
tslsdvb: error setting UniqueName on TSDuck DVB-T Tuning Space: The specified object is too large.
tslsdvb: error setting FriendlyName on TSDuck DVB-T Tuning Space: The specified object is too large.
tslsdvb: Debug: user-specified receiver filter name: ""
tslsdvb: Debug: trying direct connection from tuner (no receiver)
tslsdvb: Debug: adding filter "Tee", graph valid: true, filter not null: true
tslsdvb: failed to connect pins: Žiadna kombinácia filtrov pre prepojenie sa nedala nájsť.
tslsdvb: Debug: failed to connect filters
tslsdvb: Debug: trying receiver filter "TBS 5520 BDA Capture"
tslsdvb: Debug: adding filter "Receiver", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Tee", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Sink/Capture", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "Demux", graph valid: true, filter not null: true
tslsdvb: Debug: adding filter "TIF", graph valid: true, filter not null: true
tslsdvb: Debug: using receiver filter "TBS 5520 BDA Capture"
tslsdvb: Debug: IBDA_DigitalDemodulator in tuner: 7
tslsdvb: Debug: IBDA_DigitalDemodulator2 in tuner: 7
tslsdvb: Debug: IBDA_SignalStatistics in tuner: 7
tslsdvb: Debug: IKsPropertySet in tuner: 2
tslsdvb: Debug: found tuner device "TBS 5520 BDA Tuner"
0: "TBS 5520 BDA Tuner" (DVB-T, DVB-T2)
   Device: @device:pnp:\\?\usb#vid_734c&pid_5521#5&44cb01&0&4#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\{b45b50ff-2d09-4bf2-a87c-ee4a7ef00857}
     Signal locked:    no
     Signal strength:  0 dB

result: occasional system freeze

lelegard, 2021-10-30:
Thanks @EnoSat. Quite an impressive list of demods for one single device.

- How do you switch from one mode to another?
- In C2 mode, it appears as T/T2?
- Did you try it on Linux? Linux is much better to handle multi-standard demods.

EnoSat, 2021-10-30:
small windows utility for change mode , real test C2 not possible , no try :(
[image]
[image]
[image]

EvariSTC, 2022-10-27:
USB dongle **MyGica T230C,** support for **DVB-T & DVB-T2**, tested on Windows 10
Output of tslsdvb: "Digital TV" (DVB-T, DVB-T2)
Bought on Amazon.it: https://www.amazon.it/dp/B01BU4UIOW

tsp -I dvb -a 0 --uhf-channel 34 -P until --second 1 -P analyze -O drop

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ....... 8192 (0x2000)  |  Services: ............. 20  |
|  Bytes: ........................ 11,208,748  |  PID's: Total: ......... 75  |
|  TS packets: ....................... 59,621  |         Clear: ......... 75  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: .... 20  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ......................... 22,394,133 b/s   24,300,017 b/s  |
|  Estimated based on PCR's: ............... 22,394,133 b/s   24,300,017 b/s  |
|  Selected reference bitrate: ............. 22,394,133 b/s   24,300,017 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ..................................... 4 sec (0 min 4 sec)  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x0016  Telelombardia HD. ............................ C    1,602,344 b/s  |
|  0x0017  TopCalcio24 HD. .............................. C    1,615,866 b/s  |
|  0x0018  Antenna 3 HD. ................................ C    1,603,471 b/s  |
|  0x0019  TELECOLOR. ................................... C    1,323,267 b/s  |
|  0x001A  CREMONA1 ..................................... C    1,588,447 b/s  |
|  0x001B  TELENOVA HD .................................. C    1,603,096 b/s  |
|  0x001C  UNICA LECCO .................................. C    1,091,517 b/s  |
|  0x001D  TELELIBERTA' ................................. C    1,347,682 b/s  |
|  0x001E  TeleBoario ................................... C    1,082,503 b/s  |
|  0x001F  Milano Pavia TV .............................. C    1,058,839 b/s  |
|  0x0020  TELEREPORTER ................................. C      804,928 b/s  |
|  0x0022  Antenna 2 .................................... C      836,855 b/s  |
|  0x0023  TRS .......................................... C      798,543 b/s  |
|  0x0024  STUDIO1 ...................................... C    1,082,503 b/s  |
|  0x0025  TVS Televallassina ........................... C      845,494 b/s  |
|  0x0026  NBC MILANO ................................... C      798,919 b/s  |
|  0x0027  LOMBARDIA TV 80 .............................. C      763,987 b/s  |
|  0x0028  MILANOW. ..................................... C      582,944 b/s  |
|  0x007D  Primantenna .................................. C      810,187 b/s  |
|  0x0094  ITALIA 8 PRESTIGE ............................ C      806,055 b/s  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 923, PID's: 6 (clear: 6, scrambled: 0)                         |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C      346,686 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       45,449 b/s  |
|  0x0000  PAT .......................................... C        7,512 b/s  |
|  0x0010  NIT .......................................... C        1,502 b/s  |
|  0x0011  SDT/BAT ...................................... C       30,049 b/s  |
|  0x0012  EIT .......................................... C        4,132 b/s  |
|  0x0015  Network Synchronization ...................... C        2,254 b/s  |
|  0x1FFF  Stuffing ..................................... C      301,238 b/s  |
|=============================================================================|
|  Service: 0x0016 (22), TS: 0x2000 (8192), Original Netw: 0x217C (8572)      |
|  Service name: Telelombardia HD., provider: Rai Way                         |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 4,266, PID's: 5 (clear: 5, scrambled: 0)                       |
|  PMT PID: 0x00DE (222), PCR PID: 0x01A6 (422)                               |
[... 1301 lines cut ...]
|=============================================================================|
|  PID: 0x07EA (2026)                          MPEG-2 Private sections (AIT)  |
[... cut ...]
|  Service: 0x0028 (40) MILANOW.                                              |
|-----------------------------------------------------------------------------|
|  0x74 (116, AIT), TID ext: 0x0010 (16)                                      |
|      Repetition: ..... 499  ms  Section cnt: ........ 8                     |
|      Min repet.: ..... 486  ms  Table cnt: .......... 8                     |
|      Max repet.: ..... 515  ms  Version: ........... 21                     |
===============================================================================

lelegard, 2022-10-29:
Thanks @EvariSTC. I have added it in the list of tested devices in the user's guide.

jof-cw, 2023-01-07:
Hello all and happy new year :)  This is Jesus

I just tested in my Raspberry 3B+ a USB dongle MyGica T230C  like the one @EvariSTC have 3, and those are the results:

---------------------------------
_**Raspbian Bullseye**_

_**uname -a**_
Linux Raspberry 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux

--------------------------------
**_tslsdvb_**

0: /dev/dvb/adapter0 ("Sony CXD2880", DVB-T, DVB-T2)
1: /dev/dvb/adapter1 ("Silicon Labs Si2168", DVB-T, DVB-T2, DVB-C)

--------------------------------

 **_tslsdvb -v_**   (I omited the adapter 0)
 
1: /dev/dvb/adapter1 ("Silicon Labs Si2168", DVB-T, DVB-T2, DVB-C)
   Device: /dev/dvb/adapter1/frontend0

tslsdvb: error reading status on /dev/dvb/adapter1/frontend0: Resource temporarily unavailable
tslsdvb: error reading status on /dev/dvb/adapter1/frontend0: Resource temporarily unavailable
     Signal: locked: no, strength: -36 dB, SNR: 38.5 dB, BER: 1%

     Delivery system ........................... DVB-T2
     Frequencies:
       Current ............................ 690,000,000 Hz
       Min ................................. 48,000,000 Hz
       Max ................................ 870,000,000 Hz
       Step .................................... 62,500 Hz
       Tolerance .................................... 0 Hz
     Spectral inversion ............................ on
     Modulation ............................... 256-QAM
     Bandwidth .............................. 8,000,000 Hz
     FEC(high priority) .......................... auto
     FEC(low priority) ........................... auto
     Transmission mode ............................. 8K
     Guard interval ............................. 1/128
     Hierarchy ................................... none

     Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8,
       FEC auto, QPSK, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM, QAM auto,
       transmission mode auto, guard interval auto, hierarchy auto,
       multistream, 2nd generation, mute TS

--------------------------------

**_tsp --verbose -I dvb -a 1 -u 48 --delivery-system DVB-T2 --modulation 256-QAM --guard-interval 1/128 -P until --seconds 30 -P analyze -O drop_**

* dvb: using /dev/dvb/adapter1 (DVB-T, DVB-T2, DVB-C)
* dvb: tuned to transponder --delivery-system DVB-T2 --frequency 690,000,000 --modulation 256-QAM --high-priority-fec auto --low-priority-fec auto --bandwidth 8,000,000 --transmission-mode 8K --guard-interval 1/128 --hierarchy none
* dvb: locked: yes, SNR: 37 dB, BER: 1%
* dvb: initial input bitrate is 33,358,077 b/s

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ......... 10 (0x000A)  |  Services: .............. 2  |
|  Bytes: ....................... 133,721,204  |  PID's: Total: ......... 15  |
|  TS packets: ...................... 711,283  |         Clear: ......... 15  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 2  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ......................... 33,354,323 b/s   36,192,989 b/s  |
|  Estimated based on PCR's: ............... 33,354,230 b/s   36,192,888 b/s  |
[... 297 lines cut ...]
|=============================================================================|
|  PID: 0x0103 (259)                           MPEG-2 Private sections (AIT)  |
|  Service: 0x30D5 (12501) UHD-1                                              |
|-----------------------------------------------------------------------------|
|  0x74 (116, AIT), TID ext: 0x0010 (16)                                      |
|      Repetition: ..... 100  ms  Section cnt: ...... 320                     |
|      Min repet.: ...... 99  ms  Table cnt: ........ 320                     |
|      Max repet.: ..... 101  ms  Version: ............ 3                     |
|=============================================================================|
|  PID: 0x0203 (515)                           MPEG-2 Private sections (AIT)  |
|  Service: 0x30D6 (12502) UHD-2                                              |
|-----------------------------------------------------------------------------|
|  0x74 (116, AIT), TID ext: 0x0010 (16)                                      |
|      Repetition: ..... 100  ms  Section cnt: ...... 320                     |
|      Min repet.: ...... 99  ms  Table cnt: ........ 320                     |
|      Max repet.: ..... 101  ms  Version: ............ 0                     |
===============================================================================

------------------------------

To use this USB tuner on a Raspberry Pi,  I only needed to copy the next drivers at **_/usr/lib/firmware_**:
    **_dvb-demod-si2168-d60-01.fw
    dvb-tuner-si2141-a10-01.fw_**

     Download them from [Github Open Source Media Center](https://github.com/osmc/dvb-firmware-osmc)

EvariSTC, 2023-02-07:
There is USB dongle with support for DVB-S & DVB-S2?

lelegard, 2023-02-08:
Hi @EvariSTC 

> There is a USB dongle with support for DVB-S & DVB-S2?

I used a DVBSky S960 which was quite reliable, Linux or Windows. Not really a "dongle", more like a small box with auxiliary power supply. But definitely a USB interface and it works well.

In a "dongle" shape, I also tested the PCTV DVB-S2 Stick 461e a long time ago. But it was not stable. I do not recommend it.

EvariSTC, 2023-02-08:
@lelegard  do you happen to have a link to a store because it's not available on Amazon?

lelegard, 2023-02-08:
I had purchased mine years ago on Amazon. Maybe it is no longer available. I cannot make any commercial advice for any retailer, sorry.

EvariSTC, 2023-02-10:
There is a PCI-e card with support for DVB-s & DVB-S2 still available?

EvariSTC, 2023-04-26:
@oviano you posted in 2019 the tslsdvb of your "PCTV 292e/461e BDA 28179 TVTuner S", I opened the file and it says: PCTV 292e/461e BDA 28179 TVTuner S" (DVB-S). It only tunes DVB-S or even DVB-s2?
I ask you this information because I'm looking for a usb sat decoder compatible with Windows and I saw your post.
Thank you very much.

stulluk, 2023-10-28:
Before damaging my TBS5530 ( Where I think its firmware got damaged somehow so it does not get detected by kernel anymore), I confirm it worked perfectly with tsduck.

EvariSTC, 2024-05-03:
USB dongle **MyGica PT362**, support for **DVB-T & DVB-T2**, tested on Windows 11
Output of tslsdvb: "Digital TV" (DVB-T, DVB-T2)
Bought on Amazon.it: https://www.amazon.it/Geniatech-MyGica-PT362-DVB-T2-Ricevitore/dp/B095KD5L3N/
Driver of MyGica T230A to be downloaded from Geniatech website.

tsp -I dvb -a 0 --uhf-channel 42 -P until --second 1 -P analyze -O drop

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ...... 10252 (0x280C)  |  Services: ............. 11  |
|  Bytes: ........................ 11,396,748  |  PID's: Total: ......... 53  |
|  TS packets: ....................... 60,621  |         Clear: ......... 53  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: .... 10  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ......................... 24,882,355 b/s   27,000,003 b/s  |
|  Estimated based on PCR's: ............... 24,882,356 b/s   27,000,003 b/s  |
|  Selected reference bitrate: ............. 24,882,356 b/s   27,000,003 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ..................................... 3 sec (0 min 3 sec)  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x0028  RAS ZDF Neo HD ............................... C    2,072,811 b/s  |
|  0x0029  RAS ORF Sport+ HD ............................ C    3,613,669 b/s  |
|  0x002B  RAS ServusTV HD Oesterreich .................. C    2,049,005 b/s  |
|  0x002C  TV33 BZ ...................................... C    2,832,979 b/s  |
|  0x002D  Kronplatz - Plan de Corones .................. C    2,350,691 b/s  |
|  0x002E  Peer.tv Südtirol ............................. C    2,125,760 b/s  |
|  0x002F  RTTR ......................................... C    1,190,327 b/s  |
|  0x0030  ALTO ADIGE TV ................................ C    2,541,144 b/s  |
|  0x0031  CANALE ITALIA ................................ C    2,666,744 b/s  |
|  0x0032  TRENTINO TV HD ............................... C    2,706,558 b/s  |
|  0x003A  CANALE ITALIA ................................ C    2,667,154 b/s  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 1,768, PID's: 6 (clear: 6, scrambled: 0)                       |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C      725,689 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C      185,937 b/s  |
|  0x0000  PAT .......................................... C        6,567 b/s  |
|  0x0010  NIT (Network: 0x3071 (12401) RAS DVB-t NETWOR  C        1,231 b/s  |
|  0x0011  SDT/BAT ...................................... C        2,052 b/s  |
|  0x0012  EIT .......................................... C      173,624 b/s  |
|  0x0015  Network Synchronization ...................... C        2,463 b/s  |
|  0x1FFF  Stuffing ..................................... C      539,752 b/s  |
|=============================================================================|
|  Service: 0x0028 (40), TS: 0x280C (10252), Original Netw: 0x217C (8572)     |
|  Service name: RAS ZDF Neo HD, provider: RAS                                |
|  Service type: 0x19 (Advanced codec HD digital television service)          |
|  TS packets: 5,050, PID's: 8 (clear: 8, scrambled: 0)                       |
|  PMT PID: 0x0FA0 (4000), PCR PID: 0x0FA1 (4001)                             |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Advanced codec HD digital television service . C    2,072,811 b/s  |
|  0x0FA0  PMT .......................................... C        6,567 b/s  |
|  0x0FA1  AVC video (1280x720, high profile, level 4.0,  C    1,280,628 b/s  |
|  0x0FA2  MPEG-1 Audio (deu, Audio layer II, 256 kb/s,   C      266,387 b/s  |
|  0x0FA3  MPEG-1 Audio (mis, Audio layer II, 192 kb/s,   C      201,124 b/s  |
|  0x0FA5  Teletext (deu, Initial Teletext page) ........ C      301,686 b/s  |
[... 1102 lines cut ...]
|  Service: 0x003A (58) CANALE ITALIA                                         |
[... cut ...]
|      Max repet.: .... 1128  ms  Version: ............ 0                     |
|-----------------------------------------------------------------------------|
|  0x3C (60, DSM-CC DDM), TID ext: 0x0001 (1)                                 |
|      Repetition: .... 1128  ms  Section cnt: ........ 3                     |
|      Min repet.: .... 1128  ms  Table cnt: .......... 3                     |
|      Max repet.: .... 1128  ms  Version: ............ 2                     |
===============================================================================

castinglife, 2024-07-05:
This is Korea (ATSC? ATSC-T?).
Automatically record TV schedules (automatically remove commercial breaks) using tsduck
I'm trying to create a script.
Currently, DTA-2131B and DTA-2139C tuner cards are installed in the ubuntu22.04 server.

After scanning the channel (tsscan), take the channel ID
I understand that an automatic recording script is created using the tsp command.

Channel search using the tsscan command
Automatic recording with tsp command (removing interstitial ads)
Can you tell me how to write it?

I'm trying to scan the current channel.
The tsscan command does not work.
[image]
[image]
[image]
[image]

lelegard, 2024-07-09:
Hi @castinglife 

Dektec devices are professional tools with their own interface, the `dektec` input plugin. They are not regular commercial tuner devices, within a standard system-defined tuner framework (LinuxTV on Linux, DirectShow on Windows). The main reason is that Dektec devices are more general than tuners, they have more professional features such an signal analysis and monitoring. Commercial tuners and the Linux or Windows tuner frameworks are only designed to "watch TV". Therefore, Dektec did not develop any standard driver for LinuxTV and DirectShow.

The command `tsscan` uses the standard tuner framework and cannot use a Dektec demodulator.

If you have a Dektec demodulator, you can receive a signal in `tsp` using the `dektec` input plugin, not the `dvb` plugin for standard tuners (arguably misnamed since it is not limited to DVB tuners).

Removing ads is another topic. If you have questions on this, please open a dedicated topic. This topic is reserved to user feedback on standard tuners, to build a list of community-tested tuner devices.

garret1317, 2025-01-16:
I have these:

**Microsoft Xbox One Digital TV Tuner**
Modulation support: DVB-T, DVB-T2, DVB-C (I've only tested with T and T2)
Bus: USB
I had to add the firmware listed on https://www.linuxtv.org/wiki/index.php/Xbox_One_Digital_TV_Tuner for it to work with anything

**Realtek RTL2832** "RTL-SDR"
Modulation support: DVB-T
Bus: USB
also known as the RTL-SDR, can be used as a software defined radio https://www.rtl-sdr.com/
rtl-sdr tools say mine is a Rafael Micro R820T

Tested systems: `Linux 5.15.0-71-generic #78-Ubuntu SMP Tue Apr 18 09:00:29 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux` mint 21

```
$ tslsdvb 
0: /dev/dvb/adapter0 ("Panasonic MN88472, Microsoft Corp. Xbox USB Tuner 2.00 SN:003243230915", DVB-T, DVB-T2, DVB-C)
1: /dev/dvb/adapter1 ("Realtek RTL2832 (DVB-T), Realtek RTL2838UHIDIR 2.00 SN:00000001", DVB-T)
```
<details><summary>tslsdvb -v</summary>

```
0: /dev/dvb/adapter0 ("Panasonic MN88472, Microsoft Corp. Xbox USB Tuner 2.00 SN:003243230915", DVB-T, DVB-T2, DVB-C)
   Device: /dev/dvb/adapter0/frontend0

     Signal: locked: no

     Delivery system ............................ DVB-T 
     Frequencies:
       Current ............................ 522,000,000 Hz
       Min ................................. 42,000,000 Hz
       Max ................................ 870,000,000 Hz
       Step ......................................... 0 Hz
       Tolerance .................................... 0 Hz
     Spectral inversion ............................ on 
     Modulation ................................ 64-QAM 
     Bandwidth .............................. 8,000,000 Hz
     FEC(high priority) .......................... auto 
     FEC(low priority) ........................... auto 
     Transmission mode ............................. 8K 
     Guard interval .............................. 1/32 
     Hierarchy ................................... none 

     Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8,
       FEC auto, QPSK, 16-QAM, 32-QAM, 64-QAM, 128-QAM, 256-QAM, QAM auto,
       transmission mode auto, guard interval auto, hierarchy auto,
       multistream, 2nd generation, mute TS

1: /dev/dvb/adapter1 ("Realtek RTL2832 (DVB-T), Realtek RTL2838UHIDIR 2.00 SN:00000001", DVB-T)
   Device: /dev/dvb/adapter1/frontend0

     Signal: locked: no, strength: 92%, SNR: 31.259 dB, BER: 0%

     Delivery system ............................ DVB-T 
     Frequencies:
       Current ............................ 498,000,000 Hz
       UHF channel ................................. 24 
       Min ................................ 174,000,000 Hz
       Max ................................ 862,000,000 Hz
       Step ................................... 166,667 Hz
       Tolerance .................................... 0 Hz
     Spectral inversion ............................ on 
     Modulation ................................ 64-QAM 
     Bandwidth .............................. 8,000,000 Hz
     FEC(high priority) ........................... 2/3 
     FEC(low priority) ............................ 1/2 
     Transmission mode ............................. 8K 
     Guard interval .............................. 1/32 
     Hierarchy ................................... none 

     Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 5/6, FEC 7/8,
       FEC auto, QPSK, 16-QAM, 64-QAM, QAM auto, transmission mode auto,
       guard interval auto, hierarchy auto, recover, mute TS
```
</details>

I'm not a broadcasting person, I haven't done any detailed/stress tests or anything like that
but both seem to work fine for watching/recording stuff with tsp