tsduck/tsduck #25: can i use this application as PSI/Si sofyware to provide epg Set top box

sachin0606, 2018-01-25:
Hi All, 

sorry for noob question, i wanted to know shall use this to generate EPG for my set top box? like below link of software?

[http://www.mcbsintl.com/epg.html](url)

lelegard, 2018-01-25:
Hi,

It depends on what you really want to do. The software you mention, after briefly looking at their site, is an EPG management system, with a GUI and a database, and multiple features.

The main point of TSDuck is to manipulate transport streams and PSI/SI. To generate an EPG with TSDuck, you may create EIT tables in XML form and inject them in a transport stream which is modulated for your STB. This is feasible but much less easy than with a specialised software, essentially because EIT's are hard to tame.

And this applies to broadcast STB's only. For fully connected STB's, either for IP/TV or OTT, the EPG is usually pulled by the STB through HTTP instead of being pushed into it through broadcast and the EPG data format is proprietary to the operator.

sachin0606, 2018-01-29:
Thanks lelegard  for your briefe explination.

any link where can accomplish this with  reference ? so i can refer and design GUI for this???

lelegard, 2018-01-29:
Hi,

You did not explain what you wanted to do, precisely. So it is hard to help.

Developing GUI and XML-generation applications is off-topic here. There are zillions of ways to achieve this, native programming, Java, Web-based, etc. which are much better discussed elsewhere.

Here, we can help in using TSDuck. Precise questions about TSDuck are much more likely to be answered.

closed 2018-02-07 by lelegard

sachin0606, 2018-02-22:
thanks and its really good tool to learn and develop.

can you suggest me how can we do this using TSDuck? 

what all command i have to use and how to inject the table..

lelegard, 2018-02-22:
What do you mean by "do this" ? Do what, precisely ?

If you create tables, EIT or others, in XML format, you can inject them in a stream using the plugin `inject`. Please read the user's guide. There are many examples in addition to the reference of each command and plugin.