tsduck/tsduck #26: An absence of timestamps in XML log

alexander-nf, 2018-01-25:
Hi Thierry, 

thanks a lot for the project and for the detailed explanations again and again. I have explored and tested just a small part of TSDuck - so much features it has.

I haven't found information in the guide about the topic. Could you please shed a light on this?

I see that `tstables` / `tables` `--time-stamp` option doesn't create time-stamps in XML logs. Is it determined feature or not? It would be great to see registration time for occurrences of a table / of the first section of a table.

Could you please comment if it's possible to get timestamps in XML log?

lelegard, 2018-01-25:
You got a point. No way to get the time stamp in XML for now. I will add it.

lelegard, 2018-01-25:
Fixed in 3.7-499, commit f47bd914aaa289f4a842ace1cb29e2e27e5c9557.
Really trivial to fix, indeed.

closed 2018-01-25 by lelegard