tsduck/tsduck #27: T2-MI PLP TSes simultaneous analysis

alexander-nf, 2018-01-25:
Is it possible to do one-pass analysis of a multiple PLP T2-MI live stream?
For example I'd like to capture some tables which should have correlation in terms of time from different PLP subTSes simultaneously.

If it is not possible at the moment, what are the chances to implement this using TSDuck architecture?

I suppose the command could be the following:
```bash
tsp \
    -I [dvb|dektec|file|ip] \
    -P t2mi \
        --pid 0x1000 \
        --extract \
        --plp all \
    -P tables \
        --no-duplicate \
        --pid 18 \
        --tid 79 \
        --time-stamp \
        --xml-output EIT_other.xml \
        --text-output EIT_other.txt \
    -O drop
```

`-I dvb` - for using with DVB-S tuner.

`--plp all` - the new value of the `--plp` option I am dreaming of...
Maybe manual PLPs selection is easier to interpret, another variant: `--plp 0 --plp 1 --plp 2` or `--plp 0 --plp 2` or etc.

`EIT_other.xml` and `EIT_other.txt` - base names for EIT_other_plpN.txt and EIT_other_plpN.xml logs.

lelegard, 2018-01-25:
Hi,

I understand the idea. However, multiple PLP are independent TS. If tsp extracts all PLP's at the same time, what will be the resulting stream? A mixture of all PLP's? When you write `--pid 18`, you collect packets from all PLP's, a bit of an EIT from PLP 0, a piece of another EIT from PLP 1, etc. What a mess :)

You could try plugin `fork` after `-I dvb`. Thus, you can create other instances of `tsp`. Each instance receives the same TS from the live source and can extract its own PLP to analyse it.

lelegard, 2018-01-25:
This one works:

~~~
tsp -I file foo.ts \
    -P fork --buffered-packets 1000 'tsp -P t2mi --pid 0x1000 --plp 0 -P tables --pid 0 -O drop' \
    -P fork --buffered-packets 1000 'tsp -P t2mi --pid 0x1000 --plp 1 -P tables --pid 0 -O drop' \
    -P t2mi --pid 0x1000 --plp 2 -P tables --pid 0 -O drop
~~~

The option `--buffer-packets` is useful because the default is to pass packets one by one in the unix pipe, which is very inefficient on large streams.

alexander-nf, 2018-01-26:
I understood the idea after the first reading the guide: 1 TS in, 1 TS out. My unconscious thought was to proccess multi-TS T2-MI stream at once, not twice/trice/etc., and without piping. :)
I threw the thought away after your explanation and I've checked your recipe modified for my needs with T2-MI live stream. It works, it is a suitable solution. Thank you!

A bit more information about the user experience:
- I have tested the command with 3 piped `tsp` instances on Windows (I changed `'` to `"` and `\` to `^` before an execution).
- CPU load was just about 5% on old low-end [Xeon 3065](https://ark.intel.com/en/products/30775/Intel-Xeon-Processor-3065-4M-Cache-2_33-GHz-1333-MHz-FSB).
- `dektec` input devices I have used:
1. Device 0: DTA-2144 (DTA-2144 Quad ASI/SDI Input+Output)
  Production date: 2011.06 
2. Device 1: DTU-245 (DTU-245 FantASI USB-2 ASI/SDI Input+Output)
  Production date: 2013.11

I haven't met **PCIe** DTA-2144 / **DTA-2xxx** in the guide:
> **dektec (input)**
> **Dektec DTA-1xx and DTU-2xx ASI Devices**
> This input plugin receives packets from a DVB-ASI Dektec DTA-1xx or DTU-2xx device

So I was slightly afraid that the device is incompatible with TSDuck. My guess was PCIe DTA-2xxx has DTAPI support, but it is black-listed by TSDuck probably.

lelegard, 2018-01-26:
Hi,

The support for Dektec devices is much cleaner than DVB tuners because the same company provides all hardware and a consistent DTAPI software. So, any new Dektec device - as long as it does not introduce new original features of course - is fully supported by the existing DTAPI and I am very confident that TSDuck works with it. This is why I have not maintained a list of tested devices as large as with tuners. 

Good luck for your work with TSDuck.

closed 2018-01-26 by lelegard