tsduck/tsduck #28: Bitrate monitor fails to display bitrate when exceeding limit

djsdevel, 2018-02-06:
Running the bitrate monitor like:

tsp -I ip -l 192.168.20.2 239.1.1.1:5000 -P bitrate_monitor 221 -t 1 --min 3000000 --max 3000000 -O drop

When the bitrate is lower than the limit the actual bitrate is displayed, but when over the limit the detail is missing, see example output below:

* Warning: bitrate_monitor:  is greater than allowed maximum (3,000,000 bits/s)
* Warning: bitrate_monitor: pid 221 (0x00DD) - bitrate (2,961,376 bits/s) is lower than allowed minimum (3,000,000 bits/s)

Apart from this minor issue they are great tools and look very useful.

djsdevel, 2018-02-06:
I should have mentioned I installed from - TSDuck-Win64-3.7-512.exe

lelegard, 2018-02-06:
Good spot ! Thank you for the report, details matter.

It was "just" a '+' sign missing :(

Fixed in commit ecd00c0b25c36387c33dcee51eea965105b22dba, version 3.8-522.

closed 2018-02-06 by lelegard