tsduck/tsduck #31: Enhancement request for bitrate_monitor

djsdevel, 2018-02-06:
Hi,

Thanks for the fast responses on the other issues I raised today.

A minor enhancement request that would be very handy for spot checking variable bitrate components would be an extra option for the bitrate monitor to display the actual bitrate at the frequency of the time interval, so alarms can be shown for being outside max and min and as a confidence that a variable bitrate component is changing within the max and min the changing bitrates can be seen in the outut.

Regards

Darren

lelegard, 2018-02-06:
No problem, quite easy. 
Will do it.

lelegard, 2018-02-06:
Implemented. 
Sample output:

```
$ tsp ...  -P bitrate_monitor --periodic-bitrate 4 ...
* bitrate_monitor: 2018/02/06 23:32:04, pid 210 (0x00D2), bitrate: 2,339,622 bits/s
* bitrate_monitor: 2018/02/06 23:32:08, pid 210 (0x00D2), bitrate: 2,415,123 bits/s
* bitrate_monitor: 2018/02/06 23:32:12, pid 210 (0x00D2), bitrate: 3,259,168 bits/s
* bitrate_monitor: 2018/02/06 23:32:16, pid 210 (0x00D2), bitrate: 3,108,166 bits/s
```

closed 2018-02-06 by lelegard