tsduck/tsduck #33: Add support for another table ?

ElyKar, 2018-02-12:
Hi,
First, thanks for this amazing software ! I've just downloaded it and couldn't test it yet, but it really looks incredible !
What would be the steps to add interactive applications manipulation to tsduck ? For example, implementing a plugin listing all applications for a given service. From what I could see, it would require to:
- Implement AIT and its descriptors (defined in ts 102 809)
- Add a plugin (applist ?) to the list of plugins

Is this the right way to go ? Or I am maybe looking to do something already possible.

lelegard, 2018-02-12:
Hi,

It depends on what you mean by "application manipulation".

Implementing AIT and related descriptors (a few of them are already there) is probably a good idea. It just takes time. When I have some, I will consider implementing 102 809 signalisation. With this, you will be able to analyse the application signalisation using commands like `tstables`(or the related plugin `tables`). You will be also able to create or edit AIT's in XML format, capture them or inject them with the standard TSDuck tools.

What would be the features of an `applist` plugin in your view ?

Do you have sample streams with existing AIT's ? I probably already came across some of them but never paid attention.  I can easily capture on Astra 19.2 E and HotBird 13 E. Would you recommend any sample here?

(BTW, are you from INSA Rennes?)

ElyKar, 2018-02-13:
Indeed, it's more adding signalisation support to tsduck.
Actually, I wondered if I could try to implement this myself (ait and descriptors), based on what has been done for other tables.
After looking at the documentation and existing plugins, it looks like the way to go would be to implement an `ait` plugin performing various operations (e.g. list applications, change priority, change transport mode, modify version, ...)

I do not have any stream with an AIT right now (and can't capture any where I am currently), but I'm confident there is one in almost every multiplex (R4 I'm 99% sure).

I don't know of your plans for tsduck, but if that's ok with you I'd like to try to hack into it to implement signalisation structures.
Then if it went well, maybe we can try to determine what an `ait` plugin could do.
Is it fine with you ?

(Yes, that's me, why ?)

lelegard, 2018-02-13:
Hi,

Contributions are welcome. I will consider any pull request. I only request that the architecture, structure and coding style of TSDuck are preserved, including Doxgen completeness for header files in `libtsduck`.

I recommend to start by implementing the AIT. It has some structural similarities with a PMT. So, you may use `tsPMT.h` /  `.cpp` as template for a new `tsAIT.h` / `.cpp`. Also add the corresponding definitions in `tsduck.xml`. I always try to keep the original names from MPEG/DVB standards in the XML syntax.

At this point, you will need some real streams with real AIT's to test. I have found some in TS 6000 on Hot Bird 13° E as shown below:

~~~
$ tsp -I dvb --freq 11,919,000,000 --symb 29,900,000 --fec 5/6 --pol vert --del DVB-S \
      -P tables --tid 0x74 \
      -O drop

* AIT, TID 116 (0x74), PID 7877 (0x1EC5)
  Version: 0, sections: 1, total size: 182 bytes
  - Section 0:
    TIDext: 1 (0x0001)
    0000:  F0 00 F0 A6 00 00 00 0B 1A B5 02 F0 9D 00 09 05  ................
    0010:  00 01 01 01 01 3F 3C 01 01 17 69 74 61 13 50 72  .....?<...ita.Pr
    0020:  6F 67 72 61 6D 6D 69 20 54 56 20 42 42 20 53 41  ogrammi TV BB SA
    0030:  54 04 2B 01 2F 00 69 74 2E 6D 65 64 69 61 73 65  T.+./.it.mediase
    0040:  74 2E 73 63 68 65 64 75 6C 65 73 74 76 2E 50 6F  t.schedulestv.Po
    0050:  72 74 61 6C 65 4C 69 67 68 74 58 6C 65 74 03 00  rtaleLightXlet..
    0060:  02 48 00 03 01 30 68 74 74 70 3A 2F 2F 6D 68 70  .H...0http://mhp
    0070:  2E 64 67 74 76 2E 6D 65 64 69 61 73 65 74 2E 69  .dgtv.mediaset.i
    0080:  74 2F 61 70 70 6C 2F 50 72 6F 67 72 61 6D 6D 69  t/appl/Programmi
    0090:  54 76 53 61 74 2F 01 12 50 72 6F 67 72 61 6D 6D  TvSat/..Programm
    00A0:  69 54 76 53 61 74 2E 7A 69 70                    iTvSat.zip

* AIT, TID 116 (0x74), PID 7878 (0x1EC6)
  Version: 0, sections: 1, total size: 77 bytes
  - Section 0:
    TIDext: 1 (0x0001)
    0000:  F0 00 F0 3D 00 00 00 0B 1A B6 01 F0 34 02 05 00  ...=........4...
    0010:  01 01 7F 0A 00 09 05 00 01 01 00 02 FF 3C 01 01  .............<..
    0020:  10 65 6E 67 0C 4C 61 75 6E 63 68 65 72 20 53 41  .eng.Launcher SA
    0030:  54 03 00 04 0C 01 2F 00 62 64 2E 42 44 58 6C 65  T...../.bd.BDXle
    0040:  74                                               t
~~~

I can provide a stream capture if you do not have reception capabilities.

Then, you may implement some interesting descriptors, one by one.

Finally, it would be nice to add some non-regression tests in repository `tsduck/tsduck-test` (the input reference TS could be stripped of all non-AIT-related PID's to reduce its size).

Let me know if you plan to continue in this direction.
Thanks for your cooperation!

(just out of curiosity, I have a few contacts there)

ElyKar, 2018-02-13:
Thanks for the help !
I will then try to implement properly the AIT and make a good PR for it.

lelegard, 2018-02-13:
OK, good.

In the meantime, here is a [sample TS file](https://github.com/tsduck/tsduck/files/1719938/hb13-ts6000-ait.zip) from TS 6000 on Hot Bird 13°E. It contains only the basic PSI and AIT's. All other voluminous PID's have been removed. There are only 3 AIT's but they are the only ones on the TS. Waiting longer does not bring any other.

lelegard, 2018-02-18:
Hi @ElyKar 

If you are implementing AIT, I advise to merge with the latest content of the repository before going further because I just introduced some refactoring at `AbstractTable` level, with a small impact on the implementation of tables like PMT and, presumably, AIT.

The reason is the following. I just noticed that tables such as AIT, INT (IP/MAC Notification, EN 301 192) and UNT (SSU, Update Notification, TS 102 006) define specific descriptors which can be used only in their respective table and which, by a very unfortunate decision at some point in the past, have tags which overlap with MPEG-defined descriptors, in the range 0x00-0x3F.

This must be addressed in a specific way. TSDuck maintains a tables & descriptors factory which contains a dynamically created repository of descriptors. A descriptor is defined in one of the following categories:
- Standard MPEG / DVB
- Private in the context of DVB, with a preceding private data specifier
- DVB extension descriptor

All these cases are addressed by inspecting the descriptor list, regardless of the table into which the descriptor loop is inserted. With the AIT / INT / UNT specific descriptors, we have another category : descriptor with a standard MPEG / DVB tag but which has a different definition in a specific table.

This case cannot be addressed by inspecting the descriptor list alone, we need to know the tag of the table. To solve this, I made a modification to class `DescriptorList` so that, now, an instance of `DescriptorList` is tightly bound to its enclosing `AbstractTable`. As usual, I adopted a safe implementation, meaning that the code cannot compile if the developer forgets to bind them and the relationship between the two is automatically transferred when necessary. 

For the application using the tables, the API did not change. For implementers of tables containing descriptor loops, the link should be made in the constructor of the table. Do not worry, if don't do it correctly, it won't compile and you should not be afraid of forgetting something.

Let me know if you have problems.

ElyKar, 2018-02-19:
Thanks for the notification, I will look into it right away.
I started implementing the AIT but for now it was more about understanding the structure of tsduck and playing with it. Also, I had to get a bit more familiar with C++ because I never had the opportunity to use it in any project (until now).
I plan to spend a good chunk of time on that in the next few days, if everything goes well I should have a first functional version of the AIT after that.

lelegard, 2018-02-19:
Good luck, That will be a good exercise!

Yesterday, I forgot to push another bunch of commits (done now) to support these AIT / INT / UNT specific descriptors. The support infrastructure has been done. I have implemented only one, the simplest one, the application_usage_descriptor for the AIT. It is extremely simple (it took about 10 mn to implement, compilation included) but its registration mechanism can be used as a template for other table-specific descriptors.

If you never used C++ in the past, take care. C++ is extremely flexible, way too much flexible in fact. I have seen many different types of C++ coding, from crappy C-like in a 70's fashion to highly structured Java-like code. TSDuck clearly sticks to the latter. So please, preserve that style for any contribution.

It you used to work in C, forget everything you learned :) If you worked in Java, you should be more comfortable. I have implemented in TSDuck many classical design patterns which accelerate the coding and the robustness of the code. Use them.

Use the doxygen-generated documentation (`make doygen` or online [here](https://tsduck.github.io/doxy/html/)). The class-diagrams are usually helpful to understand the structure, not need to know UML.

Feel free to ask questions here.

lelegard, 2018-03-04:
Hi Tristan,

I just integrated your pull request #34 with your implementation of the AIT. Thanks a lot for your contribution. I will close this issue #33. If you plan to implement AIT descriptors, please directly send a pull requests with all descriptors you implemented.

Thanks
-Thierry

closed 2018-03-04 by lelegard