tsduck/tsduck #35: What is the best way to get values of signalLocked, signalQuality and signalStrength of BDA tuners under Windows

anilyildiz, 2018-02-23:
Hello,

First off, thank you for your hard work on this. It's useful in so many levels.

I'm developing a different kind of application than a video / audio one. Basically it consists of using an mpeg transport stream as a way of transferring any data. I'll be using tsp with plugins to output data stream from a dvb-s tuner (TBS 5922 and/or Technisat Skystar USB HD, both of which works fine under Windows for my use case). An **example** command line params I use are `-m --debug --buffer-size-mb 1 --bitrate-adjust-interval 1 -I dvb --tune 11111:v:0:22222 -P pcrbitrate -P regulate -P filter --pid 123 -O ip 192.168.1.88:5001`

I also need to get values of a unique string associated with the hw. Preferably mac address or a serial of some sort. Along with signal lock, quality and strength values.

What i tried so far is:

1. Use tsp with params provided above. While it is running, execute `tslsdvb -v`. It reports signal is not locked and according to the code, when signal quality and strength are below -1, prints nothing. Then prints graph details.

2. Just to see if it was possible within the `tsplugin_dvb`, I inserted a line  on method `size_t ts::DVBInput::receive (TSPacket* buffer, size_t max_packets)` right before `return`. 
Which is `tsp->info(u"SignalLocked=%s SignalQuality=%s SignalStrength=%s", { _tuner.signalLocked(*tsp), _tuner.signalQuality(*tsp), _tuner.signalStrength(*tsp) });`. 
I did this to see the values while tsp was streaming data over network. 
It always prints `* dvb: SignalLocked=0 SignalQuality=-1 SignalStrength=-1`

I know that the signal is locked and tsp is working because it is streaming to ip correctly. Also some other application (TSReader Lite, for completeness) can also lock and stream data AND it can show signal stats. So it should be possible with the current BDA drivers. 

What am I doing wrong? Could you point me to the right direction please?

lelegard, 2018-02-23:
Hi

You are not doing anything wrong. Windows is doing wrong for you :)

I have been using DVB tuners on Linux and Windows for 10 years and the experience has always been the same : all returned values, signal characteristics or tuning parameters, are incorrect and meaningless on Windows. 

If you do need these values, use Linux. I do not know any other solution. 

Concerning signal strength and quality, be aware that various Linux drivers may return different unit scales : values in 0-100%, in range 0-255, in dB, etc. So you may have to calibrate first for each tuner model. 

Concerning a unique value, like a MAC address, I am not aware of anything like that.

anilyildiz, 2018-02-23:
Alright, thanks for taking your time and writing a response.

closed 2018-02-23 by anilyildiz