tsduck/tsduck #36: HLS Support

ballerabdude, 2018-02-23:
TSDuck is great how hard would it be to support an HLS stream?

lelegard, 2018-02-24:
Hi

I have been thinking about DASH and HLS for a while now. However, TSDuck has been designed around transport streams. So, it won’t be so easy. 

Call to all users:

If you have ideas about the adaptations of the TSDuck architecture for HLS and DASH, please share them here.

ballerabdude, 2018-02-24:
So HLS now support fmp4(.m4s) which is the same container that Dash uses. I'm new to TSDuck but from my understanding is it only for mpeg ts?

ballerabdude, 2018-02-24:
I'm an idiot it's in the title. I think we need to make it easier to stream and hls into TSDuck. Currently, now I have to download each ts segment to analyze. What program can I use to stream the segments from a layer manifest into TSDuck. I couldn't do it with ffmpeg.

ballerabdude, 2018-03-02:
I figured out how to pipe hls into TSDuck using ffmpeg 
```bash
tsdump <(capture /dev/stdout | ffmpeg -i https://example.com/hls/master.m3u8 -f mpegts -) 2> /dev/null
```
Maybe that can be included in the Readme

lelegard, 2018-03-10:
Hi,

Yes, TSDuck commands can take an MPEG-TS as input, so any command like this will work:
~~~
ffmpeg ... -f mpegts - | tsp ...
~~~

closed 2018-03-10 by lelegard

lelegard, 2018-11-25:
Better late than never, TSDuck support for HLS has been added in version 3.16.