tsduck/tsduck #37: EIT Schedule table not present

PHHENS, 2018-03-02:
Hi Thierry,

When doing :
tsp -I ip 226.0.0.1:1234 -P tables --pid 18 --binary-output eit.bin -O drop
tstabcomp -d eit.bin -v
cat eit.xml
I only have 0x4E (Present/following) but no 0x50 (Schedule) ?

Thanks
Phil

lelegard, 2018-03-02:
Hi Phil,

EIT schedule are optional. Their presence is based on the good will of the operator. 

On satellite, for instance, the total EPG is so big that EITs are present on only a few transponders, possibly only one. 

On IP TV, the EPG is generally never multicasted. In that case, all STB’s are connected by definition. The content of the EPG is pulled by the STB using HTTP to some predefined server of the operator. 

Since your command reads an IP stream, if this is a real IP TV stream (and no a retransmission of a broadcast TS), it is perfectly normal to have no EIT schedule. 

Thierry

lelegard, 2018-03-02:
And by the way EIT schedule are 0x50 to 0x5F, not only 0x50. But you probably knew it already.

PHHENS, 2018-03-02:
> 5493e4bb","name":"GitHub"},"entity":{"external_key":"github/tsduck/ts
> duck","title":"tsduck/tsduck","subtitle":"GitHub
> repository","main_image_url":"https://cloud.githubusercontent.com/ass
> ets/143418/17495839/a5054eac-5d88-11e6-95fc-
> 7290892c7bb5.png","avatar_image_url":"https://cloud.githubusercontent
> .com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-
> b52498112777.png","action":{"name":"Open in GitHub","url":"https://gi
> thub.com/tsduck/tsduck"}},"updates":{"snippets":[{"icon":"PERSON","me
> ssage":"@lelegard in #37: Hi Phil,\n\nEIT schedule are optional.
> Their presence is based on the good will of the operator. \n\nOn
> present on only a few transponders, possibly only one. \n\nOn IP TV,
> using HTTP to some predefined server of the operator. \n\nSince your
> EIT schedule. \n\nThierry "}],"action":{"name":"View Issue","url":"ht
> tps://github.com/tsduck/tsduck/issues/37#issuecomment-369959974"}}}

Hi Thierry,
My stream HAS EIT Schedule :
Summary-------TS id:         6 (0x0006)Last UTC:      2018/03/02
13:24:12EITp/f actual: 32EITp/f other:  0EITs actual:   72EITs
other:    0
TS      Services  With EITp/f  With EITs  EPG days------  --------  ---
--------  ---------  ------
--
Actual         1            1          1         8Other          0     
       0          0         0
A/O  TS Id   Srv Id  Name  EITp/f  EITs  EPG days---  ------  ------  -
---  ------  ----  ------
--Act  0x0006  0x0601  TF1   yes     yes          8
But it is not shown when doing :
tsp -I ip 226.0.0.1:1234 -P tables --pid 18 --binary-output eit.bin -O
droptstabcomp -d eit.bin -vcat eit.xml
Thanks for your help

lelegard, 2018-03-02:
One hypothesis : you have EIT _sections_ but not complete EIT _tables_. Try plugin `tables` with option `-a` and use `tstabdump` on the binary file to view all sections. 

XML files must contain complete tables. Not individual sections. EIT’s are strange kinds of tables with widely spaced sections. 

A complete table is defined by the set of all sections with same table id and same table id extension, all section numbers up to _last section number_.

lelegard, 2018-03-16:
Hi @PHHENS,

You may want to try the new options `-P tables --pack-all-sections --all-once` on the EIT PID. These new options force each EIT section to become a complete table (by forcing _section_number_ and _last_section_number_ to zero). Thus, individual sections from incomplete tables can be saved as XML. Since EIT's are rarely complete, it is probably the best way to save them in XML format.

These options will be present in version 3.9, to be released this week-end I hope. Or you may pull the repository now and rebuild TSDuck.

lelegard, 2018-03-23:
Create one or more XML files containing your EIT's or modify the extracted XML files. Then, use plugin `inject` to inject them in PID 18 of an existing stream. Make sure to read the user's guide first, reference section of the plugin and various examples in the subsequent chapter.

closed 2018-04-03 by lelegard