tsduck/tsduck #38: tsp read commands from file

haraldjordan78, 2018-03-02:
Dear Thierry, 
first i'd like to ask whats the best way to give you the code of my changed PMT and MUX plugins, i added some features for my project and it would be great to have them merged in case you like them:-) Do you need to give me rights for the branching and pull request?

Second, i do create tsp command line programmatically, it can get very lengthy. More lenghty than a windows shell is able to take. Therefore it would be great if tsp was able to read the commands from file instead of parameters. 
My strategy for this would be to check if the first argument is a file and if yes, then modify argc and argv accordingly. But i fear you would not like it that way (it is a bit hacky style too). So could you recommend the way and spot where you would like me to implement this?

lelegard, 2018-03-02:
Hi Harald. 

First, thank you for your contribution. 

Please submit pull requests, preferably one per plugin (unless they share common new code in libtsduck). 

Concerning command line length, my first suggestion would be to use a real operating system, meaning switching from Windows to some flavor of Unix ;-))

More seriously, a common pattern to get command line options from a file is to use `@filename` anywhere in the command line. This replaces the `@filename` with the content of `filename`, each line being one option.

lelegard, 2018-03-02:
Concerning the `@filename` syntax, it would be preferable to implement it in class `ts::Args` so that any command could benefit from it. 

I will do that later if you prefer not to mess with that essential class.

lelegard, 2018-03-05:
Command line redirection using `@filename` has been added. See the user's guide, section 3.1.2 Partial command line redirection from a file.

closed 2018-03-10 by lelegard