tsduck/tsduck #41: Call to users: looking for streams with MPE

lelegard, 2018-03-09:
Dear TSDuck users,

I am looking for real-life captured transport streams with MPE streams. It possible, I would appreciate encapsulated TS inside UDP, being encapsulated inside MPE streams, inside TS.

I you have such streams, please post them (or link to them) here.

Recent commits of TSDuck have added detailed MPE signalisation and a new plugin named `mpe` which extracts MPE streams (saved on files or forwarded as UDP for the moment). To add more interesting features on this plugin, I need real test streams.

Thank you in advance for your cooperation.

rezna11, 2018-04-20:
Is this what you're looking for? 

https://mega.nz/#!FkRxiQAD!HmQVAD0GpaWktt4_Qbj5K_MmF56vqkuF9wBO4uUfI2E
https://mega.nz/#!I5QViLRK!_nTex16w-3oOftHbKt4wyfkX2ymtksbRJStGBdv2iag

It's a 1 minute from Tricolor OTT at 36E and some C-band stream, probably from 45W or close to it. I didn't have a chance to try MPE plugin yet so I'm not sure what would be the expected output from this. In the Tricolor OTT there are clearly video streams in it but I have no idea if there's any way to extract those properly. Please let me know if the files recorded are useful in any way.

lelegard, 2018-04-20:
Hi. 

Thank you. I am away this week and will download the file next week. I heard that Tricolor was using MPE to feed ABR streams for local pseudo-OTT redistribution. This is probably that stream. I will let you know.

lelegard, 2018-04-22:
Hi Jan,

Thanks for the files. I just downloaded them.

The first file is clearly an MPE TS with correct syntax, correct INT, etc. I do not know yet how to interpret the content of each multicast stream. Using plugin `mpe`, I extracted the raw UDP content of one stream and tried to extract a valid TS from it.
~~~
tsp -I file 36.0E_12168.835_V_27500.ts -P mpe -d 224.30.0.20 -o 224.30.0.20.udp -O drop
tsresync 224.30.0.20.udp -c -m $((3*188)) -o 224.30.0.20.ts
~~~
The strange usage of `tsresync` is a tweak. It takes a binary file with TS packets in the middle of various types of encapsulations and tries to extract as many valid TS packets as it can. This is a bit rude but it works on Wireshark capture files for instance (capture of an IP-TV UDP multicast stream). Here, I get a more or less valid SPTS file with a clearly identified service, PAT, PMT. But VLC plays it as garbage. So, I need to investigate a bit more.

The second file is not MPE, it is an RCS stream (Return Channel Satellite) as defined in ETSI EN 301 790. I have never investigated that kind of stream and I am not familiar with this standard. I keep this file for reference and I will probably get back to it later.

Thanks again.

lelegard, 2018-04-22:
In the MPE stream, using a new option `--sync-layout` in plugin `mpe`, we can display the layout of the 0x47 bytes in the UDP payload. The number between two `[47]` is the number of bytes between the two 0x47.
~~~
$ tsp -I file 36.0E_12168.835_V_27500.ts -P mpe -d 224.30.0.20 --sync-layout -O drop
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  101 [47] 78 [47] 187 [47] 187 [47] 187 [47] 187 [47] 187 [47] 187 [47] 57 [47] 53 [47] 47
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  33 [47] 14 [47] 187 [47] 40 [47] 146 [47] 187 [47] 125 [47] 61 [47] 187 [47] 187 [47] 187 [47] 103
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  104 [47] 166 [47] 20 [47] 133 [47] 53 [47] 187 [47] 60 [47] 40 [47] 85 [47] 187 [47] 182 [47] 4 [47] 187 [47] 47
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  160 [47] 187 [47] 163 [47] 10 [47] 12 [47] 136 [47] 42 [47] 7 [47] 64 [47] 122 [47] 187 [47] 187 [47] 179
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  28 [47] 57 [47] 60 [47] 40 [47] 27 [47] 61 [47] 77 [47] 47 [47] 187 [47] 153 [47] 33 [47] 187 [47] 187 [47] 187 [47] 123
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  84 [47] 108 [47] 78 [47] 81 [47] 83 [47] 21 [47] 118 [47] 68 [47] 153 [47] 33 [47] 187 [47] 187 [47] 187 [47] 67
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  140 [47] 187 [47] 145 [47] 41 [47] 126 [47] 60 [47] 25 [47] 90 [47] 70 [47] 58 [47] 128 [47] 129 [47] 57 [47] 187 [47] 11
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  196 [47] 28 [47] 158 [47] 77 [47] 58 [47] 50 [47] 54 [47] 55 [47] 76 [47] 187 [47] 187 [47] 187 [47] 143
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  64 [47] 187 [47] 187 [47] 5 [47] 181 [47] 187 [47] 29 [47] 40 [47] 116 [47] 187 [47] 127 [47] 59 [47] 87
* mpe: PID 0x07D2 (2002), src: 10.10.30.206:5050, dest: 224.30.0.20:5050 (01:00:5E:1E:00:14), 1468 bytes, fragment: 0x0000
  120 [47] 187 [47] 187 [47] 5 [47] 181 [47] 118 [47] 68 [47] 126 [47] 60 [47] 160 [47] 26 [47] 22 [47] 92 [47] 22 [47] 48 [47] 31
...
~~~

We can see the 187 values indicating packets without 0x47 in the middle. Otherwise, by grouping adjacent values, we can identify 188-bytes packets with a 0x47 in the middle.

But the global layout is very different from an IP-TV stream with a fixed size prefix and where the last TS packet always ends exactly at the end of the UDP payload.

rezna11, 2018-04-22:
Googling around, I found this topic: http://www.sat-universe.com/showthread.php?t=301973
which leads to http://www.mediafire.com/file/d9q3w3ucy4npe9x/OTTextractor%282%29.rar and with that, working video files are extracted from the MPE stream... so it can be done, somehow :)

lelegard, 2018-04-22:
Thanks for the link, I will have a look

lelegard, 2018-04-23:
I looked deeper inside the decapsulated UDP streams and found that the content format is Apple HLS. Apparently, the UDP packets contain a 20-byte header (proprietary?) followed by payload. The concatenated payloads form an original HTTP response stream, with HTTP headers, HLS manifests and HLS segments (which are in fact TS files).

I originally spotted the TS packets from the HLS segments but missed the HTTP headers and HLS manifests since they represent only a very small part of the data.

To detect missing DSM-CC sections/ MPE datagrams, a continuity counter seems to be present in the header.

The danger of that stack of protocols (TS -> sections -> DSM-CC -> MPE -> UDP -> HTTP -> HLS -> back to TS) is the impact of a missing TS packet in the satellite stream. Instead of losing one video TS packet (limited impact),  you loose one DSM-CC section, meaning several effective TS packets plus adjacent overlapping TS packets, not including the potential damages on the HLS structure. Add the fact that the reception takes place at a high latitude (Russia). It really looks like looking for trouble.

rezna11, 2018-05-20:
here's another stream but I'm not sure if it's really the kind of MPE that tsduck handles... looks like DStv service for mobile phones, recorded from 68E 12610H 
https://mega.nz/#!clph1TzD!n-YCATnLV8vYRZLez2mhv6WRAAJouRSHaR2cnDQ3kkg

lelegard, 2018-05-21:
Hi Jan,

Thanks for the stream. TSDuck can extract UDP packets from this stream. The MPE structure and the INT seem correct. Currently, I have no idea of the format of the data inside the UDP datagrams.

Looking at the MPE stream for the service named _Discovery_ for instance:
~~~
tsp -I file 68E_12610.912_H_6710.ts -P mpe --pid 2107 --log --dump-udp -O drop
~~~

The datagrams do not contain TS packets, with any kind of encapsulation. All datagrams start with either `80 60`, `80 E0` or `80 E1`. There are some repetitive patterns. Any idea of the format?

rezna11, 2018-05-21:
no idea, the only mention of something that could be this related to is DVB-H OMA/BCAST (https://www.naijatechguide.com/2008/05/nigeria-dstv-mobile-from-mtn.html)

lelegard, 2018-06-19:
Thanks for your efforts in searching, capturing and providing MPE streams.
Closing this topic now.

closed 2018-06-19 by lelegard