tsduck/tsduck #42: how to extract the NIT from ts file in xml format.

sachin0606, 2018-03-20:
below example gives the xml format NIT from stream

from user guide extraction of NIT: 
Capture a table from a stream directly in XML format
tsp -I dvb ... \
 -P tables --pid 16 --tid 0x40 --max 1 --xml nit.xml \
 -O drop
• Manually edit the XML file with a text editor

but how to extract the same from TS file.

and how to inject the same and generate stream in UDP ip?

sachin0606, 2018-03-20:
can use below command for getting xml from file.

tsp -I file 000.ts -P tables --pid 16 --tid 0x40 --max 1 --xml nit.xml

where 000.ts => file name

thanks. i get it. now only needed how to stream modified NIT in UDP

lelegard, 2018-03-20:
Hi,

Use the `file` input plugin instead of `dvb` to read from a file. Alternatively, if the only thing you need is to extract tables from a file, you can use the command `tstables`. It has the same options as the plugin `tables` and does the same thing but works directly on a file instead of being a plugin in a chain.

To inject the table in a stream and send it using UDP, use the output plugin `ip`. I suggest you read the examples in the user's guide about producing IP multicast. There are some traps to avoid, for instance accidentally broadcasting at the speed of the input file reading, which is way too fast. Details and examples are in the user's guide, specifically 5.2.3 and 5.2.4.

sachin0606, 2018-03-21:
wanted to stream the BAT and NIT from the stream, currently i am getting the NIT from existing ts file. and modified accordingly but how to extract the BAT table ? and how to combine both and stream it.

lelegard, 2018-03-21:
Hi

The BAT is on PID 17. You capture it the same way as the NIT. Just take care that PID 17 carries several tables : SDT actual and other, and the BAT’s for different bouquets. So you have to filter the one you need. 

To inject it, first collect all tables for the PID. Then inject all of them using the plugin inject. 

At this point, you really need to read the user’s guide. You will find all information you need here.

sachin0606, 2018-03-21:
Thanks a lot. and some thing i want to tell, user guide giving all information but missing examples within it. and also i find PID for the BAT but i didnt get that in user guide. 

any way. i really thanks to you for support.

lelegard, 2018-03-21:
Well, TSDuck User's Guide is a user's guide for ... TSDuck.

It is not a tutorial or a lesson on Digital TV. Otherwise, I would have to write thousands of pages. You need to get more familiar with MPEG transport streams and DVB signalization. Fortunately, DVB standards can be obtained for free on the [ETSI Standards page](http://www.etsi.org/standards-search). You should start with ETSI EN 300 468, the specifications of DVB SI ([direct link](http://www.etsi.org/deliver/etsi_en/300400_300499/300468/01.15.01_60/en_300468v011501p.pdf)).

LCN are unfortunately not specified, neither by MPEG nor DVB. There are several private descriptors to define LCN. The most commonly used in Europe is defined by EACEM TR-030 (not for free I think). Below is the NIT I just captured on the French terrestrial TV network. It uses the EACEM private descriptor for LCN. But you should first get information of the way the NIT is signaled on your target network. It may not be the same.

Please restrict this issue tracking system for questions about TSDuck itself. There are other forums for Digital TV techniques with many people willing to help.
~~~
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT version="30" current="true" network_id="0x20FA" actual="true">
    <!-- PID 0x0010 (16) -->
    <network_name_descriptor network_name="F"/>
    <transport_stream transport_stream_id="0x0001" original_network_id="0x20FA">
      <terrestrial_delivery_system_descriptor centre_frequency="42,949,672,950" bandwidth="8MHz" priority="HP" no_time_slicing="true" no_MPE_FEC="true" constellation="64-QAM" hierarchy_information="0" code_rate_HP_stream="0x05" code_rate_LP_stream="3/4" guard_interval="1/8" transmission_mode="8k" other_frequency="false"/>
      <private_data_specifier_descriptor private_data_specifier="eacem"/>
      <logical_channel_number_descriptor>
        <service service_id="0x0101" logical_channel_number="2" visible_service="true"/>
        <service service_id="0x0104" logical_channel_number="14" visible_service="true"/>
        <service service_id="0x0105" logical_channel_number="19" visible_service="true"/>
        <service service_id="0x0106" logical_channel_number="27" visible_service="true"/>
        <service service_id="0x0113" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x0115" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x0119" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x011A" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x0111" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x0112" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x011F" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x0120" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x0124" logical_channel_number="3" visible_service="true"/>
        <service service_id="0x0143" logical_channel_number="33" visible_service="true"/>
        <service service_id="0x0144" logical_channel_number="33" visible_service="true"/>
        <service service_id="0x0170" logical_channel_number="30" visible_service="true"/>
        <service service_id="0x0171" logical_channel_number="31" visible_service="true"/>
        <service service_id="0x0172" logical_channel_number="32" visible_service="true"/>
        <service service_id="0x0173" logical_channel_number="33" visible_service="true"/>
        <service service_id="0x0174" logical_channel_number="34" visible_service="true"/>
        <service service_id="0x0175" logical_channel_number="35" visible_service="true"/>
        <service service_id="0x0176" logical_channel_number="36" visible_service="true"/>
        <service service_id="0x0177" logical_channel_number="37" visible_service="true"/>
        <service service_id="0x0178" logical_channel_number="38" visible_service="true"/>
        <service service_id="0x0145" logical_channel_number="32" visible_service="true"/>
        <service service_id="0x0146" logical_channel_number="32" visible_service="true"/>
      </logical_channel_number_descriptor>
      <service_list_descriptor>
        <service service_id="0x0101" service_type="0x01"/>
        <service service_id="0x0104" service_type="0x01"/>
        <service service_id="0x0105" service_type="0x01"/>
        <service service_id="0x0106" service_type="0x01"/>
        <service service_id="0x0113" service_type="0x01"/>
        <service service_id="0x0115" service_type="0x01"/>
        <service service_id="0x0119" service_type="0x01"/>
        <service service_id="0x011A" service_type="0x01"/>
        <service service_id="0x0111" service_type="0x01"/>
        <service service_id="0x0112" service_type="0x01"/>
        <service service_id="0x011F" service_type="0x01"/>
        <service service_id="0x0120" service_type="0x01"/>
        <service service_id="0x0124" service_type="0x01"/>
        <service service_id="0x0143" service_type="0x01"/>
        <service service_id="0x0144" service_type="0x01"/>
        <service service_id="0x0170" service_type="0x01"/>
        <service service_id="0x0171" service_type="0x01"/>
        <service service_id="0x0172" service_type="0x01"/>
        <service service_id="0x0173" service_type="0x01"/>
        <service service_id="0x0174" service_type="0x01"/>
        <service service_id="0x0175" service_type="0x01"/>
[... cut ...]
        <service service_id="0x0A05" logical_channel_number="23" visible_service="true"/>
      </logical_channel_number_descriptor>
      <service_list_descriptor>
        <service service_id="0x0A01" service_type="0x19"/>
        <service service_id="0x0A03" service_type="0x19"/>
        <service service_id="0x0A02" service_type="0x19"/>
        <service service_id="0x0A04" service_type="0x19"/>
        <service service_id="0x0A05" service_type="0x19"/>
      </service_list_descriptor>
    </transport_stream>
  </NIT>
</tsduck>
~~~

sachin0606, 2018-03-21:
thanks for brief instruction. and guidance. will refer it. 

the given structure of NIT can be used with TSDuck?

lelegard, 2018-03-21:
Yes, it was produced by TSDuck and consequently usable by TSDuck.

sachin0606, 2018-03-21:
THNAKS....! got all done with two files one is BAT and one is NIT. now how can i compile transmit through UDP?
as per guidance i can use 
tstables  --ip-udp 224.222.18.9:9000 NIT.xml

or i have to compile and generated .bin file is to be streamed? and as two separate file how to transmit?

lelegard, 2018-03-21:
I told you to read the user’s guide. You obviously didn’t. So my last word in this thread can only be RTFM.

sachin0606, 2018-03-23:
thanks for you support. its really helped me to study and implement.

closed 2018-04-03 by lelegard

jimbo2018, 2022-09-05:
Hi.  I have tried for ages to get TSDuck to work but it willo not work on Windows.  It oly works on linux.  My computer is Windows. So there is no way I can use the program.

I have managed to extract PID 16 and 17 from DTT but cannot use the file. TSReader will not work, DVBStreamExplorer doesn't work and TerraToll does not work.

They ALL look for a TS live stream and it is impossible to import a file to them.

Please advise how I can get TSDuck to work on Windows 10 laptop. And which file do I download to get it working please?

Thank you

lelegard, 2022-09-05:
Hi @jimbo2018 

Are you sure this is the right thread? This one has been closed more than 4 years ago.

Otherwise, TSDuck works on Windows 10. Use the installer from here: https://tsduck.io/download/tsduck/

If you have a specific problem with TSDuck on Windows, please provide a precise description of the problem.

jimbo2018, 2022-09-05:
Thank you.

Had trouble with other programs but it is tsduck.i would prefer to use but I hope to get it sorted soon.

Thanks again. >