tsduck/tsduck #43: tsp: ip(input): Second IP input can't use a multucast group with the same port number

yaneti, 2018-03-22:
Linux: tsduck-3.9-619.fc27.x86_64
```
# tsp -m -I ip 224.168.16.211:2000 -P analyze -i3 --normalized -O drop
..... everything ok...
```
Start a second tsp on the same host
```
# tsp -m -I ip 224.168.16.222:2000 -P analyze -i3 --normalized -O drop
* Error: ip: error binding socket to local address: Address already in use

```

On a related note the way the ip input module currently works with multicast on linux results in sending igmp joins on _all_ interfaces, regardless of the routing table, which is not how multicast usually works.   eg for comparison see the way multicat or socat work

yaneti, 2018-03-22:
So I just noticed the -r --reuse-port option that  solves this particular issue for me. 
I am still generally confused about the mutlicast binding options because they don't match what I expect from other mutlicast software on Linux

To give you an example adding the -l <local_address> option that would allow me to isolate the particular igmp to a specific interface seems to prevent the plugin from accepting the above mentioned groups, even tho they are flooded in my general direction.

closed 2018-03-22 by yaneti

lelegard, 2018-03-22:
Hi @yaneti 

Using the same UDP or TCP port from two concurrent sockets has always been disallowed by default on all Unix systems (and probably Windows as well). Allowing this is the purpose of the socket options `SO_REUSEADDR` or `SO_REUSEPORT` or both (depending on the Unix flavor), which is implemented through the option `--reuse-port` in input plugin `ip`. So, the solution you found is the right one.

Routing tables usually know nothing about which interface will receive which multicast traffic. So, the application has to specify which interface should be used. I am afraid I do not fully understand what you meant about `-l <local_address>` in your last post. The option `--local-address` already exists in the input plugin `ip`. Is it what you were refering to?

Using all interfaces by default was a usability choice. Most users do not manually configure routing tables on their local machines. Moreover, I have encountered in the past systems were multicast addresses in the local routing tables applied to _outgoing_ traffic only and was ignored when listening that specific multicast address on the _any address_ interface. I do not know what is the current behaviour of recent Linux kernels. And honestly, I don't care. I prefer to ignore system-specific or version-specific behaviours and instead stick to a portable and deterministic behaviour. So, most users do not know the network topology surrounding their system and can't possibly select the right local interface at once. To avoid nervous breakdowns, waiting for traffic which in fact arrives at another interface, the default behaviour is to listen everywhere. It is the most realistic choice for most users. For advanced users who know where to listen, the option `--local-address` is the solution.

lelegard, 2018-03-23:
Hi @yaneti 

See below a good reason why `tsp` listens on all local interfaces for multicast on end-user systems.

You seem pretty knowledgeable in Linux network configuration. So, you will probably have a rational explanation to this and the correct fix to apply (other than `route add` of course). However, this is not the point here. Digital TV engineers are not all IP network specialists. They are focused on their debugging or integration tasks. They have no time to spend learning all the subtle details of multicast routing configurations. They would probably love to, but they just don't have the time.

Here is a Fedora 27 system:
~~~
$ uname -a
Linux vmfedora 4.15.10-300.fc27.x86_64 #1 SMP [...] x86_64 x86_64 x86_64 GNU/Linux
~~~

With 5 local interfaces (including localhost):
~~~
$ ifconfig | grep 'inet '
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 0.0.0.0
        inet 10.0.2.15  netmask 255.255.255.0  broadcast 10.0.2.255
        inet 192.168.56.102  netmask 255.255.255.0  broadcast 192.168.56.255
        inet 127.0.0.1  netmask 255.0.0.0
        inet 192.168.122.1  netmask 255.255.255.0  broadcast 192.168.122.255
~~~

From another machine in network 192.168.56.0, we send packets to 230.1.2.3:7777. Using WireShark or debug traces in `tsp` when we receive the stream, we can see packets source: 192.168.56.1:59160, destination: 230.1.2.3:7777.

Listening on all interfaces (the default), we receive packets:
~~~
$ tsp -v -I ip 230.1.2.3:7777 -P count -i 1000 -O drop
* ip: joining multicast group 230.1.2.3:7777 from local address 10.0.2.15
* ip: joining multicast group 230.1.2.3:7777 from local address 192.168.56.102
* ip: joining multicast group 230.1.2.3:7777 from local address 192.168.122.1
* ip: joining multicast group 230.1.2.3:7777 from local address 172.17.0.1
* count: 2018/03/23 18:19:13.864, counted: 1,000 packets, 2,816,479 b/s, total: 1,000 packets, 2,816,479 b/s
* count: 2018/03/23 18:19:14.408, counted: 2,000 packets, 2,764,705 b/s, total: 2,000 packets, 2,764,705 b/s
~~~

The four messages `joining multicast group ...` indicate that `tsp` sends four `IP_ADD_MEMBERSHIP` to the four displayed local addresses. Now, we run four individual commands with a `--local-interface` option, one for each of the local addresses. Each time, we see only one of the `joining multicast group ...` message.

Guess what? None of the four commands receives any packet. None. So, if we listen on all local interfaces, it works. But it never works when specifying one interface.

I will be very interested in understanding why. Because I am curious. But you have to understand that most Digital TV engineers are just interested in making the thing works and let `tsp` hide the dirt without system configuration.

Just because you implicitely wished it, I just added a `--default-interface` option, letting the system decide, using only one `IP_ADD_MEMBERSHIP` with zero as local address. Here, we rely on the routing tables.

~~~
$ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         gateway         0.0.0.0         UG    100    0        0 enp0s3
10.0.2.0        0.0.0.0         255.255.255.0   U     100    0        0 enp0s3
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
192.168.56.0    0.0.0.0         255.255.255.0   U     100    0        0 enp0s8
192.168.122.0   0.0.0.0         255.255.255.0   U     0      0        0 virbr0
~~~

As you can guess, with this configuration and `--default-interface`, we receive nothing.

But if we add a route through the appropriate local interface, it works, of course again.
~~~
$ sudo route add -host 230.1.2.3 dev enp0s8
$ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         gateway         0.0.0.0         UG    100    0        0 enp0s3
10.0.2.0        0.0.0.0         255.255.255.0   U     100    0        0 enp0s3
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
192.168.56.0    0.0.0.0         255.255.255.0   U     100    0        0 enp0s8
192.168.122.0   0.0.0.0         255.255.255.0   U     0      0        0 virbr0
230.1.2.3       0.0.0.0         255.255.255.255 UH    0      0        0 enp0s8
~~~

But this is precisely what Digital TV engineers don't want to do. They focus on the IP-TV experimentation, not on the local configuration of every system in the lab.

I will let the new option `--default-interface` because it is useful on systems with carefully configured routing tables.

And if you have an explanation to the weirdness above, please share it here. I am very interested.

yaneti, 2018-03-23:
Hi @lelegard , thanks for taking the time to look at the issue and adding the --default-interface option. 

I think the problem with --local-address is caused by  the specific implementation in ts::UDPReceiver::open,  where in the case of multicast you are actually bind() ing to the local address. While that's how it works for unicast, for multicast its the other way around and the bind should be to the multicast group address that you are expecting to receive.  
The local address plays a role in the ip_mreqn which is why we are observing a correct single IGMP going out, but the incoming multicast packets are not considerd for this socket because of the aforementioned local bind.

Hope that helps.

lelegard, 2018-03-24:
Hi @yaneti 

Interesting answer indeed. I didn't see multicast sockets that way. I have searched a little bit and made some tests. 

You seem to be right on Linux and BSD (macOS). 

On the other hand, on Windows, it is the opposite. The Microsoft documentation for [bind()](https://msdn.microsoft.com/en-us/library/windows/desktop/ms737550(v=vs.85).aspx) explicitly states "_For multicast operations, the preferred method is to call the bind function to associate a socket with a __local IP address__ and then join the multicast group._". And indeed, it works when you bind to 0.0.0.0 (regardless of routing tables, unlike Linux) or when you bind to the local address from which the multicast stream is received. Binding to a multicast address, you don't receive anything.

It is not the first time I saw differences between Linux and Windows on multicast reception. For instance, in the default configuration (bind to 0.0.0.0 and joining membership on all interfaces), when you join a group A but traffic exists on groups A and B (same port), on Windows you just receive A while on Linux you receive everything. And, for the worst part, if you enjoy horror movies, you may have a look in class `UDPSocket` how to call `recvmsg()` on Windows!

Part of the problem is that there is no single formal standard definition for what is a socket and how it is expected to work. So, various O/S diverge in time. When I originally developed the ancestor of TSDuck 12 years ago, multicast was no so widely used. The documentation was vague and testing results prevailed. Now, it is more precise. And the differences between O/S become more precise as well.

I need to thing about how to handle this. The final target is the user. So usability prevails. Portability as well. We cannot please Linux users at the expense of Windows users or vice-versa. Right now, the default behaviour (joining on all interfaces) works everywhere without system configuration. This will remain the default configuration because it serves usability. Now, for specific situations and advanced users, I will probably have to implement and document different options. I need to think twice to do that properly.

Thanks for your explanations.

yaneti, 2018-03-24:
Hi @lelegard

wrt the Linux issue of multiple groups while binding to just one have a look at the IP_MULTICAST_ALL. socket option.  Its on by default on Linux for historical reasons.

lelegard, 2018-03-24:
Hi @yaneti 

In fact, it was simpler than expected. I simply modified the `bind()` in `UDPReceiver`. On Unix when receiving a multicast address, we bind to the multicast address. On Unix when receiving unicast and on Windows in all cases, we bind to the local address when specified or to zero (_any address_) otherwise.

Now, all use cases (no option, `--local-address` or `--default-interface`) work as expected on Linux, macOS and Windows.

In other words, when a local interface is specified on Unix with multicast, we bind to the multicast and send the join request through the local interface. On Windows, we bind to the local interface and send the join the same way. Good to know if you want to write portable software...

So, on Windows and Unix, when specifying the "right" local interface, we receive. When specifying another interface, we receive nothing. By default (all interfaces), we always receive. With `--default-interface`, we rely on the routing tables.

The code modifications are limited to the following:
~~~
-    // The local socket address to bind is the optional local IP address and the destination port
-    SocketAddress local_addr(_local_address, _dest_addr.port());

+    // The local socket address to bind is the optional local IP address and the destination port.
+    // Except on Linux, macOS and probably most Unix, when listening to a multicast group.
+    // In that case, we bind to the multicast group, not the local interface.
+    // Note that if _dest_addr has an address, it is a multicast one (checked in load()).
+    SocketAddress local_addr(
+#if defined(TS_UNIX)
+        _dest_addr.hasAddress() ? _dest_addr.address() : _local_address,
+#else
+        _local_address,
+#endif
+        _dest_addr.port());
~~~

Thanks for having me learnt something on multicast on Linux.