tsduck/tsduck #44: New presentation "introduction to MPEG transport streams"

lelegard, 2018-03-22:
Dear TSDuck users,

I have added a new presentation in the [doc](https://github.com/tsduck/tsduck/tree/master/doc) subdirectory named "_an introduction to MPEG transport streams - all you should know before using TSDuck_" (direct link to [PDF](https://github.com/tsduck/tsduck/raw/master/doc/mpegts-introduction.pdf), [PowerPoint](https://github.com/tsduck/tsduck/raw/master/doc/mpegts-introduction.pptx)).

Nothing new here, just old stuff. The presentation is just a extract of various presentations I made in the past years. But this may be a useful introduction.

Feel free to post your feedback in this thread.

haraldjordan78, 2018-03-24:
This is a great presentation Thierry, 
i am new to Encryption systems, probably i will never deal with it in real life but could you please explain why you have Simulcrypt in this presentation? Is Simulcrypt on the Same level as Nagra and they all are Conditional Access Systems?

By the way, i would like to add my sources of DVB knowledge. 
A DVB Syntax Poster from Tektronix that helped me designing my first DVB splitter in Perl code: 
https://www.tek.com/poster/mpeg-poster-dvb

The only really deep free source of DVB and MPEG syntax i found on the web (unfortunately in german only):
http://www.fr-an.de/projects/mpeg/index.htm

haraldjordan78, 2018-03-24:
Added the pdf link: https://en.wikipedia.org/wiki/MPEG_transport_stream to wikipedia mpeg-ts if you don't mind (see "general introduction to DVB MPEG-TS"):
https://en.wikipedia.org/wiki/MPEG_transport_stream

lelegard, 2018-03-24:
> could you please explain why you have Simulcrypt in this presentation? 

DVB SimulCrypt is a public standard. All DVB-compliant CAS follow that standard (Nagra, NDS, Viaccess, Conax, Irdeto, etc.) This is an essential part of Digital TV systems. Digital TV exists only because TV operators exist. TV operators exist only because they get revenue from their business. Revenue can be enforced only because Conditional Access Systems exist. And all CAS coexist only because DVB SimulCrypt exist. This is why DVB SimulCrypt is an essential part of Digital TV.

Moreover, the `tsp` plugins `scrambler` and `datainject` are fully DVB SimulCrypt compliant. Understanding this standard is required to use them correctly.

That being said, while DVB SimulCrypt is public and DVB CSA-2 common to all systems, each CAS has its own proprietary secret technology with a much more sophisticated crypto than CSA-2. DVB SimulCrypt is the guarantee of interoperability between the CAS and the network.

rezna11, 2018-03-26:
some valuable info in there, especially it's nice to understand how simulcrypt really works at the headend, thank you

closed 2018-05-04 by lelegard