tsduck/tsduck #47: Cannot build TSDuck Binaries in

leobin, 2018-03-28:
Hi,
I try to build TSDuck Binaries from source by following the tutorial

> Execute the command make at top level. The TSDuck binaries, executables and shared objects (.so), are built in the src directory tree in subdirectories release-i386 and release-x86_64 for 32-bit and 64-bit platforms respectively.

But it doesn't work

> make[1]: *** No rule to make target 'default'.  Stop.

I can still build it by going into "src" folder and run make there

lelegard, 2018-03-29:
Hi,

Let's try to find a solution. Maybe there is a missing prerequisite, something I forgot to list in the tutorial.

What is your platform? Linux? Which distro and version?

If this the only message you get or are there preceding messages? Could you try with `make -n` and post the complete output?

In which directory are you when you run `make`? Can you check that you have the three files `Makefile`, `Makefile.tsduck` and `Makefile.common`?

You mention that `make` works in `src`, which is good because it builds the binaries. But could you try `make` in each other directory to see where is fails (the default `make` action at top level is to run `make` recursively in all subdirectories)? With the default target (building the product), having `make` run at least once in `dektec` is important because it downloads the latest version of the Dektec API. If the Dektec API is not present, the build in `src` succeeds but without Dektec support (which is not a problem for you if you do not use Dektec devices).

leobin, 2018-03-29:
Hi, 
I think I figured out the problem. That's because I import tsduck to CLion which automatically create a CMAKE file. 
Which cause some cache trouble when build the application. 
I resolve the issue by:
- remove CMakeLists.txt
- run make clean
- run make 
It works fine now. 
Sorry for any problem, and thanks for your awesome work

closed 2018-03-29 by leobin

jbglaw, 2021-09-30:
Hi,

just to mention: I'm seeing the very same issue. Though no special IDEs or cmake or something like that. Just a plain master checkout from git.

lelegard, 2021-09-30:
Hi @leobin 

Please provide precise information. Everyone, including myself, run make without issue at the top level, several times a day.

Which OS, distro, command, output, after `make distclean`, etc

leobin, 2021-09-30:
Hi @lelegard,
It's a long time ago. I don't have these information anymore. But I could confirm that it worked fine now on my Ubuntu 18.04.

lelegard, 2021-09-30:
Hi @leobin, sorry I meant @jbglaw who just reported a similar issue.

Hi @jbglaw
Please provide precise information. Everyone, including myself, run make without issue at the top level, several times a day.
Which OS, distro, command, output, after make distclean, etc