tsduck/tsduck #49: tsanalyze: tsAVCParser.h:111: size_t ts::AVCParser::remainingBits() const: Assertion `_byte <= _end' failed.

yaneti, 2018-03-30:
Hi,
Attached is  a piece of stream that crashes tsanalyze from a locally built tsduck on f27 
tsduck-3.10-636.fc27.x86_64 . Gzipped because github

```
(gdb) bt
#0  0x00007f42cae66660 in raise () from /lib64/libc.so.6
#1  0x00007f42cae67c41 in abort () from /lib64/libc.so.6
#2  0x00007f42cae5ef7a in __assert_fail_base () from /lib64/libc.so.6
#3  0x00007f42cae5eff2 in __assert_fail () from /lib64/libc.so.6
#4  0x00007f42cc811eb5 in bool ts::AVCParser::readBits<unsigned char, (void*)0>(unsigned char&, unsigned long) () from /usr/bin/tsduck.so
#5  0x00007f42cc8126e1 in ts::AVCSequenceParameterSet::parseBody(ts::AVCParser&) () from /usr/bin/tsduck.so
#6  0x00007f42cc81a480 in ts::AbstractAVCAccessUnit::parse(void const*, unsigned long) () from /usr/bin/tsduck.so
#7  0x00007f42cc81514e in ts::AVCSequenceParameterSet::AVCSequenceParameterSet(void const*, unsigned long) () from /usr/bin/tsduck.so
#8  0x00007f42cc8103e0 in ts::AVCAttributes::moreBinaryData(void const*, unsigned long) () from /usr/bin/tsduck.so
#9  0x00007f42cc90506e in ts::PESDemux::processPESPacket(unsigned short, ts::PESDemux::PIDContext&) () from /usr/bin/tsduck.so
#10 0x00007f42cc9053f9 in ts::PESDemux::processPacket(ts::TSPacket const&) () from /usr/bin/tsduck.so
#11 0x00007f42cc9057b4 in ts::PESDemux::feedPacket(ts::TSPacket const&) () from /usr/bin/tsduck.so
#12 0x00007f42cc96b7eb in ts::TSAnalyzer::feedPacket(ts::TSPacket const&) () from /usr/bin/tsduck.so
#13 0x00007f42c6403f46 in ts::AnalyzePlugin::processPacket(ts::TSPacket&, bool&, bool&) () from /usr/bin/tsplugin_analyze.so
#14 0x0000000000406730 in ts::tsp::ProcessorExecutor::main() ()
#15 0x00007f42cc9a86ae in ts::Thread::ThreadProc(void*) () from /usr/bin/tsduck.so
#16 0x00007f42cb95950b in start_thread () from /lib64/libpthread.so.0
#17 0x00007f42caf2716f in clone () from /lib64/libc.so.6
```
[ekcrash.ts.gz](https://github.com/tsduck/tsduck/files/1862897/ekcrash.ts.gz)

lelegard, 2018-03-30:
Hello @yaneti 

Thank you for reporting this. This is a bug of course. The problem was in class `PESDemux`. It is now fixed in commit 3393cc7e5aa13d5dac34b5bf5beac1d5df7dc556. You may pull and rebuild. Sorry for the inconvenience.

Your stream has been added in the TSDuck test suite as `test-025`. So, should the bug reappear some day, it will be immediately detected when running the non-regression test suite.

The stream seems incorrect to me, but there is no excuse for a crash. All applications shall always recover from unexpected input.

Let me explain what I found.

The video PID of the unique service in the stream is signalled as MPEG-2 video. MediaInfo reports it as MPEG-2 video too. VLC plays it correctly and also reports it as MPEG-2 Video. But, with TSDuck, the content is interpreted as a mixture of MPEG-2 Video and H.264.

It is not unusual that a service is incorrectly signalled, such as H.264 video being identified as MPEG-2 Video in the PMT. For this reason, the tools and plugins in TSDuck analyze the actual content of each PES packet independently of the signalled format from the PMT. The initial start code in the _payload_ of the PES packet gives a hint on the content of the packet. For MPEG-2 Video, the PES _payload_ must start with `00 00 01`. For H.264, the payload must start with `00 00 00 [00...] 01`. Etc.

If you look at the first 16 bytes of each payload of each PES packet in the video PID, you get something like this:
~~~
$ tsp -I file ekcrash.ts -P pes -p 4096 --payload -m 16 -O drop
...
* PID 0x1000, stream_id 0xE0 (Video 0), size: 29481 bytes (header: 19, payload: 29462)
  PES payload (truncated):
    0000:  00 00 00 00 01 00 02 D4 27 AB 80 00 00 01 B5 84  ........'.......
* PID 0x1000, stream_id 0xE0 (Video 0), size: 11499 bytes (header: 14, payload: 11485)
  PES payload (truncated):
    0000:  00 00 01 00 02 5C 05 03 B8 00 00 01 B5 82 32 33  .....\........23
* PID 0x1000, stream_id 0xE0 (Video 0), size: 14197 bytes (header: 14, payload: 14183)
  PES payload (truncated):
    0000:  00 00 00 01 00 02 9C 3C 23 B8 00 00 01 B5 83 32  .......<#......2
* PID 0x1000, stream_id 0xE0 (Video 0), size: 34630 bytes (header: 19, payload: 34611)
  PES payload (truncated):
    0000:  00 00 00 01 00 03 94 65 CB 80 00 00 01 B5 84 3F  .......e.......?
* PID 0x1000, stream_id 0xE0 (Video 0), size: 14285 bytes (header: 14, payload: 14271)
  PES payload (truncated):
    0000:  00 00 00 00 00 00 00 01 00 03 1C 29 63 B8 00 00  ...........)c...
...
~~~
You can see some payloads starting with `00 00 01` (interpreted as MPEG-2 Video) and some others starting with 3 to 7 `00` before the `01` (interpreted as H.264). When analyzing the video packets, `tsanalyze` finds a weird list of exotic resolutions and profiles (after the bug fix of course). This is explained by the fact that TSDuck interprets MPEG-2 video data as H.264 data. Analysing incorrect H.264 data also triggered the crash in one specific case (now, TSDuck properly recovers).

Honestly, in the last 12 years of TSDuck usage, I have never seen an MPEG-2 video stream with PES payload starting with something else than `00 00 01`. But I must admit I never had the opportunity to manipulate Bulgarian TV streams. I assume that this PES packetization of MPEG-2 video is incorrect. But I am not a real expert here, so I may be wrong.

I assume that VLC trusts the PMT and is tolerant in the video format, ignoring extra leading zeroes in the PES payload.

The problem is that I do not want to blindly trust the video type as declared in the PMT because I know by experience that there are streams with video PID identified as MPEG-2 in the PMT but carrying H.264 in fact. So, in both cases, the stream is incorrect somewhere.

I will think about it. Maybe the solution is to add an option somewhere which says "in case of conflict between the PMT and the content, trust the PMT". The default behaviour will remain the current one. This will not be the first time I add an option to guide TSDuck on how to process incorrect content (see option `--default-charset` for instance).

closed 2018-03-30 by lelegard

yaneti, 2018-03-30:
Hi @lelegard . This is awesome round-trip time for a bugfix  :). 
Thanks for the detailed analysis and your continued work on tsduck. 

In my mostly user experience of various DVB IP equipment software,  I've found it predominantly dodgy, so the fact the this particular stream was coming out broken from that particular encoder is.not a  big surprise.

My only regret is not having heard of or noticed tsduck before this month.