tsduck/tsduck #50: Debian: tsfixcc conflicts with opencaster

KantarBruceAdams, 2018-04-13:
Hi,
     I successfully built tsduck_3.10-674_amd64.deb from source on Debian Stretch 9.4 (current master commit = e58d3c12feaff581c4718e0eac72844bef30c871 Apr 11 23:23:44 2018 +0200).
However, I cannot install the .deb created by make deb as it conflicts with tsfixcc which is also provided by opencaster:

```
    >sudo dpkg -i installers/tsduck_*.deb
    (Reading database ... 138664 files and directories currently installed.)
    Preparing to unpack .../tsduck_3.10-674_amd64.deb ...
    Unpacking tsduck (3.10-674) ...
    dpkg: error processing archive installers/tsduck_3.10-674_amd64.deb (--install):
     trying to overwrite '/usr/bin/tsfixcc', which is also in package opencaster 3.2.2+dfsg-1.1+b1
```
A workaround for me is to uninstall opencaster as I am not currently using it but ideally there should be no conflicting names.
Another option would be to document the equivalent of:
 
./configure --prefix=/usr/local

for whatever build system you are using.
I think as it stands it looks like make install installs to /usr when source installs should install to /usr/local (http://www.pathname.com/fhs/pub/fhs-2.3.html#USRLOCALLOCALHIERARCHY)

Is there any expectation of this package being accepted into Debian in the near future? 

Regards,

Bruce.

PS I also note there is a unit test failure - though this may be okay as this is not a release tag:

```
 make test
source release-x86_64/setenv.sh && release-x86_64/utest

!!!FAILURES!!!
Test Results:
Run:  341   Failures: 1   Errors: 0

1) test: DemuxTest::testTOT (F) line: 375 utestDemux.cpp
assertion failed
- Expression: tot.regions[0].next_change == ts::Time(2008, 3, 30, 1, 0, 0)

Makefile:61: recipe for target 'test' failed
make[1]: *** [test] Error 1
Makefile:55: recipe for target 'test' failed
make: *** [test] Error 2

```

lelegard, 2018-04-13:
Hello,

Thank you for reporting this. Let's take all points one by one.

1) Opencaster

I was not aware that opencaster provided a command named `tsfixcc`. I never used opencaster. All TSDuck commands starts with the prefix `ts`, probably too short to be unique. On the other hand, I would rather not have to type long command names such as `tsduck-fixcc` or whatever.

`tsfixcc`, `tsanalyze`, `tstables` were commands I wrote 12 years ago (although not released in open source at the time) and which were more or less superseded with `tsp` plugins later. But the old commands remained, both for compatibility and because they were shorter to type than a complete `tsp` command.

I can offer no solution right now, except removing `tsfixcc` or ask opencaster to remove theirs :)

2) configure

TSDuck does not use a `configure` script. I have always hated those software with hard-coded paths. I have always considered this as bad design. And the fact that so many people copy those practices is not a proof of good choice, just a lack of imagination. 

On the contrary, all TSDuck commands work relatively to the directory were the executable is located.

So, simply move the whole set of TSDuck binary files somewhere else and they will still work. No need to patch and rebuild.

3) Use of `/usr/local`

Well, this is a disputed topic. Each Unix flavor, and even each Linux distro, has its own set of rules, all being self-proclaimed standards. The FHS is just one of these self-proclaimed standards and it is not without ambiguity. It states that _the `/usr/local` hierarchy is for use by the system administrator when installing software locally_, without specifying what _installing software locally_ means. Many people interpret this as _if you manually copy files for all users, put them in `/usr/local`_, which also means that the `/usr` hierarchy outside `/usr/local` shall be kept clean and maintained by the system packager, `rpm`, `dpkg`, etc, which ultimately means _if you build a software package using the standard system packager, install your files in `/usr`_. Honestly, I cannot recall a `.rpm` or  `.deb` package installing files in `/usr/local`.

There is only one context into which TSDuck installs its files in `/usr/local`, this is macOS. This is because macOS uses a System Integrity Protection infrastructure which prevents even an administrator from writing in `/usr`. This is why commonly-used macOS packagers such as Homebrew use `/usr/local` as secondary - but still centrally managed - software deployment area.

So, the TSDuck packages, `.rpm`, `.deb` or Homebrew, follow the same usages as their corresponding packagers.

That being said, since you rebuilt TSDuck from sources, you may install it anywhere else using
~~~
make install SYSPREFIX=/some/where/else
~~~

4) The failing unitary test

It is a coincidence that, probably at the same time you typed your message, I released version 3.10-674.

So, your failing test is quite worrying for me. But I cannot reproduce it. I build TSDuck on :
- Windows Intel 32 and 64
- Fedora Intel 64
- Ubuntu Intel 64
- Raspbian, ARM 32
- and sometimes CentOS Intel 64

There is no such failure on any of these platforms. Is this the only failing test?

The test manipulates time values, which are built and analyzed by system libraries. So, when handling values which are before the epoch of one specific OS, there could be non-portability issues. But here the date is in 2008, the year I wrote this test. So, this is not the point.

Could you please make sure there was no build issue? At worst:
~~~
make distclean
make test
~~~

Let me know, I don't like this one.

lelegard, 2018-04-13:
OK, I think I get it.

The time from the test is, by definition, the transition time - UTC - for daylight saving time (the next change in a TOT). It is a UTC time.

The system libraries compute times, interpreting them as local time. The last Sunday of March, the range 1 AM (UTC) to 2 AM "does not exist". But this depends on the local time settings of your system.

I am located in Paris (UTC + 1) :
~~~
$ date --date='2008-03-30 01:00:00'
Sun Mar 30 01:00:00 CET 2008
$ date --date='2008-03-30 02:00:00'
date: invalid date ‘2008-03-30 02:00:00’
$ date --date='2008-03-30 02:30:00'
date: invalid date ‘2008-03-30 02:30:00’
$ date --date='2008-03-30 03:00:00'
Sun Mar 30 03:00:00 CEST 2008
~~~

So, the local time of the change (2 AM) is UTC 1 AM. But when the system libraries manipulate "1 AM", they use one 1 AM local time. Anywhere on the planet, except on London meridian, the test works. I see from your profile that you are based in London, this is why it fails on your system. Many people use TSDuck, in Europe, Russia, Asia or America. But it seems obvious that you are the first one to run the unitary tests in London (or somewhere where local time is UTC).

This was unexpected but interesting. I will think at the best way to fix this.

For the time being, the only impact is on very specific time values, so this will likely not impact normal usage.

Interesting, really.

KantarBruceAdams, 2018-04-13:
Thanks for the quick response. I heartily agree regarding hard coded paths. 
I was looking for the SYSPREFIX option. It may be a good idea to mention it in src/doc/building.md and didn't think to look for it in makefile.common

Regarding the use of /usr/local I am not claiming it should be the default for installing a system package like a .deb or .rpm. What I am really saying is that when building from source using "make install" should not install to /usr as that IS reserved for system packages. The conventions I'm familiar with have source installs done to /usr/local for this reason.  
I agree the FHS can be too vague but I think this is one of the points where it is clearer (I sent you an old link when I should have sent you https://refspecs.linuxfoundation.org/FHS_3.0/fhs-3.0.pdf but you seem familiar with it anyway). And while there are some relatively obscure distributions like gobolinux that don't follow it most accept the FHS (as part of the LSB).

lelegard, 2018-04-13:
Hi @KantarBruceAdams 

I have fixed the time bug around DST in commit 97482f7f0a9d052e7a907c057923fc26139c7f43

I added enough unitary tests to spot the error in any local time zone. Since you seem to be a Unix programmer, the core problem is the impossibility on Unix to build a UTC time from a broken-down time (individual values for year, month, etc). You must convert it to local time with obvious issues when the corresponding local time does not exist (ie. the missing hour the last Sunday of March).

You may pull and rebuild if you like (and `make install SYSPREFIX=/usr/local` if you prefer).

Let me know if this is OK for you.

lelegard, 2018-04-14:
> I was looking for the SYSPREFIX option. It may be a good idea to mention it in src/doc/building.md 

Just done.

KantarBruceAdams, 2018-04-16:
Looks good.

closed 2018-04-16 by KantarBruceAdams