tsduck/tsduck #52: no input signal

anupampr, 2018-04-19:
tsp -I dvb --frequency 11,670,000,000 --symbol-rate 29,500,000 --fec-inner 3/4 --polarity vertical --delivery-system DVB-S --modulation QPSK --satellite-number 2 

i am using the code fr locking the signal but the error comes constantly

* Error: dvb: no input DVB signal after 5,000 milliseconds

i am usong TBS 5520se dvb tuner dvb-s

please help

lelegard, 2018-04-19:
Hello,

What is your OS and version?

As far as I tested it years ago `—satellite-number` did not work on Windows. Any dish behind a switch with non-zero index was not accessible. This was one of several limitations of DVB support on Windows. I think it is documented in the TSDuck user‘s guide. 

If you are using Linux, can you receive any other TS on the same dish?

anupampr, 2018-04-19:
Thanks for the replying 
i am using both Linux and Windows 10 
On linux when i list the lsdvbusb only frontend0 is selected i have to select frontend1
it is not happening
please suggest 
tslsdvb
/dev/dvb/adapter0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T, DVB-T, DVB-T2)

dmesg | grep frontend
[    7.841091] usb 1-8: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[    7.841123] usb 1-8: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...

i have to receive signal from DVB-S

lelegard, 2018-04-19:
Ok, so your device has 2 tuners. Frontend 0 is the default and is DVB-T/C. You have to specify frontend 1 to use DVB-S. On Linux, add `—device-name /dev/dvb/adapter0:1`

I have not tested that kind of tuner but it should probably work like this. See the chapter on device naming in the TSDuck user’s guide. 

When I originally developed TSDuck, devices with heterogeneous tuners did not exist. Since they are more common now, I should probably improve `tslsdvb` to report them more accurately.

anupampr, 2018-04-20:
Sure sir, 
please do it will very helpful for me 
till now what should i do???

lelegard, 2018-04-20:
Did you try what I suggested ?
~~~
tsp -I dvb --device-name /dev/dvb/adapter0:1...
~~~

This selects frontend 1 in adapter 0.

anupampr, 2018-04-22:
when i lock the signal the signal also not locked ...
tsp -I dvb --frequency 11,670,000,000 --symbol-rate 29,500,000 --fec-inner 3/4 --polarity vertical --delivery-system DVB-S --modulation QPSK --satellite-number 2
                                                �ë*�5�u�k�8}�X��:FU�p���Y8�e
        "[1�Nk1#����p��H�	�����r�$F���{
                                                R�p�g̻M�R��88J��0��/�K�
                                                                       ����ȣ��ZmN�d�Fx"��ư�ּGr?�� �mN��������l=�O{G��������������������������������������������������

 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_*�,@Y>Zوe�I������f��*+��p8��R���W��-�`v���$�0P-=��	���#��"�0(��
��$eB����$�t��e�d����GH]�
d�P�
     �T�����4��A#E
���A�              �@��B/���q��}����(��"��.���{O��{� ;�M�_}���[���
       ���~�&���_�x��z(pT���j4YA:��?d�J� �(gM�>�
                                                     ��Ms.��D+�ӧ�\�z��k����Pϯ̪�xb�.\�G������������������������������������������������

 !"#$%&'()*+,-./0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_��2�6ZɿԋFm�[��A���|}~���������GQl����0�����9�8
                            @4Er�p-d��E�
                                              6�-��蠀@���`�
��2o�"ц��@�P@ @�G��M�A���                       ���9�8
������ ����@����Z���d�=���n��^�pz1�ZAy��j�֐4a�T
                 �������h`):68!��D	2:��N��	�K����E�,kM[�|�;���?�����1$pE
�$M|z���X�����d�-7,���ln��̪�ɗ������Ffe�UFM��{|�0D9���Hx2��;� �,�4l�F��|N���nAz����N����#T05L<Zֳ^�u��L_�yTtء�a�L<l.2�����7_G'�r��!�`�Dj�N�5Ub������dt˨�{(�tڜ�!��:��8#.bR���
                                 �6
�y^�5d
\A�
�P
����'�3��컌��̷_�U��k7`9���I���݅D�oFЋ��N��DBN�6���D�T"�R�u"&����ȪG3GB]���{\�����}�E� G8LVA���,��.	`��,R�İ��������z� XTG�I4#��9�pn8�5����N`Bb3�3/��������bx�	8��NS��@Ad�>ϺSGС����,�CN�~��a���
                                                     �4ȸ�g�/��xR\�n0-���Y���GK<�(;����_����
                     4	���#c���S��8*k_����s���!i�.�p
                                                        7@>O�����Dx�2($�yd�Â��'�	�?�x
            К�
               q���5''����;�=DP:َ�A\(.��js�3ZU��eT���k�,�
`>-8p0
]^[[?62;c^[[?62;c^[[?6

something is like this

lelegard, 2018-04-22:
What you see is probably the binary transport stream on your screen.

Please read the TSDuck user's guide carefully. To use `tsp`, you need to specify one input plugin, any number of packet processing plugins and one output plugin. The default input and output plugins are `file` which reads/writes files. And the default files are the standard input/output. This means that, when you do not specify any output plugin, the transport stream goes to your standard output. This is fine when the output is a pipe going to some other application. But not if this is your terminal :)

TSDuck is a _toolbox_, not an _application_. There is no unique way to use it. It can do almost everything you want to do. But you need to precisely define first what you want to do.

lelegard, 2018-04-23:
1. Can you describe precisely what you want to acheive.

2. What do you expect commands like these to do ? I mean on a Unix shell syntactic standpoint, regardless of TSDuck commands.
~~~
tsdate --all tsp -I dvb --device-name /dev/dvb/adapter0:1 --tune 11670v2:29500 -P until --seconds 20
tsp -I dvb --device-name /dev/dvb/adapter0:1 --tune 11670v2:29500 -P until --seconds 20 tsdate --all
~~~

3. Please use markdown formatting for shell commands and text output.

anupampr, 2018-04-24:
i want to know that 
in 
tsanalyze [options] [input-file]
Input file:
MPEG transport stream, either a capture file or a pipe from a live stream. Must be a binary stream of
188-byte packets. If omitted, standard input is used.

how to use [input file] as a pipe from a live steam

i am interested in analyze of dvb data .

lelegard, 2018-04-24:
OK. So, to analyze 30 seconds of live DVB input and store the result in a text file, use:
~~~
tsp -I dvb ... -P until --seconds 30 -P analyze -o analyze.txt -O drop
~~~
This is the same but uses less resources than the equivalent using a pipe:
~~~
tsp -I dvb ... -P until --seconds 30 | tsanalyze >analyze.txt 
~~~
There are a few commands such as `tsanalyze` or `tspsi` which work on files (as a command) but have an equivalent filter `analyze` or `psi` which work in the context of a `tsp` processing chain.

closed 2018-05-04 by lelegard

anupampr, 2018-05-14:
Hello Sir I was Out of station Today i MA showing that the post is closed 
Earlier tuning was possible Bute now it shows error
 tsp -I dvb --tune 11470:h:0:32720 -P until --seconds 20 -O file ts_capture.mpg
* Error: dvb: no input DVB signal after 5,000 milliseconds

please help me again

anupampr, 2018-05-14:
when using tslsdvb -v
it is showing no signal locked ??

bastiaan85, 2019-06-18:
In case anyone (like me) finds this issue by searching `tsp: until: no parameter allowed, use options only`: this stems from copy/pasting from the manual PDF that provides a long dash `–` instead of a regular dash `-`. Retype the parameters using the `-` on your keyboard and the command will work.

picatchoof, 2021-08-04:
hello
can you try the  **--signal-timeout 0**

like:
mylinux@SBlab:~$ tsp -vvv -I dvb --freq 11078000000 --symbol 5000000 --fec 9/10 --polarity vertical --delivery DVB-S2 --pilots on --modulation 8-PSK --signal-timeout 0