tsduck/tsduck #54: IP output via VPN stream errors

rezna11, 2018-05-02:
I don't think this is TSDuck issue but maybe there's something in TSDuck that could help me overcome a problem I'm having or at least get some ideas from smarter people here.

I have TSDuck running at LOCATION-1 and the output is IP. When I play the UDP stream locally at LOCATION-1 in VLC, it's all smooth and error free. But when I use the destination IP of a computer at LOCATION-2, I get errors in the stream when I play this one from LOCATION-2.

Locations 1 and 2 are connected over SSTP VPN between 2 Mikrotik routers and the bandwidth is good enough (10Mbps+), yet even a 3Mbps video stream fails to play without errors. I did try to change TTL and packets on the output IP stream but that didn't seem to help.

First I'd like to ask for an idea how to analyze the incoming stream, to see what's wrong. Could you please suggest some tools that you'd consider good for such analysis, to find the cause of the errors? To see what I mean, I also attach a file with a sample captured stream.
[telecaribe.zip](https://github.com/tsduck/tsduck/files/1967004/telecaribe.zip)

Then I also think the issue might simply be UDP over VPN not being reliable. But in such case, would there be another way to deliver the output from TSDuck over the internet to another TSDuck client (or directly a player like VLC)?

Thanks!

lelegard, 2018-05-02:
Hi Jan,

You probably run into the very same problem I got 12 years ago while sending the 2006 football world cup live transmission from my machine with a 100 Mb/s (at the time) Ethernet access to some of my colleagues' computers with 10 Mb/s Ethernet interfaces.

By default, `tsp` privileges performances over latency. Thus, the default options create bursts. When the end-to-end network path has bottlenecks in the middle, some UDP packets are dropped because they can't all make it through the bottleneck during the short period of a burst.

To solve this issue, you need to smoothen the output stream by creating more frequent but smaller bursts.

This is explained in the example section 5.2.4 in the TSDuck user's guide (which I wrote immediately after solving the aforementioned issue). Of course, the values in the example are just examples. But this section explains the problem and the solution. I am sure you will find the right tuning.

I hope that this will solve your issue.

rezna11, 2018-05-02:
Thanks! Simply using the values from your example already improved things to almost perfect stream, so it was definitely the issue and now I'll just try to do some tweaks to get it 100%

lelegard, 2018-05-11:
Hi Jan,
Did you make it?
Can we close this one?

rezna11, 2018-05-13:
Yes, I think this can be closed as the issue was solved for me, although I haven't done any more tests yet.

closed 2018-05-13 by rezna11