tsduck/tsduck #56: dvb: no input DVB signal after 5,000 milliseconds

anupampr, 2018-05-14:
i am connecting in ubntu 
previously the data was coming but now no input dvb signal after 5000 ms comes
please help

lelegard, 2018-05-14:
I am sure that you understand that providing help from such a tiny description is rather challenging.

anupampr, 2018-05-15:
i have TBS5520se tu tuner usb box 

i have to locked the input signal coming to it
while using ts reader the signal is locked and it is working in window 
but in Ubuntu while i am using 
tsp -I dvb --device-name /dev/dvb/adapter0:1 --tune 11670:v:2:29500 -P until --seconds 20 -O file ts1.mpg

earlier the device was tuned and data was saved .
but now it shows error no input DVB signal after 5000 milliseconds

please help

lelegard, 2018-05-15:
When one configuration does not work and the other one works or used to, you must analyze the differences step by step.

If it used to work and no longer does on Ubuntu with TSDuck, what did change in the meantime?

Does it work on any other transponder?

If it works on Windows with TSReader and does not on Ubuntu with TSDuck, what is the relationship between the two machines? Two machines, same machine (dual-boot), one VM (which one is the guest and which one is the host)? Same physical tuner? If it is not the same tuner, did you check antenna connectivity?

anupampr, 2018-05-16:
i am using another configuration it is also not working ...
* The relation is that ts reader is also used for locking the signals
* i have dual boot machine  Ubntu with window 
and tuner is same TBS2250SE Tv Tuner Box
antenna connectivity is same for both the process

but signal is neither locked so that data may occur 
please help

anupampr, 2018-05-16:
i am using command 
tsp -I dvb --device-name /dev/dvb/adapter0:1 --tune 11510:v:0:32720 -P until --seconds 10 -O file tsnew1.mpg
* Error: dvb: no input DVB signal lock after 5000 milliseconds

it comes

anupampr, 2018-05-16:
hello sir 
As u said i did i runs
Thanks a lot 

now please tell me 2 things 
1) i have to save anything that is on the terminal screen such as 
i am extracting pcrverify it shows me on terminal screen but not save on the file command is 
tsp -I dvb --tune 11510:v:0:32720 -P until --seconds 100 -P pcrverify --jitter-max 0 | -O file >pcrextract.txt

it is showing binary data in file and on the screen it is different thing 
i want to save whatever is didplaying on screen

2) how to add local system time in nay .txt file .

anupampr, 2018-05-16:
sier where is ts duck repository in ubntu i am not finding anywhere

please help 

or tell me how to run files from tsduck-master filke in ubntu

lelegard, 2018-05-16:
TSDuck is currently not in any official distro, neither Ubuntu, nor Fedora. It is probably too specific and nobody asked. So, you have to grab the binary packages from the [release area](https://github.com/tsduck/tsduck/releases/latest/).

To rebuild TSDuck from a freshly pulled repository, follow the [builds instructions](https://tsduck.io/doxy/building.html). But you probably already did it.

After building, you have several options:

1. Build the `.deb` package and install it:
~~~
make deb
~~~

2. Directly install in the system location:
~~~
make install
~~~

3. Install somewhere else, for instance:
~~~
make install SYSPREFIX=/usr/local
~~~

4. Run directly from the build directory. This is currently not well documented and I should probably improve this. The appropriate environment is setup using a script which is automatically generated in the bin directory. Depending on your target (release vs. debug and 32 vs. 64 bits), execute one of:
~~~
source src/tstools/release-x86_64/setenv.sh
source src/tstools/debug-x86_64/setenv.sh
source src/tstools/release-i386/setenv.sh
source src/tstools/debug-i386/setenv.sh
~~~
Example:
~~~
$ source src/tstools/release-x86_64/setenv.sh 
$ which tsp
~/tsduck/src/tstools/release-x86_64/tsp
$ tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.12-730
~~~

anupampr, 2018-05-17:
ok sir i will do what u suggesgts thanks a lot ...............

closed 2018-05-17 by anupampr