tsduck/tsduck #57: Support for HiDes UT-100C

lars18th, 2018-05-15:
Hi,

The current version of TSDuck has support for Dektec modulator devices.

One cheap DVB-T modulator is the HiDes UT-100C device, based on the ITE IT-9507 chip:
http://www.hides.com.tw/product_cg74469_eng.html

You can found the official SDK for Linux & Windows here:
https://www.dropbox.com/sh/zcrqdf5xrvfa4wz/AACib5Z7EBErqk35N-DZGt5Pa?dl=0
(See the page http://www.hides.com.tw/downloads_eng.html)

And some opensource Linux driver here:
https://github.com/linuxstb/it9507

The suggestion is if anyone is interested in adding support to this kind of device.
Regards.

lelegard, 2018-05-15:
Interesting device. $169 for a DVB-T modulator. It seems too good to be true. And it looks quite small for a modulator. Did anybody try it ?

They have only DVB-T for now. Any plan for DVB-T2 or -S/S2 ?

What is the target market? According to the sample use case on the site, they seem to target hospitality or very local redistribution. I would have assumed that for such a usage, IP streaming would be preferred. Any details ?

Supporting it in TSDuck could be done through some new `hides` plugin, equivalent to the `dektec` plugin. But this needs of course devices to test. I can review code contributions.

lars18th, 2018-05-16:
Hi @lelegard ,

> It seems too good to be true. Did anybody try it ?

Yes, I have one! And a lot of users around the world (almost in DVB areas) are using it.

> They have only DVB-T for now. Any plan for DVB-T2 or -S/S2 ?

I don't know. It's not a new device (6 years in the market). But the ISDB-T version is available now.

> What is the target market? Any details ?

The target is *testing/development*, also *local distribution* with server devices, and *home-made* environments. I don't do personal tests of 24/7 runs, but the devices seems very stable.

Regarding the software, regular users are using it with:
- Linux, with basic tools like `tsrfsend` or `mpts2dvbt` (opensource tools).
- Astra-SM in Windows (opensource version of the Astra Cesbo).
- Other custom tools based on the SDK.

> Supporting it in TSDuck could be done through some new `hides` plugin, equivalent to the `dektec` plugin. But this needs of course devices to test. I can review code contributions.

If you prepare some skeleton of the `hides` plugin (using the headers in the SDK) then I can do some basic testing/coding.

Regards.

lars18th, 2018-05-23:
> the surrounding software ecosystem isn't great

Sure! And this is the reason for requesting to support it with TSDuck.

@lelegard, what we can do to help you with this enhancement?

F5OEO, 2018-05-23:
Just for your information, piping tsduck to https://github.com/F5OEO/it950x_cmd is working well (Linux only). UT-100C is so working properly. The best should of course shoud be to include it in tsuduck plugin. Will maybe manage to do that later.

lars18th, 2018-05-23:
Hi @F5OEO ,

> Just for your information, piping tsduck to https://github.com/F5OEO/it950x_cmd is working well (Linux only).

Good to point it. Thank you!
However, the idea is to have _multiplatform_ support (as TSDuck runs on Mac, Linux & Windows).

Until the plugin support is added, perhaps you like to add some Mac and/or Windows support to the tool `it950x_cmd_tx`.

F5OEO, 2018-05-23:
As Hides API is the same (Windows/Linux) it should be not so difficult.

lars18th, 2018-05-23:
> As Hides API is the same (Windows/Linux) it should be not so difficult.

So I hope to see soon a multiplatform (Linux/Windows at minimum) of the tool `it950x_cmd_tx`. It will simplifies the work using notebooks with the UT-100C.

Regards.

lelegard, 2018-05-23:
> @lelegard, what we can do to help you with this enhancement?

Write an output plugin of course :)

More seriously, when I have time (ie. not now), I can write a plugin skeleton based on the vendor API.  Then you would have to test it and modify the code when it does not work. However, since I do not have any Hides device, I won't be able to support this plugin, analyze and fix issues. So, in case of problem, you would also have to provide support.

lars18th, 2018-05-23:
Hi @lelegard ,

> when I have time (ie. not now), I can write a plugin skeleton based on the vendor API.

Just to know: a plugin skeleton for HiDes devices, or for _generic_ devices?

> Then you would have to test it and modify the code when it does not work.
...
> So, in case of problem, you would also have to provide support.

Sure! :smile:

lelegard, 2018-05-23:
> Just to know: a plugin skeleton for HiDes devices, or for _generic_ devices?

By experience, there is no such thing as a _generic_ device. Even with a professional and portable API such as the Dektec DTAPI, some devices need specific workflows. For instance, do we need to fill a buffer and up to which volume before starting transmission or not? In which order should the configuration values be set for the device to work correctly? Etc.

The best example of generic device is the Linux DVB API which handles all tuners in a uniform way.

The worst example is the Windows way of handling DVB tuners through DirectShow and BDA drivers.

lelegard, 2018-05-24:
I just had a look at the software environment. What a mess! This is amateurish work, really. I do not know if the code itself is clean or not. But these guys never worked in a professional software delivery environment. 

- No real distribution channel, no deterministic URL to check. Just a page with unclear tables and links. Then a link to dropbox, pointing to a structure of files.
- No way to automatically detect and integrate new version. Only manual checking, search and download.
- No software packaging for the development environment.
- Different integration approaches on Linux and Windows (although the API itself is similar).
- The API based on Win32 types, using types such as `WORD` on Linux. Clashing with PCSC which uses the same ugly trick on Linux.
- On Linux, the makefile is common to the API and the driver. But we need to deploy the driver on execution hosts while the API shall be built on development hosts.
- No proper packaging of Linux driver, no DMKS package (no DKMS package either for Dektec but better versioning and packaging).
- The only clean installation is the Windows drivers. (But still through dropbox, without proper distribution channel).

This may be fine to build a dedicated test application. But it creates a mess to integrate with a larger application with multiple dependencies such as TSDuck. I cannot just drop the HiDes source code into the TSDuck source tree. But there is no clean software environment to install either. My concern is to keep TSDuck easily buildable on all systems. I need to think about it and probably recreate a brand new separate software packaging for HiDes SDK and drivers.

The modulator is extremely cheap, compared to Dektec, which is in turn extremely cheap compared to standard production equipment. This may explain the absence of investment in proper software packaging and delivery. But most free open source applications are still better packaged.

I have always been concerned with software quality and high quality product management. So, that kind of software delivery drives me crazy !

lelegard, 2018-05-24:
I understand the suggestion. But that means that the transport stream would also go through a pipe. It will work. But the architecture of `tsp` was precisely designed to avoid this. There is a large circular buffer inside `tsp`, possibly locked in physical memory if the process has the appropriate privilege. A TS packet is written at one place in the buffer by the input plugin and is never moved until it is read from the same place by the output plugin to be emitted somehow. The various plugins inspect or modify this packet, one after the other, without moving it.

In practice, the TS packets don't move. The various plugins move their sliding view of the buffer. That is better for performances.

lars18th, 2018-05-24:
Hi @lelegard ,

Your opinion is very professional and I agree with all your statements.

However, regarding the idea of supporting the HiDes DVB-T modulator let me to comment some ideas that you can consider:

- TSDuck has three types of "modules": input, output and processing. The target in this case is only for the "output" type. And perhaps too for the "input" if you consider it for different hardware (Dectek, HiDes, etc.). So that's only for I/O modules.
- For these I/O modules, like current `file-input` and `file-output` the code reads/writes from/to the circular buffer. No more, no less. So adding new `pipe-input` and `pipe-output` modules will be useful from a multiplatform point of view (in Windows it's hard to use "mkfifo" for simulate pipes with files!).
- Based on these PIPE modules, will be easy to create a simple API for _external_ (aka plugins) modules. The API can be a simple PIPE for the Transport Plane, and a set of calls for the Control Plane.
- Then current `dektec-input` and `dektec-output` can be ported to this _new_ API for I/O modules. And futhermore, it could perhaps be compiled from outside the TSDuck in a independent module: "TSDuck-IO-dektec".
- And based on this "TSDuck-IO-dektec" module new modules can be developed: "TSDuck-IO-hides", "TSDuck-IO-BDA", "TSDuck-IO-LinuxDVB", "TSDuck-IO-SDR", etc.

Perhaps a bit code rewriting, but with this new I/O API the TSDuck core will continue to be clean every time. And the development of _plugins_ (IO modules) will be external.

What you think about this way?

lelegard, 2018-05-25:
Well, this is reinventing plugins. Plugin can always be developed outside the TSDuck source tree. Plugins or sets of plugins can be developed _on top of_ TSDuck without being _as part of_ TSDuck.

I have done this several times for plugins manipulating company-confidential data or concepts. In that cases, the plugins never went to open source, but they used TSDuck as a framework (just like proprietary applications running on top of the open-source Linux kernel and libraries). The license and technical architecture of TSDuck make this possible.

So, we could develop some open-source `hides` plugin outside TSDuck, in a separate `tsduck-hides` project. I am just a bit reluctant to it.

skunge2000, 2018-06-02:
https://github.com/linuxstb/it9507

Dave Chapman - linuxstb (who also worked on TV Headend) was working on a different driver for the UT100C at one point. That might be of interest.

lars18th, 2018-06-04:
Hi @skunge2000 ,

Thank you for commenting it. However, all people that has the UT-100C and uses it in Linux are in fact using some of the alternative drivers.

The objective here is have an opensouce tool that targets too the Windwos platform. So, as the TSDuck is Linux/MacOS/Windows compatible, and as one SDK for Windows & Linux exists, then it will be interesting to support it.

However, @lelegard has discovered that the SDK has sirious memory leaks. I hope, in any case, that someone will prepare a _dirty_ plugin for supporting it in Windows for _pure testing_.

Regards.

lelegard, 2018-06-11:
Hi,

Did anyone manage to write a portable application using the UTC-100C ? By "portable", I mean a single application source code which can be compiled indifferently on Windows and Linux.

The API is _similar_ but in fact very different in many details between Windows and Linux.

@F5OEO, you mentioned: 

> As Hides API is the same (Windows/Linux) it should be not so difficult.

Did you manage to produce a single source code version of your application which runs on Windows and Linux (without excessive `#ifdef`) ?

Moreover, the implementation of the API source code is available on Linux while it is not on Windows (binary DLL's only). So we, as external developers, cannot merge them.

Why bother defining and developing an API layer if it is different between operating systems?

The provided "test application" is functionally equivalent but has different source code between Windows and Linux. This is a strange interpretation of the concept of portability....

lars18th, 2018-06-11:
Hi @lelegard ,

I try to explain some issues based on my (very) limited understanding of this device:

- The company that manufactures this device seems that has few resources for software development.
- No one has created any open source tools for this device other than for the Linux platform.
- I feel the Windows and Linux API are different, because only one SDK has created at start. And after a lot of time some other person has developed the SDK for the other platform.

So, if you feel it's impossible to support this device, then don't waste your time. And thank you in any case. Your TSDuck project is impresive!

lelegard, 2018-06-11:
OK, probably the best solution would be to obtain that the Windows API becomes open-source. Then, independent software developers could merge the two into a portable solution. No need to open the source code of the Windows _driver_, just the API.

lars18th, 2018-06-11:
Hi @lelegard ,

Regarding the "API", you're talking about the `ITEAPIEXTest_TX.h/cpp` files in the "SDK_Windows"?

lars18th, 2018-06-11:
Hi @lelegard ,

The reputed developer CrazyCat has implemented Windows support for this device in his fork of the Astra-SM tool:
http://gitlab.com/crazycat69/astra-sm/tree/event_monitor/src/hwdev/it95x

I feel you can use the API files from this directory to implement support for it.
So, you don't need nothing more, and you can avoid the ugly SDK!
Regards! :smile:

lelegard, 2018-06-12:
Hi @lars18th 

>Regarding the "API", you're talking about the ITEAPIEXTest_TX.h/cpp files in the "SDK_Windows"?

Yes for the `.h` and no for the `.cpp`. And this is probably here the most insane part of this software. This is where we understand that, despite some effort in the code presentation, the developer has absolutely zero understanding of basic software engineering.  You may find me a bit harsh but this is the kind of thing that makes me mad.

It is an established practice to write modular software. A module (_object_, _class_, _package_, you name it depending on your implementation language) has a public interface and a private implementation. In C or C++, the implementation is in a file `foo.c[pp]` and the corresponding interface is in `foo.h`.

But in the so-called SDK_Windows, `ITEAPIEXTest_TX.cpp` is a test application program (with a `main()`) and `ITEAPIEXTest_TX.h` is the declaration of the API in the DLL. I cannot believe it!

In proper software:
- A `.h` declares what is in the corresponding `.cpp`.
- The source file containing the `main` is probably the **only** `.cpp` without a `.h`.
- A generic and reusable API is not in a file named `*Test*`.
- A generic and reusable API is not mixed with an application, especially a _test_ application.
- The interface of a generic and reusable API is system-agnostic.

On the contrary, the other pointer you mentioned from CrazyCat fully complies with these requirements and seems to be good software. We may do something with it. My only concern is : Is it compatible with the HiDes drivers (closed source on Windows, open source on Linux). If the answer is yes, then this is the most promising track (and it will also preserves my nerve, which I tend to lose when looking at the HiDes SDK :angry: )

lars18th, 2018-06-12:
Hi @lelegard ,

First of all, thank you for your effort with this (good) hardware and (poor) SDK!

Regarding the _official_ SDK I feel you don't need to waste much more effort. Instead of it, use the code from CrazyCat. I confirm that it works in Windows with the _closed source_ driver. I check it before write here about it. However, I don't know if it's works with the Linux _open source_ driver.

In any case, please keep calm. :smile:

lelegard, 2018-06-12:
Since nothing is simple in this world, the code from CrazyCat is only implemented for Windows, there is no Linux implementation. The file `api.h` defines a system-agnostic API to IT95x devices. The file `api-win32.c` is a Windows implementation for `api.h`. But I did not find any Linux implementation. Note that the file `api.c` contains only a few portable bitrate computation routines. It is not a Linux counterpart of `api-win32.c`.

There is an open source implementation for Linux in the HiDes software. But the API is different. So, based on these two incompatible implementations, I suppose we can derive a new API that we can re-implement on Windows _and_ Linux. Just more work...

BTW, the original author seems to be Artem Kharitonov from the Astra project, in the branch event_monitor of his Astra project.

I do not have any HiDes device to test. Before starting any work in this, I need the commitments of at least two testers, one for Windows and one for Linux, who can debug and update the code. @lars18th ? @F5OEO ?

F5OEO, 2018-06-12:
@lelegard No problem if you want some tester / contributor : my favorite environment is Linux (particularly on Raspberry) but have also dev environment on Windows. Sorry about confusion on common Hides API between Linux and Windows. In conclusion, Hides is an integrator who deal with ITE (chip manufacturer). Hides market is mainly amator radio and hospitalities. They sell a reference design from ITE and use SDK and driver provided by ITE. There software skills are not very good as you noticed. Wonder if we can't rewrite driver on userland using libusb and rewrite a common API in order to be OS compatible.

lars18th, 2018-06-12:
Hi @lelegard ,

>  I need the commitments of at least two testers, one for Windows and one for Linux, who can debug and update the code.

I'll try it!

Hi @F5OEO ,

>  Wonder if we can't rewrite driver on userland using libusb and rewrite a common API in order to be OS compatible.

This will be a _third_ API! And, yes libusb is portable. However, at time no drivers for this device exists using it. Instead of this option, an statand Windows driver and a Linux driver are available. So, why not use them? I feel much more complex to rewrite a libusb driver and an API, than only an API. Do you see it as well?

lelegard, 2018-06-12:
> Wonder if we can't rewrite driver on userland using libusb

One of the most important software engineering rules is _"if it ain't broken, don't fix it"_. If we have working drivers on Windows and Linux, just keep them as they are. I can write a DKMS package for the Linux driver. I have already done this for Dektec drivers.

> and rewrite a common API in order to be OS compatible.

Yes, definitely, this is the goal. The implementation of the API hides the system-specific ioctl's.

Volunteers, please contact me directly at contact @ tsduck.io. No need to add too many more messages in this thread.

lelegard, 2018-06-12:
@lars18th , I think we cross-posted the same thing at the same time. Let's continue offline.

lelegard, 2018-06-15:
Hi @lars18th, please send a mail to contact @ tsduck.io if you would like to participate in test and debug of a HiDes plugin for TSDuck.

lars18th, 2018-06-22:
> Hi @lars18th, please send a mail to contact @ tsduck.io if you would like to participate in test and debug of a HiDes plugin for TSDuck.

Sorry! Very busy at time... I need to complete some internal project. :disappointed:

lelegard, 2018-06-22:
Hi,

I pushed a preliminary version of HiDes support in commit 8f93107bf7393df140bf7265c140174ed4f7792e. It is not final. A few stuff are known to not work correctly. But testing is possible. Please provide feeedback here.

You need to pull and rebuild first. Windows users without development environment can download versions for [Win32](https://tsduck.io/download/test/TSDuck-Win32-3.13-803.exe) and [Win64](https://tsduck.io/download/test/TSDuck-Win64-3.13-803.exe). An [updated documentation](https://tsduck.io/download/docs/tsduck.pdf) is also available.

Requirements: The HiDes drivers are available [here](https://github.com/tsduck/hides-drivers/releases).

Note that everything was coded without direct test since I have no HiDes device. Making this work was made possible only because a couple of users made intensive testing and provided valuable feedback. Thanks a lot, guys.

However, a couple of things do not work correctly yet and need direct testing. I am expecting a UT-100C as a loan from a TSDuck user. I expect to finalize the plugin before returning the device to its legitimate owner.

Notably:

- The output regulation on Linux is not very efficient.
- There is probably a driver interaction problem with 32-bit TSDuck running on a 64-bit Windows.

About the software architecture: I kept the two drivers, Linux and Windows, as they are and completely threw away the pseudo-SDK. I set up the project [hides-drivers](https://github.com/tsduck/hides-drivers) to keep a reference copy of the drivers. The Linux driver has been repackaged as a DKMS package to make kernel upgrades fully transparent. The pseudo-SDK has been replaced by a C++ class named `HiDesDevice`. The API of this class is system-agnostic. The utility `tshides` lists installed HiDes devices and their characteristics. The output plugin `hides` drives the modulator.

About the HiDes / ITE software: After having been driven crazy by the pseudo-SDK (see previous posts), I realized that the Linux driver was no better. See the (angry) comments in the Linux implementation of `HiDesDevice` in the TSDuck library.

lars18th, 2018-06-22:
Hi @lelegard ,

Thank you for provide this pre-release!
I'll test the Windows versions this weekend and I provide some feedback.

Regards.

lars18th, 2018-06-22:
Hi @lelegard ,

Very fast & dirty test:

- Windows 7 64bit.
- UT-100C with stock windows driver.
- Installed "TSDuck-Win64-3.13-803" & "TSDuck-Win32-3.13-803".
- Source: One network DVB-T streamer (HDHomeRun), with pid filtering activated for only catch one service (4~7Mbps). Target streaming to multicast 239.0.0.1:10000.
- Execute: `tsp.exe -I ip 239.0.0.1:10000 -O hides -b 8 -c QPSK -f 474000000 -g 1/32 -h 3/4 -t 8K`.
- **PERFECT** reception with both versions, 32 & 64 bit (the 32bit is a little more CPU time consuming).

So, congratulations! It works. :smile:

lelegard, 2018-06-22:
Thanks @lars18th.

What is the average CPU consumption with the 32 and 64 bits versions ?

Could you post the output of `tshides` and `tshides -v` with the 32 and 64 versions ? In case of failure, retry with `--debug=2`.

lars18th, 2018-06-25:
Hi @lelegard ,

Feedback:

> What is the average CPU consumption with the 32 and 64 bits versions ?

Insignificant. Around 1-4% CPU time. But 32bit is usually 3-4%, and 64bit 1%.

> Could you post the output of tshides and tshides -v with the 32 and 64 versions ? In case of failure, retry with --debug=2.

(64bit)
```
$ tshides.exe
0: "IT9507 TX Filter"

$ tshides.exe -v --debug=2
debug level set to 2
HiDes: CLSID_Proxy: 17cca71becd711d0b90800a0c9223196
HiDes: checking "IT9507 TX Filter", CLSID {17CCA71B-ECD7-11D0-B908-00A0C9223196}
HiDes: found device "IT9507 TX Filter"
HiDesDevice: getting device information
HiDesDevice: get filter instance
HiDesDevice: calling GetHandleFromObject
WinUtils.GetHandleFromObject: getting IKsObject interface
WinUtils.GetHandleFromObject: IKsObject found, calling KsGetObjectHandle
WinUtils.GetHandleFromObject: handle: 0x00000000000001CC
HiDesDevice: GetHandleFromObject successful
HiDesDevice: creating event for overlapped
HiDesDevice: checking support for property 0, index 0
HiDesDevice: checking support for property 0, index 1
HiDesDevice: checking support for property 1, index 2
HiDesDevice: checking support for property 1, index 3
HiDesDevice: checking support for property 5, index 4
HiDesDevice: getting USB mode
HiDesDevice: getting driver information
HiDesDevice: getting chip type
HiDesDevice: getting device type
Found 1 HiDes devices

Index: 0
Name: "IT9507 TX Filter"
Device: \\?\usb#vid_048d&pid_9507#ut100cv5201801150904#{fbf6f530-07b9-11d2-a71e-0000f8004788}\{9963cc0e-ee70-11e0-ba8f-92d34824019b}
USB mode: 0x0200
Vendor id: 0x048D
Product id: 0x9507
Chip type: 0x9507
Device type: 11
Driver version: 21.17.39.1
Link-level firmware version: 255.39.2.0
OFDM firmware version: 255.9.11.0
```

(32bit)
```
$ tshides.exe
0: "IT9507 TX Filter"

$ tshides.exe -v --debug=2
debug level set to 2
HiDes: CLSID_Proxy: 17cca71becd711d0b90800a0c9223196
HiDes: checking "IT9507 TX Filter", CLSID {17CCA71B-ECD7-11D0-B908-00A0C9223196}
HiDes: found device "IT9507 TX Filter"
HiDesDevice: getting device information
HiDesDevice: get filter instance
HiDesDevice: calling GetHandleFromObject
WinUtils.GetHandleFromObject: getting IKsObject interface
WinUtils.GetHandleFromObject: IKsObject found, calling KsGetObjectHandle
WinUtils.GetHandleFromObject: handle: 0x000001D8
HiDesDevice: GetHandleFromObject successful
HiDesDevice: creating event for overlapped
HiDesDevice: checking support for property 0, index 0
HiDesDevice: checking support for property 0, index 1
HiDesDevice: checking support for property 1, index 2
HiDesDevice: checking support for property 1, index 3
HiDesDevice: checking support for property 5, index 4
HiDesDevice: getting USB mode
HiDesDevice: getting driver information
HiDesDevice: getting chip type
HiDesDevice: getting device type
Found 1 HiDes devices

Index: 0
Name: "IT9507 TX Filter"
Device: \\?\usb#vid_048d&pid_9507#ut100cv5201801150904#{fbf6f530-07b9-11d2-a71e-0000f8004788}\{9963cc0e-ee70-11e0-ba8f-92d34824019b}
USB mode: 0x0200
Vendor id: 0x048D
Product id: 0x9507
Chip type: 0x9507
Device type: 11
Driver version: 21.17.39.1
Link-level firmware version: 255.39.2.0
OFDM firmware version: 255.9.11.0
```

However, please help me with a trouble with PCR values and this device...

The Hides UT-100C doesn't reestamp the PCR values before broadcasting the stream. So, with my current muxer tool (mptsd) I'm generting a slower bitrate for the MPTS. And the PCR stamp values with this tool are correct (checked it with an external tool for validation). But the problem is when injesting this TS with the tsp tool. The result when capturing the RF signal with a tuner has incorrect timing PCR values. Any idea about how to solve this problem?

For sure, I prefer to generate a slower MPTS. This assures no overflow with the modulator device. And futhermore, solves the problem of the writes with the Hides buggy driver. As you write less data, the internal buffer never goes full. :smile: 

I hope you can help.
Thank you, and you do a very good job! :+1:

lelegard, 2018-06-25:
Thanks for the feedback.

I worried a bit about the 32 bit version on Windows 64 bits. But it is working, so it seems ok.

I worry a little bit more on Linux about the CPU load because of the insane polling architecture of the driver. 

I will try to adjust the details when I receive one device.

The problem with PCR maybe also linked to the insertion of null packets. On a modulator, the output bitrate is fixed and only depends on the modulation parameters. When you provide more that this, the device will slow you down. But if you provide less, it should insert null packets. I have no idea of the HiDes behavior in this case.

lars18th, 2018-06-25:
Hi @lelegard ,

> The problem with PCR maybe also linked to the insertion of null packets. On a modulator, the output bitrate is fixed and only depends on the modulation parameters. When you provide more that this, [...]

As I see, when the bitrate is less, the HiDes modulator just insert null packets to complete the bitrate. And in fact, I prefer to push a slower TS, because is more safe. However, in this case I need a method for _adapting_ the PCR values. Any idea about how to achieve it using the `tsp` tool when doing the `-O hides`?

lelegard, 2018-06-25:
If the modulator inserts null packets but does not fix PCR's, then they are impossible to fix in theory. 

We could easily fix the PCR in the `hides` plugin. We know the precise bitrate from the modulation parameters and we can fix PCR's based on the total number of TS packets between two PCR's in the same PID. The plugin `merge` already does this (based on a less precise bitrate however).

However, if the modulator inserts one or more null packets between the two, the PCR becomes incorrect again. And this time, we have no way to fix this in software because it is too late.

I think that a reasonable approach could be to fix PCR's in `hides`, which will give an acceptable value if the input stream is not too low and not too many null packets are added.

What do you think ?

lars18th, 2018-06-25:
> What do you think ?

Well, I'm not sure... I suspose that the other modulator plugin (`dektec`) doesn't need it, as the driver or hardware fixes the values. Right? In this case, for _compatibility_ then will be interesting to do it inside the `hides` plugin. However, I feel for other scenarios the same functionality (PCR fixing) will be interesting.

In any case: If I like to do it without the `hides` plugin... I can do it?
If not, why not create first an specific plugin for it; and after include the same code (class) inside the 'hides' plugin for automatically apply it?

lars18th, 2018-07-02:
Hi @lelegard ,

I really need a solution for PCR and bitrate _adaptation_. I'm using a tool (mptsd) that generates one output stream with a bitrate that isn't the _exact_ value of the DVB-T modulation bitrate. So, I'm trying to use it with a _lower_ bitrate (around less than 10Kbps of difference). This tool generates the correct PCR values for the target bitrate (checked with an external tool). But, when I like to input this stream to the `hides` plugin I can't obtain a good signal.

At time I'm using this line:
`tsp.exe -r -b 16076256 -I ip 239.0.0.1:10000 --buffer-size 8388608 -O hides -b 8 -c 16-QAM -f 474000000 -g 1/32 -h 2/3 -t 8K`
An the input stream has a bitrate of **16,076,256 bps**; and the modulation has **16,085,561 b/s** (in fact, with a half bit more).

Any idea of the best method for _adapting_ the bitrate?

lelegard, 2018-07-02:
Hello happy TSDuck users,

It seems that the support for HiDes modulators is working fine now. Please test and report feedback here. Next week, if all tests are satisfactory, I will release version 3.13 with HiDes support.

The HiDes / ITE device drivers can be found at https://tsduck.io, then Download -> HiDes drivers ([direct link](https://tsduck.io/download/hides/)).

### Windows

Install the driver. If you already had it, there is no change, no need to reinstall.

Pre-built test installers are available here:

- https://tsduck.io/download/test/TSDuck-Win64-3.13-817.exe
- https://tsduck.io/download/test/TSDuck-Win32-3.13-817.exe
- https://tsduck.io/download/test/TSDuck-Win64-3.13-817-Portable.zip
- https://tsduck.io/download/test/TSDuck-Win32-3.13-817-Portable.zip

### Linux

The original driver from HiDes / ITE was hopeless, with an insane "polling design". I had to modify it (and consequently create a fork of the driver). You must upgrade the DKMS package of the driver. This is version `16.11.10.1w` with `w` for "waiting write", as opposed to the original "polling write".

Then, pull and rebuild TSDuck.

The modified driver is compatible with the previous version. So, if you have legacy applications using the old it950x driver, they should still work. They should even work better since all write operations will succeed at once and no polling / retry is necessary.

lars18th, 2018-07-04:
Hi,

In Windows it works like a charm! :open_mouth: 
Also with the HTTP input plugin... so reading from HTTP and modulating with one UT-100C.
It seems a very powerfull tool for demos using a notebook and a cheap USB modulator.

Perhaps you like to include a simple tool `tsmodulate` that has simple options for setting the DVB-T parameters, select Hides or Dektec (or any future module), and read from HTTP or Multicast.

Regards!

Note: For the HTTP input in Windows I need to allways use `--max-queue` as the default value (1000) is quite small. Perhaps you like to increment it in the source, or comment about it in the documentation.

lelegard, 2018-07-04:
> Perhaps you like to include a simple tool tsmodulate that has simple options for setting the DVB-T parameters, select Hides or Dektec (or any future module), and read from HTTP or Multicast.

Can't you write a simple one-line script to do this ?

TSDuck is all about scripting. It is a set of basic tools. A small number of tools giving the opportunity to write an infinite number of scripts.

lelegard, 2018-07-04:
> It seems a very powerfull tool for demos using a notebook and a cheap USB modulator.

TSDuck was already used for such demos using a Dektec DTU-315 USB modulator.

The DTU-315 is more expensive than a UT-100C (although way much cheaper than a professional / production modulator) but it has the advantage to support all types of modulation (not only DVB-T like UT-100C).

lars18th, 2018-07-04:
> TSDuck was already used for such demos using a Dektec DTU-315 USB modulator.
The DTU-315 is more expensive than a UT-100C  but it has the advantage to support all types of modulation (not only DVB-T like UT-100C).

:laughing: In the order of 30 or 40 times more!!! You can adquire one new UTC-100C as lower as €120!

Seriously, Dektec products are **PRO** devices, even with things for _testing_ and _development_. For home or entusiats the HiDes is the only affordable solution. The only thing I miss with it is the DVB-S modulation!

skunge2000, 2018-07-05:
Keep an eye out for DTA-111s on eBay.  They were selling for <£100 in the UK recently. I think they were from DVB-T ring-main modulators used in shops to play a promotional loop.

PCI not PCI-E - but they work very nicely with TSDuck - and allow for DVB-C modulation (which is useful if you are trying to encode 40Mbs HEVC streams that are too high bitrate for DVB-T)

lars18th, 2018-07-05:
Hi @skunge2000 ,

Thank you for sharing this info! :+1: 
We have now some different low cost options.

In any case, personally I prefer to use only USB devices (I hate internal devices). And the only advantage of the DTA-111s over the UT-100C is the DVB-C modulation (as you mentioned). If you know about a _low cost_ DVB-S or DVB-S2 device, please advise us.

Regards. :smile:

lelegard, 2018-07-06:
Any other feedback or problem with HiDes support in TSDuck?

If there is none, I will release version 3.13 next week.

lars18th, 2018-07-06:
> Any other feedback or problem with HiDes support in TSDuck?

My only problem with it is how to achieve a perfect bitrate with PCR marks. However, I feel this isn't a problem to do a release.

Thank you!

lelegard, 2018-07-09:
I just published TSDuck version 3.13-834 with first official HiDes support.

I close this issue which was the original request for this. Please open new individual issues in case of problem with HiDes devices. 

Thanks again Evariste and Steve.

closed 2018-07-09 by lelegard

Zibri, 2020-02-21:
Small question.. what about UT100A ? is it just an UT100C + demodulator or is better to get an UT100C because it is somehow optimized?