tsduck/tsduck #58: error in buiding

anupampr, 2018-05-19:
i have got an error while building please help me i have googled so much but no luck 

error is 

dpkg --build /tmp/tsduck-deb-root /home/anupam/Documents/TSduck/tsduck-3.11-716/installers
dpkg-deb: warning: parsing file '/tmp/tsduck-deb-root/DEBIAN/control' near line 13 package 'tsduck':
 empty value for architecture
dpkg-deb: error: package architecture is missing or empty
Makefile:126: recipe for target 'deb-main' failed
make[1]: *** [deb-main] Error 2
Makefile:85: recipe for target 'deb' failed
make: *** [deb] Error 2
 please help

lelegard, 2018-05-19:
Try command:
~~~
dpkg-architecture -qDEB_BUILD_ARCH
~~~
If it is not found, then the package `dpkg-dev` is missing. Install it and retry `make deb`. The package `dpkg-dev` was not listed in the pre-requisites to build TSDuck. I just added it right now.

lelegard, 2018-05-26:
Did you manage to build the deb package?
Can we close this one?

closed 2018-06-19 by lelegard