tsduck/tsduck #59: Basic mux capabilities ?

F5OEO, 2018-05-23:
Tsduck (command line) seems not very easy for muxing.
A classical example is to input MPTS , removing one PID and inject it from an other source (usually SPTS).
Don't find (even RTFM) a plugin to remove only one PID (replaced by stuffing). Filter plugin do the opposite.
Inject could be the solution but I assume input is section not ts.

Have I miss something ?

For now, I have 2 solutions : 
- Mix with opencaster tools
- Use tsduck library to implement it

lelegard, 2018-05-23:
To remove one PID and replace it with stuffing, use `-P filter --pid xxx --negate --stuffing`.

To free bandwidth in a TS, I usually use `-P svremove --stuffing service-name-or-id`.

That being said, I agree that TSDuck is not a real MUX. The plugin named `mux` provides some basic capabilities to merge the main TS with a transport stream file. But there is no way to merge two live streams. Professional muxing also requires timing synchronization, which need some specific processing. I had in mind to build a `merge` plugin, using the output of some command to merge into the main stream, some kind of opposite to the existing `fork` plugin. But I never had the time to do it properly.

F5OEO, 2018-05-23:
Ok, sorry for not having seen --negate !
Good to know that you plan a merge plugin ...that will be very usefull.
I try for now to use mux , but unfortunatelly even by using -r 1 (for no looping), if the file is non regular one (fifo) and there is no packet available, the function terminate.

lelegard, 2018-05-23:
> I  try for now to use mux , but unfortunatelly even by using -r 1 (for no  looping), if the file is non regular one (fifo) and there is no packet  available, the function terminate.

Reading from a named pipe returns "end of file" when there is no more writer. So, if your writer terminates when it finished pushing data, the `mux` plugin interprets this as an end of file and terminates if you use the option `--terminate`.

But, anyway, the plugin `mux` is not designed to work with asynchronous incoming data. It reads the file in the same thread as the packet processing. So, if the input is blocking, the whole stream processing is blocked. This is fine for a disk file, probably not for asynchronous devices.

Generally speaking, `tsp` is highly multi-threaded. Each plugin executes in its own thread. This way, real time input or output devices are not blocked by CPU-intensive plugins (such as scrambling for instance). Some plugins which process asynchronous input (`datainject`, `mpeinject`, `spliceinject`, etc) have an additional thread which reads and may block on asynchronous data. An intermediate buffer is used to passed incoming data so that the main thread of the plugin can immediately inserts data in the stream (if any data is available). This is the way a real mux or hypothetical merge plugin should work.

lars18th, 2018-05-24:
I agree with the idea to create a `merge` module and improve the `mux` one.

From my point of view create a reliable MPTS from realtime sources is a must have function (at minimum for testing)! :smile: 

I hope to see it someday, though not too late.

lars18th, 2018-05-24:
Hi @rezna11 ,

>  I realize TSDuck is not meant to be a commercial product and as mentioned it's a lot of work to implement live mux so I'd also totally understand if and why it's not going to happen.

In fact, at time you can found one open source project that can generate a realtime mux:
http://github.com/gfto/mptsd

lelegard, 2018-05-31:
Hi @F5OEO, @lars18th, @rezna11,

You wanted a `merge` plugin ? So, here we go...

I pushed an initial version of a `merge` plugin in the repository. The documentation is not yet in the user's guide. But, `tsp -P merge --help` works as usual. There are not many options anyway for now.

You may pull, rebuild and try. Your feedback is welcome here.

In short, `merge` works the other way round when compared to `fork`. A command is started and the plugin reads the transport stream which comes from its output. The command can be another instance of `tsp` or anything producing a transport stream.

The obvious requirement is that the stuffing bitrate of the main stream is greater than the overall bitrate of the merged stream. Otherwise, it wouldn't fit in the main transport steam.

Currently, only the basic features are provided, so that the resultant stream remains consistent:
- Merge services into the main stream, meaning merging the content of the PAT and SDT.
- Merge EMM streams, meaning merging the CAT.
- Checking and rejecting PID conflicts.
- Dropping all predefined PID's below 0x20, except that the information from the PAT, CAT and SDT are merged into the corresponding tables in the main stream.
- Adjust PCR's in the merged stream according to their insertion point into the main stream.

What it does not do (but probably should do):
- Smoothen the insertion of packets based on instant bitrates of the two streams.
- Any suggestion ?

Here is a tested example (tested on macOS, Linux and Windows). The idea is to merge a service from one live TS into another live TS.

I take my two TS from Astra 19.2 E. Since I do not have two tuners on the same computer at the moment (and most of my devs and tests are run on a Mac for which no tuner exists 😢), I successively capture 30 seconds of each TS:

~~~
tsp -v \
    -I dvb --frequency 11,538,000,000 --symbol 22,000,000 --fec 5/6 --polarity vertical --delivery DVB-S \
    -P until --seconds 30 \
    -P analyze -o ts1022.txt \
    -O file ts1022.ts

tsp -v \
    -I dvb --frequency 11,626,500,000 --symbol 22,000,000 --fec 5/6 --polarity vertical --delivery DVB-S \
    -P until --seconds 30 \
    -P analyze -o ts1028.txt \
    -O file ts1028.ts
~~~

Then, each "live" TS will be a `tsp -I file` with `-P regulate`. But of course, it should work identically with `ip`, `dvb` or `dektec`.

Here is the sample `merge`:
~~~
BITRATE=33,790,850

tsp -v -b $BITRATE \
    -I file -i ts1028.ts \
    -P regulate \
    -P svremove -s 0x114E \
    -P svremove -s 0x1158 \
    -P merge "tsp -b $BITRATE -I file -i ts1022.ts -P regulate -P zap TV5MondeEurope" \
    -P analyze -i 30 -o merged.txt \
    -O drop
~~~

In this command, I first remove the two services with ids `0x114E` and `0x1158` and replace them with stuffing. The idea is of course to have room for insertion.

In the created command, I read another TS, remove everything except the service "TV5 Monde Europe". So, the output of the command is a live SPTS. The unique service from this SPTS is then merged into the main MPTS.

I also replaced the output with:
~~~
    -P zap TV5MondeEurope \
    -O play
~~~
to verify that the merged service is played correctly.

During the captures, I produced two analysis files, `ts1022.txt` and `ts1028.txt` (1022 and 1028 are the transport stream ids of the two streams). After the `merge` plugin, I also produce the analysis file `merged.txt`. Comparing the three files, we can see that the merge was correctly performed.

lars18th, 2018-06-01:
> You may pull, rebuild and try. Your feedback is welcome here.

Great. Good job! :+1: 

Just a question: What about merging the EPG? EIT table pid 18 (0x12) is less than 0x20!

Thank you.

lelegard, 2018-06-01:
> Just a question: What about merging the EPG? EIT table pid 18 (0x12) is less than 0x20!

The idea behind removing PID's less than 0x20 was that they define stream-specific PSI/SI and we should keep the original PSI/SI from the main TS, while merging essential information from the merged stream (currently PAT, CAT, SDT).

But you are right, there are more PSI/SI to merge. EIT for sure. But also BAT and NIT. But these three types of tables are less straightforward to merge. If you look at the source code for the `merge` plugin, merging PAT, CAT and SDT takes less than 30 lines of C++ code each (thanks to the basic building blocks in TSDuck).

There is no such thing as "the EPG" in a TS. "The EPG" is a database of information coming from various TS which are collected by the receiver. The allocation and spreading strategy of the information which collectively form "the EPG" highly depends on the operator and/or broadcaster.

Here are some options which vary among the networks I worked on:
- EITp/f (present/following) only or also EITs (schedule) ?
- When EITs are present, are they on all TS or on one or a few TS with specific bandwidth allocation for the EPG?
- EITp/f Actual only or also EITp/f Other ?
- When EITs are present, does each TS carry its own EITs Actual only ?
- Or do we also have EITs Other ?
- EITs lifetime and repetition rate.
- Network-specific constraints (typically satellite vs. terrestrial, where the number of TS and bandwidth are radically different).

So, here are some of the challenges of merging EIT's:
- Shall we merge EITp/f only or shall we also include EITs ?
- Does the main TS already contain EITs or is it inappropriate to import them?
- Merge EITp/f and/or EITs Actual for the merged services only?
- Or shall we turn EIT Actual for services which were previously removed from the merged stream into EIT Other?
- What should we do with EIT Other from the _merged_ stream which happen to become EIT Actual in the main stream?
- What should we do with EIT Other from the _main_ stream which become EIT Actual because they refer to merged services?
- How to detect duplicate EIT Other? Or combination of EIT Other _and_ Actual (coming from either stream) which become duplicate EIT Other _or_ Actual on merged stream?
- How to keep the global EIT bandwidth below some predefined threshold (which is a critical parameter for operators)?
- Probably more but they don't immediately come to my mind...

You would probably say that all these alternatives should be triggered by options in the plugin. But it is a lot if work. I suspect that implementing a clever merge of EIT is more work than the complete `merge` plugin as it is today. If you look at the code, the `merge` plugin is unexpectedly simple after all.

Let's keep that list of questions for further reference if / when EIT merge is implemented later.

lars18th, 2018-06-01:
Hi @lelegard ,

Thank you for your explanation. And of course I agree with the complexity of the "mixing EIT info". However, to improve this new `merge` plugin, I feel you can do some "simple" enhancements:

- Include support to bypass any PSI/SI table (pids 0x0-0x20) from selectable inputs. That's an option for the user to select one or another from specific source. Example: merge two MPTS from different sources (sat & cable) and use the NIT from one, and the TOT from another. At time, I see that only the "main TS" bypasses all these pids. Why not make it user selectable?

- Also to Implement the option for "external processing" of the EIT tables. Then, in the future a new `eti-merge` plugin can be developed that will does this _hard_ work.

What you think?

lelegard, 2018-06-01:
Yes, this is quite doable.

For each PSI/SI PID from the merged stream, we could have the option to drop (current), replace or remap to some other PID. This means that a future `eitmerge` plugin could identify the two sources of EIT and merge accordingly.

I keep the idea.

lelegard, 2018-06-01:
Thinking twice about it, I will only propose individual or global drop/replace options for PID's below 0x20. A `remap` plugin already exits. So, to support a future `eitmerge` plugin, simply add a `-P remap 0x20=0x1234` (or any other value) in the `tsp` command which processes the TS to merge. The remapped PID is naturally included in the main TS. Then, in the `tsp` command which processes the main TS, use a syntax similar to `-P eitmerge --merge-pid 0x1234`.

I prefer this approach which lets each plugin focus on a dedicated function.

lelegard, 2018-06-01:
Commit dcd48e49e38a433a8131391931db9f4ca55ab8fb contains an improvement of the `merge` plugin with selected options to disable or replace the default service and PID processing.

The corresponding documentation has been added in the user's guide and a complete example is at the end of the example chapter.

I close this issue since the original question is answered using this plugin. Please open new individual issues in case of problem with the `merge` plugin.

closed 2018-06-01 by lelegard