tsduck/tsduck #60: Windows portable binaries missing

lars18th, 2018-05-24:
Hi,

After last commit a new release of Windows binaries is added: the _portable_ version.

However, the files like `TSDuck-Win32-3.12-745-Portable.zip` are still missing in the release section of the github.

Please, upload/update the files.
Thank you!

lelegard, 2018-05-24:
Wow ! You are going too fast !

This is just the last commit, as you noticed, dating back from last night only. That means that the build procedure now creates the portable packages in addition to the system installers. The portable packages will be available with the next release 3.12-_something_.

lars18th, 2018-05-24:
:laughing: 
The reason is for testing these _PORTABLE_ binaries in Windows. :wink:

lars18th, 2018-05-25:
Hi @lelegard ,

I know very well what means _portable_... I'm an intensive user of PortableApps. :smile: 

However, if the binaries are the same, then I only need to extract de EXE files from the installer.

Thank you! :ok_hand:

closed 2018-05-25 by lars18th