tsduck/tsduck #61: support output mpts and write log messages to multiple-files like analyze plugin

samerajarmeh, 2018-05-30:
thank you for this full functional free software.
it will be complete solution if support output to mpts like paid ASTRA  http://cesbo.com/
the other thing is to write output log messages for all plugins to file with multiple-files like analyse plugin
the last thing i need how to filter only scrambled video & audio & caid-provider cas encryption system like (nds)
thanks.

samerajarmeh, 2018-05-30:
first of all thank you for your fast reply
ASTRA here more info about it ("https://cesbo.com/en/astra/") i am interested with ("Multiplexing - preparing streams for DVB modulators")
log messages i know can see them with daemon log but its more functional logs like analyze plagin with Date Time files
for third question i have channels with more than one encryption i want to choose only one encryption pid if filter plugin can choose pids by Descriptor (audio , video, (ecm by caid or provider) and sub) and drop others because streams sometimes change their audio video pids and sure update them when updated from source without restart program
finaly its very helpful software with all documents thanks alot..

lelegard, 2018-05-30:
Sorry, but this isn't more clear than your initial post.
Please explain _precisely_ what you would like to do. 

I have no intention to read and analyze the documentation of ASTRA. Please explain using which mechanism and which format a TS could be injected in it. And, BTW, `tsp` can directly inject into a modulator.

Concerning the log, please precisely which log you want to save, tsp log, a specific plugin log, a specific plugin output?

About the last point, does the `zap` plugin match your requirements? It monitors the PMT and adapt itself to the PID modifications.

closed 2018-06-19 by samerajarmeh

slaweb, 2019-08-26:
Hello, does anyone knows how can I do some cache keys from CI module...I open only 3 ch and I have information that I can open 8-10 ch from that card and module? Thanks

Saentist, 2019-12-25:
@lelegard MPTS can be generated via free software not only [Cesbo ASTRA](https://cesbo.com/en/)
for exmple 
https://github.com/gfto/mptsd
https://github.com/Saentist/mplex13818
https://github.com/linuxstb/dvb2dvb