tsduck/tsduck #63: Filtering AVC NALs for scrambling

clustor, 2018-06-05:
I am trying to write a plugin that needs to set scrambling bits in each TS packet for AVC payload but keep the TS packets in clear (keep scrambling bits as 0) if the packets match criterion such as the following:

- Any transport packets containing any part of a NAL unit Type 1 to 5 header
- Any transport packet containing any part of Filler Data (NAL unit type 12)
- Any transport packets containing any part of an H.264 Video Slice Header
- The entire PPS

I have looked at pes and scrambler plugins and reviewed PES Demux code but it's unclear to me how this could be accomplished with tsduck.  PES Demux operates on an entire PES packet boundary and is able to detect NAL types but doesn't have access to TS packet or TS packet number to be able to set or not set scrambling bits. 

I have a high level working understanding of MPEG2 TS but I didn't yet had to understand the AVC byte stream.  It would be ideal, if a TS packet can be identified if it is part of a NAL unit or if it has NAL units in them with the types. 

Do you have any suggestions on how to go about this? I would appreciate any pointers to write a POC plugin.

lelegard, 2018-06-06:
Hi,

Interesting use case. First question: When you say _"set scrambling bits in each TS packet for AVC payload"_, do you want to also actually _scramble the packet_? Or simply _set scrambling bits_ without actual scrambling? In other words, do you want to do content-based scrambling?

What you want to achieve is quite feasible, although not completely trivial because you want to work at two levels at the same time. Scrambling usually works at packet level only, regardless of the packet content. PES processing, on the other hand, works at a higher level and the interface between the two is the class `PESDemux` which rebuilds a complete PES packet from many TS packets. But at the time the complete PES packet becomes available, as you noted, the corresponding TS packets went downstream a while ago.

I see two classes of difficulties to solve (both are solvable with some effort): detection and modification.

First: detection. To detect NALunit boundaries and type, you need to parse the AVC content. In the general case, parsing AVC requires some interpretation layer (see class `AVCParser`). However, NALunit boundaries are much easier to detect since they are synchronization / recovery points. You need to find sequences of 0000..01 (see H.264 appendix B) and the NALunit type is the lower 5 bits of the next byte (ib. 7.3).

It is simple if you are in a complete PES packet. However, TS packets are only fragments of a PES one. If the sequence 0000..01nn is entirely inside a TS payload, so far so good. But what about a TS packet ending in 00 or 00..00? Is this just a 00 data or is it followed by 00..01nn at the beginning of the next TS packet of the same PID? In short, you need to keep a context per PID. The context shall contain the "end" of the previous TS packet (or the complete TS packet for simplicity).

Second: modification (scrambling). Assuming that you have implemented the detection as described above, you can decide whether or not you need to scramble a TS packet inside the `processPacket()` method of the plugin. There are pathological cases such as the example trailing 00 at the end of a clear packet followed by a 00..01nn to scramble in the next packet. You only discover that you have to scramble it at the next packet (if the NALunit type matches) and we cannot act backward on the previous TS packet. But, I assume that leaving a clear trailing 00 is not a problem, even if it is the start of the header of a NALunit to scramble.

Now, scrambling and ECM generation is a problem which is already solved in the `scrambler` plugin and you should not reimplement it. It is not a good idea to _modify_ the `scrambler` plugin because your need is too specific. However, if you decide to implement the NALunit detection as described above, I can propose the following.

We keep the `scrambler` plugin as it is today with one modification: When the plugin decides that a packet needs to be scrambled (because it belongs to the right PID's of the right service), it invokes a new virtual method called something like `bool needScrambling(const TSPacket&)`. The default implementation always returns `true` and the plugin `scrambler` behaves as it is today.

However, it is possible to implement content-based scrambling (as you want to do) by _deriving_ the `ScramblerPlugin` class and overriding the `needScrambling()` method. The content-based detection is implemented here. You simply return `true` or `false` and the superclass `ScramblerPlugin` does the rest.

This way, I keep TSDuck generic enough but I open the door to any kind of content-based scrambling, yours or others.

Let me know if you are interested.

clustor, 2018-06-06:
Thanks for the prompt response. 

I just need to set the scrambling bits and a downstream device will perform the scrambling based on the bits. 

If it is possible to peek ahead one TS packet, it seems to me that I should be able to achieve what I need,unless I am missing something, with context maintained per PID. 

If I have a current TS packet and next TS packet, I should be able to determine if the NAL start is spanning across two TS packets. I should be able to maintain the current NAL type in the context and skip the packet setting the scrambling bits appropriately until the NAL ends.  I am not sure if I really need to parse the AVC. 

Btw, thanks for open sourcing this project. It's very powerful, extensible and elegantly done.

lelegard, 2018-06-06:
OK, if you need to work on two packets, the simplest way is to organise a one-packet "time shift" per PID. 

You keep a context per PID. 

1. The first time you see a packet from a PID:
- create the context
- load the packet in the context
- replace the content of the packet by a null packet.

2. For each subsequent packet in the PID:
- you have your two packets, one in the context, the next one in the stream
- do what you have to do
- swap the packets between the context and the current packet in the stream 
- if you want to be clean, adjust the PCR when the packet contains one

Since the video bitrate is usually higher than any other data stream, the added latency is negligible.

Philip-Wells, 2018-06-29:
I am working with clustor on the packet scrambler.

I'm trying out the two packet buffer scheme and I've run  into a problem.  To replace the current packet with a null packet I return from the processPacket routine with TSP_NULL.  The next packet that I am fed is the PAT and then the PMT, so it seems the stream is being restarted..

Is returning TSP_NULL the same thing as replacing the contents with a null packet?

lelegard, 2018-06-29:
>  The next packet that I am fed is the PAT and then the PMT, 

PAT and PMT are cycled, so you can get a packet containing a part of them at any time. This is just a coincidence.

> so it seems the stream is being restarted.

No, there is absolutely no way for a plugin to get a packet backward or restarted. No way. It is like this by design.

> Is returning TSP_NULL the same thing as replacing the contents with a null packet?

Yes it is.

lelegard, 2018-06-29:
And if you want to be sure, try `tsdump` on the original file (if you use a TS file for your tests). You will see all packets, from your PID, PAT, PMT, etc. This is quite verbose but precise.

Philip-Wells, 2018-06-29:
The lesson here is "Know your content".  The first 6 packets are PAT, PMT, VIDEO, PAT, PMT, VIDEO.  I was surprised to see the PAT on the 3rd packet and jumped to the wrong conclusion.

closed 2018-08-06 by lelegard