tsduck/tsduck #64: merge and ad-hoc UDP source

rezna11, 2018-06-08:
I've run into an issue when merging an UDP source didn't quite work if the UDP source goes offline and then comes back. TSP needs to be restarted so that the playout continues correctly. 

Here's my source:

`ffmpeg.exe -re -i D:\SW\input1.mp4 -c:a copy -c:v copy -f mpegts -mpegts_service_id 3 -mpegts_pmt_start_pid 5002 -mpegts_start_pid 2001 -metadata service_name="CH-3" -muxrate 9000000.0 "udp://10.0.0.11:10000?pkt_size=1316&bitrate=9000000"`

And this is my tsp command:

`tsp -v --bitrate 30000000 -I null -P regulate -P merge "tsp -I ip 10000 -P pcrbitrate -P regulate" -O dektec --device 0 --modulation DVB-S --convolutional-rate 7/8 --lnb 5150 --satellite-frequency 4095000000 --level -390 --stuffing --bitrate 30000000`

As output I've also done UDP to test with VLC.

Both run on the same machine (10.0.0.11).

When I stop the source and then run it again, the player isn't working properly - if streaming to VLC, only some frames from video are shown. If modulating and playing on a Dreambox satellite receiver, video works but there's no audio. After restarting tsp (while ffmpeg keeps running), it works, so something that's done at the merge start is missing when UDP source resumes. 

If I use the ffmpeg UDP source directly as tsp input, not merging, it works - after stopping and starting ffmpeg again, the player works fine. This is why I think it's related to merging and looks like I need some additional parameters in my command but I'm not sure what that could be.

lelegard, 2018-06-08:
Hi Jan,

I bet a beer at next IBC that the problem would be solved by removing `-P pcrbitrate -P regulate` from the merged command (the one with `-I ip`) :beers: 

Let me explain why. 

The plugins `pcrbitrate` and `regulate` are not panaceas. They serve very specific purposes

A stream has a _rendering_ speed (the speed of the audio / video) and a _transmission_ speed (the speed at which packets go through `tsp`). The plugin `pcrbitrate` is designed to _evaluate the rendering_ speed. The plugin `regulate` is designed to _alter the transmission_ speed.

Real-time streams, from broadcast or multicast, have identical transmission and rendering speeds because they are transmitted to watch TV.

Files, on the other hand, have a very high transmission speed, maybe 1 Gb/s or more on SSD. Consider that you have recorded a 6 Mb/s single program transport stream. Then, using `tsp -I file`, you read it and send it to `-O ip`. The effective reading speed of the file will be 500 M/b for instance. So, on a gigabit network, you send a 6 Mb/s video stream at 1 Gb/s, 166 times faster as it should be. Thus, a 15 mn video is received in 5 seconds and the player displays almost nothing.

In this case, you __must__ use the plugin `regulate` between `-I file` and `-O ip`. The plugin acts as a bottleneck and lets packets flow out at 6 Mb/s only.

But, when the source has the same _transmission_ and _rendering_ speeds, `regulate` is useless. At best, it does nothing. At worst, it introduces undesirable artifacts.

In your main command, `-P regulate` is necessary because `-I null` produces packets at an infinite speed (at CPU speed in fact). But, in the merged command, `-I ip` produces packets when they arrive, with identical transmission and rendering speeds, and `-P regulate` is useless.

So, why does it fail in your case? Because of the behaviour of `-P pcrbitrate` between `-I ip` and `-P regulate` when the transmission is interrupted and restarted.

How does `regulate` know the regulation speed? First, it can be specified in its command line options. Otherwise, it is transmitted from the previous plugin. As a general rule, `tsp` collects the input bitrate, either from the input plugin itself which extracts the bitrate from a hardware input device (this is the case for ASI cards for instance) or, if the input plugin is not able to report a bitrate, `tsp` automatically analyzes PCR's at the output of the input plugin and computes the corresponding bitrate (which implies, by the way, that specifying `-P pcrbitrate` immediately after the input plugin is redundant).

Then, the bitrate is transmitted from plugin to plugin. Some plugins may inadvertently propagate incorrect bitrates while some plugins may force a (correct) recomputation of the bitrate. To illustrate the first case, consider `-I file ... -P zap ...` using sample bitrate values. You reads a complete 36 Mb/s input and `tsp` evaluates this bitrate. Then, `-P zap` extracts a 4 Mb/s service and removes everthing else. But it does not recomputes the TS bitrate. So, the propagated bitrate information is still 36 Mb/s. If this information is not used downstream in others plugins, we don't care. But if we use the bitrate information in `-P regulate -O ip ...`, we will regulate at 36 Mb/s a stream which should be played at 4 Mb/s. This is why, in specific situations like this, we need to recompute the bitrate using `-P pcrbitrate` before `-P regulate`.

OK, I wrote this to re-explain the principles of bitrate propagation / computation in the `tsp` chain.

Now, back to your issue, `-I ip` reads packets and `tsp` computes the bitrate based on PCR's (let's ignore the next `-P pcrbirate` which is redundant). Then, `-P regulate` uses the evaluated bitrate to alter the transmission speed.

As long as your IP source produces packets (with identical rendering and transmission speeds), so far so good. But note that `-P regulate` is useless since it changes the transmission speed at the rendering speed, which are identical anyway.

When your IP source stops transmitting packets, the evaluated bitrate (or _rendering speed_) is ... zero. This information is propagated to `-P regulate` which sets the _transmission speed_ to ... zero. And when the source restarts, packets are stopped by `regulate` (in fact they are _regulated at speed zero_). In practice, since new packets are received, PCR's are extracted and a non zero bitrate is evaluated and transmitted to `regulate`. But it increases slowly. Since packets are arriving faster, the buffer is full and packets are dropped. And the evaluated bitrate is incorrect. And so on...

As a conclusion, use the plugins `pcrbitrate` and `regulate` when necessary only.

rezna11, 2018-06-08:
thanks for great explanation of pcrbitrate and regulate plugins, which certainly helps to understand better when I really should use these... but despite removing the plugins, I don't really see any improvement... my command is now this:

`tsp --bitrate 30000000 -I null -P regulate -P merge -p 0 -p 17 "tsp -v -I ip 10005" -O ip 10.0.0.11:22000`

but if I restart ffmpeg, VLC won't resume playback, not even after reopening the stream, it just shows 1st frame

lelegard, 2018-06-08:
OK, so I lost the beer...

I suggest to replace VLC with another instance of `tsp -I ip...`. In this instance, use plugin `analyze` to create multiple analysis file names every 20 or 30 seconds. Also save the output in a .ts file. Looking at the various analysis files, you can see what is received at various periods of time, before, during and after the interruption.

rezna11, 2018-06-08:
nah, you still deserve the beer, that's the least I could do :)

anyway I did try analyze but don't really see much in there... I'm adding 3 logs, one from fresh start, another with merge source offline and another after resuming ffmpeg source

[analyze_out_20180608_192306_Running.txt](https://github.com/tsduck/tsduck/files/2085633/analyze_out_20180608_192306_Running.txt)
[analyze_out_20180608_192346_Stopped.txt](https://github.com/tsduck/tsduck/files/2085634/analyze_out_20180608_192346_Stopped.txt)
[analyze_out_20180608_192426_Restarted.txt](https://github.com/tsduck/tsduck/files/2085635/analyze_out_20180608_192426_Restarted.txt)

I think the next test for me will be trying merging the UDP stream into something else than null packets, which is what you were doing in your example, I'll just use UDP instead of files

rezna11, 2018-06-08:
No change even when merging into a file input instead of null. But it looks like this might be a network issue. There are some warning messages related to network traffic - screenshot attached. These warnings appear at the moment when the UDP (ffmpeg) source is restarted.

[image]

lelegard, 2018-06-08:
Do not worry about the warning. It says that two distinct sources were detected sending packets to the same destination (the one we are listening to). This is a warning because if two distinct sources are sending packets **simultaneously**, we get a mixture of the two, resulting in garbage. But here, you stopped the first source and restarted it later (using another client port, being identified as a second source). So, no problem in your case.

I looked at the 3 files. They looked perfectly OK. After restart, you receive the same service, using the same PID's, etc. 

You mentioned that it worked when using the IP stream as direct source for `tsp`, but not when it is merged. The only difference I see is the vast amount of stuffing which is dropped into VLC. During the time where the source is stopped, VLC receives 30 Mb/s of pure stuffing without anything to play. Does this disturb VLC? It shouldn't.

Anyway, just to retry without this difference between the working and non-working cases, you may add `-P filter -n -p 0x1FFF` just before the final `-O ip`. Thus, VLC receives no stuffing, just like the working case.

Let us know the result.

rezna11, 2018-06-08:
I tried streaming without stuffing but don't see any difference.

`tsp -v --bitrate 30000000 -I null -P regulate -P merge -p 0 -p 17 "tsp -v -I ip 10005" -P filter -n -p 0x1FFF -O ip 10.0.0.11:22000`
 
I've also tried streaming stuffing only (30 Mbps) by simply running tsp before running ffmpeg, which worked fine (VLC showing 30 Mbps incoming bitrate but no stream yet), and then I started the ffmpeg - this works, the stream is merged and VLC plays it just fine. The issue only comes when the merged UDP stream stops and starts again. If I use the same stream as input, the player is able to continue after ffmpeg restarts.

rezna11, 2018-06-09:
I've done a test with another UDP source, other than ffmpeg, and it works. So the problem is not caused by TSDuck, therefore this ticket can probably be closed. However, I'd like to find out how to configure ffmpeg output to work with the merge plugin. The only difference I see with ffmpeg and D9854 UDP output stream is that with D9854, I don't get the IP warning messages when rebooting the receiver.

lelegard, 2018-06-10:
The IP warning indicates that distinct pairs of source address:port are detected for the same destination. When you run ffmpeg (or any other similar application program) on a desktop/server O/S, it uses a new port number each time. On the contrary, I would not be surprised if an embedded equipment such as an IRD Pro explicitly reuses the same port number for the same purpose.

I let you close this issue when you consider it can be.

rezna11, 2018-06-11:
VLC's UDP output stream also works if it's restarted, so it looks like it's ffmpeg related. I'll test some other UDP sources later.

rezna11, 2018-06-12:
...and not working with astra-sm output, same symptoms like ffmpeg

closed 2019-01-17 by lelegard