tsduck/tsduck #66: error while extracting tables

sachin0606, 2018-06-16:
Hi while i am using the the dump of tables from the ts file getting below error. 
**command:**
tsp -I file 331.ts -P tables --pid 0 --binary-output PAT.bin -O drop
**Error:**
Error: file: synchronization lost after 0 packets, got 0xFC instead of 0x47

what is the issue?

lelegard, 2018-06-16:
This means that your .ts file is corrupted (or is not a transport stream file). A TS packet always starts with 0x47. So, the first byte of a valid TS file is always 0x47 and this value is repeated every 188 bytes all along the file. 

The message tells you that a TS packet starts with 0xFC instead of 0x47 and _"after 0 packets"_ means that this is the very first packet.

Check that your file is valid. If it should be and is simply corrupted, use the utility `tsresync` to get rid of the garbage and extract as many TS packets as possible. There are many options, be sure to read the documentation first.

sachin0606, 2018-06-19:
thanks for the explanation. used tsresync and its worked.

lelegard, 2018-06-19:
OK, good, closing this one, then.

closed 2018-06-19 by lelegard