tsduck/tsduck #67: TBS5520SE can't lock

IPrium, 2018-06-16:
I have Windows 10 x64 and TBS USB Tuner TBS5520SE. I'm using it for DVB-S receiving from EutelSat 13 and it works very good with many programs for TV watching. Today I tried TSDuck, but I can't lock signal.

tslsdvb.exe -ve

0: "TBS 5520 BDA Tuner" (DVB-S)

  Signal locked:    no

  DirectShow graph:
    
    - Filter "NetworkProvider"
      class GUID: CLSID_Microsoft Network Provider
      Some supported interfaces:
        interface IAMovieSetup
        interface IBDA_DiseqCommand
        interface IBDA_NetworkProvider
        interface IBDA_TIF_REGISTRATION
        interface IBaseFilter
        interface IFrequencyMap
        interface IMPEG2_TIF_CONTROL
        interface IMediaFilter
        interface IScanningTuner
        interface IScanningTunerEx
        interface ITuner
        interface ITunerCap
      ITuner:
        Tuning space "Default Digital DVB-S Tuning Space" (DVB-S), DVB type: DVB_Satellite
        Tuning space "Default Digital ISDB-S Tuning Space" (ISDB-S), DVB type: ISDB_Satellite
    
    - Output pin "Antenna Out", id "Antenna Out"
      Some supported interfaces:
        interface IPin
        interface IQualityControl
    
    - Media major type KSDATAFORMAT_TYPE_BDA_ANTENNA
      subtype MEDIASUBTYPE_None
      format FORMAT_None
    
    - Input pin "Input0", id "0"
      Some supported interfaces:
        interface IKsControl
        interface IKsPin
        interface IKsPropertySet
        interface IMemInputPin
        interface IPin
        interface IQualityControl
        interface IStreamBuilder
      IKsPropertySet properties support:
        SIGNAL_STRENGTH (SignalStats) : get
        SIGNAL_QUALITY (SignalStats) : get
        SIGNAL_PRESENT (SignalStats) : get
        SIGNAL_LOCKED (SignalStats) : get
        RF_TUNER_FREQUENCY (FrequencyFilter) : get set
        RF_TUNER_POLARITY (FrequencyFilter) : get set
        RF_TUNER_RANGE (FrequencyFilter) : get set
        RF_TUNER_FREQUENCY_MULTIPLIER (FrequencyFilter) : get set
        LNB_LOF_LOW_BAND (LNBInfo) : get set
        LNB_LOF_HIGH_BAND (LNBInfo) : get set
        LNB_SWITCH_FREQUENCY (LNBInfo) : get set
      IKsControl properties support:
        SIGNAL_STRENGTH (SignalStats) : get
        SIGNAL_QUALITY (SignalStats) : get
        SIGNAL_PRESENT (SignalStats) : get
        SIGNAL_LOCKED (SignalStats) : get
        RF_TUNER_FREQUENCY (FrequencyFilter) : get set
        RF_TUNER_POLARITY (FrequencyFilter) : get set
        RF_TUNER_RANGE (FrequencyFilter) : get set
        RF_TUNER_FREQUENCY_MULTIPLIER (FrequencyFilter) : get set
        LNB_LOF_LOW_BAND (LNBInfo) : get set
        LNB_LOF_HIGH_BAND (LNBInfo) : get set
[... 477 lines cut ...]

=== Tuning spaces

  Tuning space "Local Analog Antenna" (Antenna)
  Tuning space "Local ATSC Digital Antenna" (ATSC)
  Tuning space "Local ATSC Digital Cable" (ATSCCable)
  Tuning space "Analog Auxiliary Input #1" (AuxIn1)
  Tuning space "Local Analog Cable" (Cable)
  Tuning space "Local Digital Cable" (Digital Cable)
  Tuning space "Default Digital DVB-S Tuning Space" (DVB-S), DVB type: DVB_Satellite
  Tuning space "Local DVB-T Digital Antenna" (DVB-T), DVB type: DVB_Terrestrial
  Tuning space "Local Analog FM Radio" (FM Radio)
  Tuning space "Default Digital ISDB-S Tuning Space" (ISDB-S), DVB type: ISDB_Satellite
  Tuning space "Local ISDB-T Digital Antenna" (ISDB-T), DVB type: ISDB_Terrestrial
  Tuning space "Freestyle generic tuning space for STB" ({adb10da8-5286-4318-9ccb-cbedc854f0dc})

tsp.exe -v -I dvb --delivery-system "DVB-S" --fec-inner "5/6" --frequency "10,775,000,000" --modulation "QPSK" --polarity "horizontal" --symbol-rate "27,500,000"

* dvb: using TBS 5520 BDA Tuner (DVB-S)
* dvb: tuned to transponder --frequency 10775000000 --symbol-rate 27500000 --fec-inner 5/6 --spectral-inversion auto --polarity horizontal --delivery-system DVB-S --modulation QPSK --pilots off --roll-off 0.35 -
-lnb 9750,10600,11700 --satellite-number 0
* dvb: expected bitrate from tuning parameters: 42,238,562 b/s
* Error: dvb: no input DVB signal after 5,000 milliseconds

Please suggest, thank you!

IPrium, 2018-06-17:
I connected to remote graph with GraphEdit when the program was dumping all TS data. Please see below.

[image]

IPrium, 2018-06-17:
This is TSDuck's graph
[image]

IPrium, 2018-06-17:
I managed to build graph manually in GraphEdit and it works correct! Tuner gets LOCK to satellite signal.

[image]

lelegard, 2018-06-17:
Good job ! I realise that my technical explanations were superfluous, you master the subject !

Now I need to understand why, in your system, TSDuck built the graph this way. I need some time to think about it and cannot do it right now. So, I keep this one open and will get back to it.

IPrium, 2018-06-18:
Part of log (with translate errors from Russian):

> 
> * Debug: debug level set to 1
> trying to load D:\tsduck\tsplugin_dvb.dll
> symbol tspNewOutput not found in D:\tsduck\tsplugin_dvb.dll
> symbol tspNewProcessor not found in D:\tsduck\tsplugin_dvb.dll
> trying to load D:\tsduck\tsplugin_file.dll
> using --max-input-packets 1,000 --max-flushed-packets 1,000
> * Debug: dvb: debug level set to 1
> * Debug: file: debug level set to 1
> * Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
> * Debug: dvb: found tuner filter "TBS 5520 BDA Tuner"
> * Debug: dvb: found tuning space "Local Analog Antenna" (Antenna)
> * Error: dvb: fail to set default tuning space "Local Analog Antenna": Wrong parameter (translate).
> * Debug: dvb: found tuning space "Local ATSC Digital Antenna" (ATSC)
> * Error: dvb: fail to set default tuning space "Local ATSC Digital Antenna": Wrong parameter (translate).
> * Debug: dvb: found tuning space "Local ATSC Digital Cable" (ATSCCable)
> * Error: dvb: fail to set default tuning space "Local ATSC Digital Cable": Wrong parameter (translate).
> * Debug: dvb: found tuning space "Analog Auxiliary Input #1" (AuxIn1)
> * Error: dvb: fail to set default tuning space "Analog Auxiliary Input #1": Wrong parameter (translate).
> * Debug: dvb: found tuning space "Local Analog Cable" (Cable)
> * Error: dvb: fail to set default tuning space "Local Analog Cable": Wrong parameter (translate).
> * Debug: dvb: found tuning space "Local Digital Cable" (Digital Cable)
> * Error: dvb: fail to set default tuning space "Local Digital Cable": Wrong parameter (translate).
> * Debug: dvb: found tuning space "Default Digital DVB-S Tuning Space" (DVB-S)
> * Debug: dvb: DVB system type is 2 for tuning space "Default Digital DVB-S Tuning Space"
> * Debug: dvb: using tuning space "DVB-S" ("Default Digital DVB-S Tuning Space")
> * Debug: dvb: trying direct connection from tuner (no receiver)
> * Error: dvb: failed to connect pins: No combination of intermediate filters could be found to make the connection (translate).
> * Debug: dvb: SinkPin constructor, ref=1
> * Debug: dvb: SinkFilter constructor, ref=1
>

IPrium, 2018-06-18:
I have full environment to check any of your patches quickly (Visual Studio, GraphEdit, Tuner). I can make additional debug if you need it and you can say to me what to do.

IPrium, 2018-06-18:
I found a workaround:
(1) Unplug TBS5520SE USB tuner
(2) Plug the tuner
(3) use 'tsp' but only once.

In that case I can get LOCK. 'tsp' works only once after tuner connect to PC, if I interrupt 'tsp' I need to make power cycle for the tuner.

closed 2018-06-19 by IPrium